gpiozero/gpiozero #145: 'GitHub' link on pythonhosted docs is broken

claremacrae, 2016-01-12:
Repro:

1. Load http://pythonhosted.org/gpiozero/
2. Click on 'GitHub' at bottom-left
3. On both Firefox and Chrome, this does not go to GitHub...

claremacrae, 2016-01-12:
To be clearer, it does not anywhere!

bennuttall, 2016-01-12:
I've noticed this. Guess it's a bug in the mkdocs template. The "Edit on GitHub" link at the top works but the logo link in the bottom doesn't.

@d0ugal where's the codebase for the readthedocs theme?

waveform80, 2016-01-13:
@bennuttall the original rtd theme is based here: https://github.com/snide/sphinx_rtd_theme but I'm not sure if that's the one pythonhosted is using as it's intended for use on rtd itself (which has a lot more in the bottom left as can be seen in the rest-docs experimental output: http://gpiozero.readthedocs.org/)

d0ugal, 2016-01-13:
@bennuttall https://github.com/mkdocs/mkdocs/tree/master/mkdocs/themes/readthedocs

lurch, 2016-02-09:
A) Is it possible for http://pythonhosted.org/gpiozero/ to now redirect to http://gpiozero.readthedocs.org/ ?
B) There's still a couple of links to http://pythonhosted.org/gpiozero/ on https://pypi.python.org/pypi/gpiozero/1.1.0

bennuttall, 2016-02-10:
Moved to readthedocs - issue fixed

closed 2016-02-10 by bennuttall

lurch, 2016-02-10:
There's still two 'documentation' links on https://pypi.python.org/pypi/gpiozero/1.1.0 which link to http://pythonhosted.org/gpiozero/ - can they be fixed? (Never having added a project to PyPI myself, I've no idea how/where those links are set.)

bennuttall, 2016-02-10:
Ah, yes, that's because pypi and pythonhosted are tied. I'm not sure I can remove them but I'll see if I can.

bennuttall, 2016-02-10:
Hmm. I can't see a way to remove those links.