gpiozero/gpiozero #146: Unable to change gpio pins state from output to input or vice versa.

scottluker, 2016-01-13:
So i'm working on a Simon Says game that has the same eletronics as the [babbage lights and sounds](https://github.com/raspberrypilearning/babbage-lights-n-sounds) project. Within that projects code you are able to switch gpio pins state from outputting to taking an input or vice versa. But I see no where in the documentation to do this with gpiozero.

bennuttall, 2016-01-13:
Can you paste in some example `RPi.GPIO` code so I know exactly what you're trying to do?

You can do this - you have to create an output object, close it, then create an input object on the same pin, for example:

```python
from gpiozero import LED, Button

led = LED(4)
# do something with led
led.close()
button = Button(4)
# do something with button
button.close()
```

Or to use the generic device classes:

```python
from gpiozero import InputDevice, OutputDevice

out = OutputDevice(4)
# do something with out
out.close()
inp = InputDevice(4)
# do something with inp
inp.close()
```

Does that help?

scottluker, 2016-01-13:
Yeah that helps very much actually, thank you. Is there any note of this in the documentation as I went through it and couldn't see anything?

Also would there be any chance of automating the change so if you were to write:

```
def control_led():
    led = LED(4)
    # Control led

while True:
    button = Button(4)
    if button.is_pressed:
        control_led()
```
It would automatically change itself instead of manually having to close it everytime?

Thank you for the help though as i can get it working now using .close() anyway.

bennuttall, 2016-01-13:
My first thought was to add `button.close()` into your `control_led` function, but then where do you close the LED? And your `button = Button(4)` line would fail if `button` already exists. Hard to say without understanding the logic. I don't understand how you are switching from one to the next? Or are they both wired up to the same pin simultaneously? (why?)

scottluker, 2016-01-13:
So they are both wired up to the same pin simultaneously, as in the way [babbage's backpack - gamekit](http://cdn.shopify.com/s/files/1/0188/6794/products/Backpack_4_of_4_1024x1024.JPG?v=1421946820) is. To make the game more compacted and less wiring. But I can use .close() to do it just when I meant automatically I was meaning behind the scenes within the gpiozero module itself.

bennuttall, 2016-01-13:
I guess you can do that without code at all, if it's wired up so that the button press connects the circuit.

Re: docs - the current docs are missing the "advanced" generic `InputDevice` and `OutputDevice` classes. We're moving to a new docs model soon so they will be included.

Glad to have shed some light anyway!

closed 2016-01-13 by bennuttall

scottluker, 2016-01-14:
So my current project uses `RPi.GPIO` and to swap the states from output to input I using `gpio.setup`, something like this:
```python
gpio_pins = [22, 23, 24, 25]
def gpio_setup(state):
    global gpio_pins,
    if state == 0:
        for pins in gpio_pins:
            GPIO.setup(pins, GPIO.OUT, initial=False)
    elif state == 1:
        for pins in gpio_pins:
            GPIO.setup(pins, GPIO.IN, pull_up_down = GPIO.PUD_UP)
```
Where as with `gpiozero` i would have to use this function to do the same thing:
```python
def gpio_setup(state):
    global gpio_pins
        if state == 0:
            redbutton.close()
            bluebutton.close()
            greenbutton.close()
            yellowbutton.close()
            redlight = LED(gpio_pins[0])
            bluelight = LED(gpio_pins[1])
            greenlight= LED(gpio_pins[2])
            yellowlight = LED(gpio_pins[3])
        elif state == 1:
            redlight.close()
            bluelight.close()
            greenlight.close()
            yellowlight.close()
            redbutton = Button(gpio_pins[0])
            bluebutton = Button(gpio_pins[1])
            greenbutton = Button(gpio_pins[2])
            yellowbutton = Button(gpio_pins[3])
```
Because `gpiozero` does `gpio.setup` automatically, you would assume that this code (see below) would work:
```python
def gpio_setup(state):
    global gpio_pins
        if state == 0:
            redlight = LED(gpio_pins[0])
            bluelight = LED(gpio_pins[1])
            greenlight= LED(gpio_pins[2])
            yellowlight = LED(gpio_pins[3])
        elif state == 1:
            redbutton = Button(gpio_pins[0])
            bluebutton = Button(gpio_pins[1])
            greenbutton = Button(gpio_pins[2])
            yellowbutton = Button(gpio_pins[3])
```
As your redefining what you want to do with a pin, so the state of the pin would change from output to input or vice versa. I don't know if this is something that is worth implementing within `gpiozero` that when a pin is redefined it doesn't cause a conflict and simpler code.

bennuttall, 2016-01-14:
How about:

```python
def gpio_setup(state):
    if state:
        [button.close() for button in buttons]
        leds = [LED(pin) for pin in pins]
    else:
        [led.close() for led in leds]
        buttons = [Button(pin) for pin in pins]
```

waveform80, 2016-01-14:
Hmmm... let's cover the points in order: firstly why close is necessary. There's two reasons for this: firstly that gpiozero guards against the case of multiple devices being attached to a single pin on the assumption that this is more likely to be a programming error than deliberate (so far that's been the case; i.e. as far as I can tell at the moment, people using dual-wiring like this is a rarity). The second reason is that Python is a garbage collected language. Consider the following sequence:

```python
from gpiozero import LED, Button

led = LED(4)
btn = Button(4)
```

At the moment you'd get an error when you create the `Button` because pin 4 is already in use by the `LED` object. Let's imagine we get rid of close and have it automatically close when the object is destroyed. In that case you still need to do something prior to instantiating the `Button`:

```python
from gpiozero import LED, Button

led = LED(4)
led = None
btn = Button(4)
```

Now when led's reference count drops to zero, the object is destroyed, the pin is freed and the button can be created. However, this assumes that the garbage collector *immediately* frees objects whose reference counts drop to zero. This is *generally* the case in the CPython implementation but there's no guarantee of this in the language specification and in other interpreters objects may hang around longer. This is the primary reason for the existence of `close`; to provide an explicit means of stating that you want the pin back *now*, not when the GC gets around to it.

Indeed, if you check the current implementation you'll find that `close` is called from `__del__`, the destructor ([devices.py L87](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/devices.py#L87)). However, for various reasons to complicated to go into here (to do with callback references from underlying C libraries) the destructor doesn't always fire in the obvious places. Try the code above in an interpreter and you'll find it'll work, but try it the other way around with the `Button` first and you'll find it fails without an explicit close on the button (this is because of the aforementioned callbacks maintaining a reference on their owning object).

So, that's why you need an explicit `close` method on devices no matter what. Why doesn't RPi.GPIO need such a method? Because it deals with *pins* not *devices*. In the near future, gpiozero may be making such a distinction more explicit by including a `Pin` implementation (see PR #141) below the Device implementation (so if you want to address pins directly and tell them to do things like in RPi.GPIO you can do, although the API's a bit different).

Now, why can't we have a device that combines LEDs and Buttons? After all, there's already a class that flips its pins between input and output: `LightSensor`. However, it does this in order to discharge then time the charging of a capacitor. Fundamentally it's still an input device and thus its `value` is a read-only property providing the current state of the device. In a device that combines a LED and a Button what does `value` (a rather crucial property that lots of other stuff depends on) do? Should the property flip between being read-write and read-only depending on the configuration of the device, or should reading the property always provide the button's state and writing it always alter the LEDs state? (In the latter case you'll no longer be able to chain components with the `source` mechanism; the former requires more difficult questions about what to do with the `source` property in the read-only case).

If this were a common use-case I'd be tempted to sit down and think rather hard about the answers to those questions but as far as I can tell at the moment (and do correct me if I'm wrong!), I think this is a rarity. Given that, I'd much rather keep the internals of the library as simple as possible (my hunch is that this case can't be covered simply with multiple inheritance of `InputDevice` and `OutputDevice`, but would require a novel class combining their functionality and quite a lot of workarounds besides).