gpiozero/gpiozero #147: v1.1 release

bennuttall, 2016-01-13:
Let's get a release out soon. We can leave out any proposed, half-complete, new components for now (they can come in v1.2). Let's migrate to sphynx docs and get the new pin class and other foundation stuff in.

https://github.com/RPi-Distro/python-gpiozero/milestones/v1.1

I know there's a lot of under-the-hood work you've been doing @waveform80, so I'm keen to get that in - whatever's ready. Can you please add issues for each of these changes (and add them here), so I can track where we're up to?

I propose:

- [ ] #143 RGBLED missing active_high param
- [ ] #121 Parent class of PWMOutputDevice should be changed
- [ ] #101 Generic docs need RST

(more to come)

Any documentation issues related to the mkdocs/pythonhosted docs should be checked in the sphynx docs and closed.

I'm going to create a milestone for v1.2, and the new components can go in there. Perhaps it would be good to move to a (roughly) monthly release?

lgcproductions, 2016-01-16:
Sorry if this has been covered. Still getting my feet with using github. Did do a search but could not find a definite yes.

Will the HC-SR04 sonic sensor going to be added or has this already been done? 

Did notice during the search you mentioned about getting your hands on one. More than welcome to send over one of mine. I got 5+ of them.

pcopa, 2016-01-17:
Any word on the MCP 3208 integration?  I would like to help, but I'm unfamiliar with the github concept and new to Python.

I have gotten my MCP3208 to work by modifying a "bitbanger" solution for MCP3008 that I fond on the internet.  

Can I make a copy of my ..usr...py3..gpiozero/input_devices.py, make changes to test and revert to the copy if things don't work out? 

I also have some spare MCP3208's to send if that can help.

waveform80, 2016-01-17:
I'll get on with 1.1 as soon as I can but I've been horribly busy this weekend (not least looking after junior while Holly's sneezing her nose off) and I'm mostly booked up next week. I'll try and find some spare time to knock the items on Ben's list off (they're mostly simple and/or the work's largely done already).

The ultra-sonic sensor will largely depend on how quickly I can get hold of one (sorry Ben; saw Robie and Sam last weekend and completely forgot to remind them to bring over the one you dropped off! I'll try and rectify this soon :), and whether it turns out to be simple to integrate. No promises on this one, in other words.

For the MCP3208, or #134, I'll need to track down the data-sheet to be 100% certain but if it's as simple as I think, that can definitely go in 1.1 (along with its various siblings).

Anyway, I'll try and hack on this in evenings next week but don't let that discourage anyone else from hacking on stuff. In particular, @pcopa if you want to try editing stuff here's a rapidly-bashed-together guide to hacking on gpiozero yourself:

1. Fork the repository (click the Fork button at the top right). This'll make a clone of our repo under your user ID (so you can commit to it; you can't commit to the "official" one without permission for fairly obvious reasons :)

2. Install some useful stuff on your Pi for development: `sudo apt-get install build-essential python-dev python3-dev python-virtualenv python3-virtualenv`

3. Clone your forked repo on your Pi. From your home directory: `git clone https://github.com/pcopa/python-gpiozero.git`

4. Set up a virtualenv to work within (so you don't have to overwrite your system installation of gpiozero). Again, from your home directory: `virtualenv -p python3 gpiozero-env`

5. Activate the virtualenv: `source gpiozero-env/bin/activate`

6. Install your forked clone as a development copy in the virtualenv: `cd python-gpiozero; python setup.py develop` (I hope this bit will work; it largely depends on whether I've got all the deps right above - if it doesn't let me know!)

7. You're now ready to hack! Fiddle with the code under /home/pi/python-gpiozero and test it simply by running python within the same virtualenv (under that virtualenv when you do `from gpiozero import *` you'll be reading your forked version, not the system version).

8. When you're happy with your changes: `git commit -a` (if it whinges about configuration like name+email, follow the instructions it gives; this is one-time setup) to commit the changes to your cloned repo on the Pi, then `git push` to push your commit up to your forked repo on GitHub.

9. If you now visit your GitHub page you should see a prompt near the top inviting you to open a pull request to the official repo with your changes. Feel free to do so! This won't alter the official repo; just open a ticket with your changes which we can review and accept/reject (or request further changes to).

Phew ... hope I've got all that right. I ought to bung all this in a document sometime ... I realize this all seems like a lot of work just to hack on stuff, but there's actually a good reason for each bit. Incidentally, if anything goes horribly wrong you can restart simply like so:

1. Deactivate the virtualenv (if you're in it): `deactivate`

2. Change to your home directory and blow away the virtualenv and the cloned repo: `cd; rm -fr gpiozero-env/ python-gpiozero/`

3. You can now restart from step 3 above.

Anyway, must go, have fun!

pcopa, 2016-01-18:
Thanks @waveform80. This just what I was looking for.

pcopa, 2016-01-18:
@waveform80 FYI
The datasheet for my MCP can be found at: http://ww1.microchip.com/downloads/en/DeviceDoc/21298c.pdf 
I got a lot of 5 pieces for $9.76 in 3 weeks with free shipping at:
http://www.aliexpress.com/item/IC-MCP3208-CI-SL-MCP3208-C-MCP3208-Original-authentic-and-new-Free-Shipping-IC/32258505994.html

pcopa, 2016-01-18:
@waveform80 I've installed everything and made a copy of the process. Maybe you can use it.
Now I can start with the fun part.
--------------------------------------
pi@raspberrypi:~ $ sudo apt-get install build-essential python-dev python3-dev python-virtualenv python3-virtualenv
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version.
python3-dev is already the newest version.
python3-dev set to manually installed.
The following extra packages will be installed:
  libpython-dev libpython2.7-dev python-chardet-whl python-colorama-whl
  python-distlib-whl python-html5lib-whl python-pip-whl python-requests-whl
  python-setuptools-whl python-six-whl python-urllib3-whl python2.7-dev
  virtualenv
The following NEW packages will be installed:
  libpython-dev libpython2.7-dev python-chardet-whl python-colorama-whl
  python-dev python-distlib-whl python-html5lib-whl python-pip-whl
  python-requests-whl python-setuptools-whl python-six-whl python-urllib3-whl
  python-virtualenv python2.7-dev python3-virtualenv virtualenv
0 upgraded, 16 newly installed, 0 to remove and 0 not upgraded.
Need to get 19.5 MB of archives.
After this operation, 27.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython2.7-dev armhf 2.7.9-2 [17.9 MB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-dev armhf 2.7.9-1 [1,188 B]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpython-dev armhf 2.7.9-1 [19.6 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-chardet-whl all 2.3.0-1 [170 kB]
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-colorama-whl all 0.3.2-1 [20.2 kB]
Get:6 http://mirrordirector.raspbian.org/raspbian/ jessie/main python2.7-dev armhf 2.7.9-2 [281 kB]
Get:7 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-distlib-whl all 0.1.9-1 [141 kB]
Get:8 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-html5lib-whl all 0.999-3 [112 kB]
Get:9 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-six-whl all 1.8.0-1 [14.8 kB]
Get:10 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-urllib3-whl all 1.9.1-3 [76.8 kB]
Get:11 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-requests-whl all 2.4.3-6 [241 kB]
Get:12 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-setuptools-whl all 5.5.1-1 [233 kB]
Get:13 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-pip-whl all 1.5.6-5 [126 kB]
Get:14 http://mirrordirector.raspbian.org/raspbian/ jessie/main python-virtualenv all 1.11.6+ds-1 [61.2 kB]
Get:15 http://mirrordirector.raspbian.org/raspbian/ jessie/main python3-virtualenv all 1.11.6+ds-1 [60.5 kB]
Get:16 http://mirrordirector.raspbian.org/raspbian/ jessie/main virtualenv all 1.11.6+ds-1 [17.2 kB]
Fetched 19.5 MB in 12s (1,596 kB/s)                                            
Selecting previously unselected package libpython2.7-dev:armhf.
(Reading database ... 123994 files and directories currently installed.)
Preparing to unpack .../libpython2.7-dev_2.7.9-2_armhf.deb ...
Unpacking libpython2.7-dev:armhf (2.7.9-2) ...
Selecting previously unselected package libpython-dev:armhf.
Preparing to unpack .../libpython-dev_2.7.9-1_armhf.deb ...
Unpacking libpython-dev:armhf (2.7.9-1) ...
Selecting previously unselected package python-chardet-whl.
Preparing to unpack .../python-chardet-whl_2.3.0-1_all.deb ...
Unpacking python-chardet-whl (2.3.0-1) ...
Selecting previously unselected package python-colorama-whl.
Preparing to unpack .../python-colorama-whl_0.3.2-1_all.deb ...
Unpacking python-colorama-whl (0.3.2-1) ...
Selecting previously unselected package python2.7-dev.
Preparing to unpack .../python2.7-dev_2.7.9-2_armhf.deb ...
Unpacking python2.7-dev (2.7.9-2) ...
Selecting previously unselected package python-dev.
Preparing to unpack .../python-dev_2.7.9-1_armhf.deb ...
Unpacking python-dev (2.7.9-1) ...
Selecting previously unselected package python-distlib-whl.
Preparing to unpack .../python-distlib-whl_0.1.9-1_all.deb ...
Unpacking python-distlib-whl (0.1.9-1) ...
Selecting previously unselected package python-html5lib-whl.
Preparing to unpack .../python-html5lib-whl_0.999-3_all.deb ...
Unpacking python-html5lib-whl (0.999-3) ...
Selecting previously unselected package python-six-whl.
Preparing to unpack .../python-six-whl_1.8.0-1_all.deb ...
Unpacking python-six-whl (1.8.0-1) ...
Selecting previously unselected package python-urllib3-whl.
[... 103 lines cut ...]
source/event_gpio.c:75:5: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     write(fd, str_gpio, len);
     ^
source/event_gpio.c: In function ‘gpio_unexport’:
source/event_gpio.c:90:5: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     write(fd, str_gpio, len);
     ^
source/event_gpio.c: In function ‘gpio_set_direction’:
source/event_gpio.c:117:9: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
         write(fd, "in", 3);
         ^
source/event_gpio.c:119:9: warning: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
         write(fd, "out", 4);
         ^
source/event_gpio.c: In function ‘gpio_set_edge’:
[... cut ...] e of ‘write’, declared with attribute warn_unused_result [-Wunused-result]
     write(fd, stredge[edge], strlen(stredge[edge]) + 1);
     ^
zip_safe flag not set; analyzing archive contents...
RPi.__pycache__.GPIO.cpython-34: module references __file__
Adding RPi.GPIO 0.6.1 to easy-install.pth file

Installed /home/pi/gpiozero-env/lib/python3.4/site-packages/RPi.GPIO-0.6.1-py3.4-linux-armv7l.egg
Finished processing dependencies for gpiozero==1.0.0
(gpiozero-env)pi@raspberrypi:~/python-gpiozero $

waveform80, 2016-01-19:
Excellent stuff! Just remember, in case you reboot your Pi or login again or something, that before hacking on gpiozero you want to enable the virtualenv with `source gpiozero-env/bin/activate` from your home directory. In other words, when your prompt starts with "(gpiozero-env)" you're in the virtualenv so Python will be looking for packages (like gpiozero) within the virtualenv (which in turn points to the files in your cloned repo). When your prompt *doesn't* start with this, Python will look for packages in the regular system locations.

That's really all you need for development of gpiozero at this stage. Later on if you get to creating new files or updating PRs there's more to learn but we can cross that bridge if/when you come to it.

One final thing I will add, because it's probably going to be important when I start merging some of the bigger PRs like rest-docs and pins: it's worth working on a separate branch to master (as I'm going to be merging stuff into master). So, before you commit anything to your clone (don't worry if you already have; we can sort it out :), you might want to do something like `git checkout -b my-cool-changes` which will create a new branch called "my-cool-changes" from master's current head and check it out immediately. Any commits you make at this point will be made on the new branch rather than master which means that as we evolve master you can more easily "rebase" your commits to keep any potential pull requests mergeable with the current head.

If all that sounds horribly confusing, don't worry - and if you've already made some commits, don't worry either - just let me know and I'll tell you how to get your new commits onto a new branch and reset master to where it was.

pcopa, 2016-01-20:
Ooh -ooh, hope I didn't screw thins up. Did  notice that after restarting my test.py, Python was working out of the wrong directory, but after moving my test.py into the python-gpiozero folder, and starting it there, everything worked fine.

waveform80, 2016-01-31:
Okay, few other things to the 1.1 list:

- [ ] #135 add blink to RGBLED (maybe with fancy color-fade ... mmm)
- [ ] #126 add an LED graph class
- [ ] #94 get blink working on composite boards (whatever we use on #135 should be useful as a fix here)
- [ ] #118 add initial parameter to OuptutDevice constructor

waveform80, 2016-01-31:
Oh, and before I forget it, the pins stuff in #141 (I've been keeping that rebased as I do stuff so it should be mergeable easily although I'd like to add some more docs to it and if I get the time, support for pigpio pins)

waveform80, 2016-02-08:
Okay, everything on this list is now in so if there're no objections I'm going to close this, and start prepping the 1.1 release.

bennuttall, 2016-02-09:
I made a gist of the instructions from above: https://gist.github.com/bennuttall/9f35ab1f41136fbf10fd

Mostly for my own reference, but if they help anyone else, great.

closed 2016-02-10 by bennuttall