gpiozero/gpiozero #148: Unable to use gpiozero with onboard ACT LED (led0)

shmick, 2016-01-16:
I set the led to gpio mode with:
`echo gpio | sudo tee /sys/class/leds/led0/trigger`

Yet I'm still unable to address the onboard green / activity led.

raspbian jessie
python2.7 = 2.7.9-2
python-gpiozero = 1.0.0

Tested with and without `active_high=False`
```
from gpiozero import LED
from signal import pause
from time import sleep

led1 = LED(16, active_high=False)
led1.blink()

while True:
  print(led1.is_lit)
  sleep(.3)

pause()
```
I get a stream of True and False results, but the LED never comes on.

```
True
True
True
True
False
False
False
True
True
True
```

shmick, 2016-01-16:
Blurg.

Turns out I was using the wrong GPIO number. 16 and active_high=False is only for the mode B.

The newer B+, A+, 2B, and Zero** use:
```
red = LED(35)
green = LED(47)
```

**No onboard red LED on the zero

waveform80, 2016-01-17:
Huh ... didn't even know that was possible - thanks for the info! I wonder if that should be bunged in the docs somewhere (in an advanced section, obviously).

bennuttall, 2016-01-17:
And camera LED? Or leave it to picamera

bennuttall, 2016-01-17:
Knew it was possible to toggle the activity light, never knew which pin number it was! Thanks for the info.

I forget, do we have an upper limit on pin numbers? Don't think so. I must give the compute module a try :)

shmick, 2016-01-18:
I think it's worthwhile to add this into the docs. It's a no-cost solution to 1 or 2 LED outputs, depending on the board you have.

waveform80, 2016-01-18:
After the pins PR is merged, the pin limit will be defined by whatever module is implementing the pins (naturally). Both RPi.GPIO and the native implementation allow all pin numbers from 0 to 53 (as that's all the GPIO registers in /dev/mem) - I think RPIO was stricter but can't recall off the top of my head.

waveform80, 2016-02-08:
Yup, let's add these to the docs (I'll add a little recipe somewhere at the end); as mentioned above - it's a straight-forward way of playing with the library without any additional hardware at all which is a decent use-case.

bennuttall, 2016-02-08:
Look ma: no additional components!

https://vine.co/v/i1pEVeJ2X9Y

waveform80, 2016-02-08:
Very nice :)

waveform80, 2016-02-08:
Out of idle curiosity, can you use PWM on them?

bennuttall, 2016-02-08:
Yep! Really nice for demos actually!

https://gist.github.com/bennuttall/aae0ec488216033774ac

waveform80, 2016-02-08:
Oooh ... that's nice! Like the demos too...

bennuttall, 2016-02-08:
https://vine.co/v/i1DhW2xQYXU

bennuttall, 2016-02-08:
These (or similar) should definitely go in the docs

lurch, 2016-02-09:
> Knew it was possible to toggle the activity light, never knew which pin number it was!

I believe the relevant GPIOs to use (and whether they're active_high or active_low) can be snagged from `/proc/cmdline` ? Maybe it'd be worth having two built-in classes for these two LEDs (e.g. PiPowerLED and PiActivityLED) which automatically do the right mapping (much like the devices in boards.py), based on the current cmdline? (which would *also* have the advantage of working transparently for those users who have remapped those functions to off-board LEDs by overriding them in `cmdline.txt` - I remember seeing this being discussed on the RPi forums once). I guess these classes could automatically save / restore the previous trigger on open / close too?

Although doesn't the PowerLED (on some models) also have something to do with over-current/low-voltage detection? Is it 'safe' to mess about with it from user-code? And regarding the CameraLED I guess there might be similar issues if both user-gpio-code and camera-apps code are trying to twiddle the LED at the same time? \*shrug\*  I guess Ben will be able to ask the hardware engineers for more info :)

> I must give the compute module a try :)

See the bottom of http://wiringpi.com/wiringpi-and-the-raspberry-pi-compute-board/  :-)

> Both RPi.GPIO and the native implementation allow all pin numbers from 0 to 53

Given that GpioZero is designed (primarily) for beginner users, and using the info available in `/proc/cpuinfo` we can determine which flavour of RaspberryPi we're running on and therefore which GPIOs are available on the headers; would it make sense for GpioZero by default to throw an error (or warning) if the user tries to use a GPIO that *isn't* on the header of the current-flavour-Pi, to help make typos more obvious? (with maybe an init-option to turn this off for users who feel this is too Nanny-ish)

waveform80, 2016-04-01:
@bennuttall any word on controlling these LEDs on the Pi 3?

@lurch interesting idea about warnings for non-physical pins - given how flexible Python's warning system is that's worth exploring - could you open a ticket to remind me? I won't get around to it in 1.2 but maybe next release...

closed 2016-04-01 by waveform80

lurch, 2016-04-03:
(I think) I remember seeing some C-code on the forum for controlling the LEDs on a RPi3 via its gpio-expander using the "Mailbox" mechanism - I'll have to see if I can find it again...

waveform80, 2016-04-03:
I've had a brief read of whatever I could find on the forums - looks like the Pi3 has an I2C-based GPIO expander (ah ha ... that's one of the use-cases of the pins implementation justified already! :) so assuming we can code up a device to represent it, it should be fairly simple to control the LEDs. Related, I definitely like the PiPowerLED and PiActivityLED ideas - they're worth exploring in the future once GPIO expanders have been explored a bit further.

lurch, 2016-04-07:
IMHO the current inability to control the Power and Activity LEDs on a Pi3 is no reason to prevent those classes (or accessor functions / whatever) being added to the current code - they'd simply raise some kind of NotSupportedError if used on a Pi3 (as would trying to access the Power LED on an original Model B or PiZero), and then the Pi3 implementation could be 'filled in' later once the necessary i2c expander functionality has been added at a later date. But perhaps this should be a separate issue targeted at v1.3 ;)

@bennuttall Any news from the hardware engineers about my earlier question? "Although doesn't the PowerLED (on some models) also have something to do with over-current/low-voltage detection?"
Or does changing the LED's trigger-mode ( http://gpiozero.readthedocs.org/en/latest/recipes.html#controlling-the-pi-s-own-leds ) automatically disable that functionality too?

waveform80, 2016-04-07:
@lurch That's a good point - if you want to bash something together feel free! I'm probably going to have my hands full today and tomorrow, then next week it's picademy time again so I really want to get this release out before then, but otherwise it'll be a 1.3 thing

dglaude, 2016-06-14:
I am on a PiZero and I have problem applying the recipes:
```
pi@ZeroView:~ $ sudo apt-get install python-gpiozero
pi@ZeroView:~ $ echo none | sudo tee /sys/class/leds/led0/trigger
pi@ZeroView:~ $ echo gpio | sudo tee /sys/class/leds/led1/trigger
pi@ZeroView:~ $ cat blink.py 
from gpiozero import LED
from signal import pause

activity = LED(47, active_high=False)
activity.blink()
pause()

pi@ZeroView:~ $ python blink.py 
/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py:95: PinNonPhysical: no physical pins exist for GPIO47
  'no physical pins exist for GPIO%d' % number))
```
Maybe I hit a limitation of rpigpio and I should try another pin factory that accept to talk to GPIO47?

Please notice that this is a Jessie-Lite up to date distribution, so it might lack some things that would be present in Jessie-Fat. Also this is not the Pi on witch I do my "environmental variable controled pin factory" test/developpement. But if needed, I "master" that topic.

Any help or hint will be appreciated.

PS: Since this is a ZeroView, with a camera, having a way to blink the camera LED would also be fun...

lurch, 2016-06-15:
@dglaude That's only a Warning, not an Exception https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L93

When I run your code on a Zero, I get the same warning message, but the activity LED does then blink :-)

@waveform80 I've always wondered - in http://gpiozero.readthedocs.io/en/v1.2.0/recipes.html#controlling-the-pi-s-own-leds why does one of the echo commands use `none` and the other uses `gpio`? And it would be nice if you could explain which of led0 / led1 match up to ACT / PWR, so the user can choose to only assign GPIO-control to one of them, rather than both at the same time.

lurch, 2016-06-15:
> PS: Since this is a ZeroView, with a camera, having a way to blink the camera LED would also be fun...

I haven't tested myself, but it shouldn't be too hard to adapt the example [here](http://www.raspberrypi-spy.co.uk/2013/05/how-to-disable-the-red-led-on-the-pi-camera-module/) from RPi.GPIO to GpioZero.

lurch, 2016-06-15:
P.S. Just found https://docs.python.org/2/library/warnings.html#temporarily-suppressing-warnings
and can confirm that changing your example to:
```python
from gpiozero import LED
from signal import pause
import warnings

with warnings.catch_warnings():
    warnings.simplefilter("ignore")
    activity = LED(47, active_high=False)
activity.blink()
pause()
```
stops the warning being displayed.

waveform80, 2016-08-14:
> @waveform80 I've always wondered - in http://gpiozero.readthedocs.io/en/v1.2.0/recipes.html#controlling-the-pi-s-own-leds why does one of the echo commands use none and the other uses gpio?

Good question. The short answer is "I don't know for certain". The longer has to do with which triggers are available for those devices (that's the bit I don't know :).

> And it would be nice if you could explain which of led0 / led1 match up to ACT / PWR, so the user can choose to only assign GPIO-control to one of them, rather than both at the same time.

Nice idea - I'll fix that in a mo.

waveform80, 2016-08-14:
>> @dglaude That's only a Warning, not an Exception https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L93

> When I run your code on a Zero, I get the same warning message, but the activity LED does then blink :-)

Yup, that warning's expected. It's basically saying "there's no header on which GPIO47 is exposed ... are you really sure you wanted that?" The intention of that warning is so if people mis-type a GPIO pin number and select something which isn't available via any header, they at least get a hint as to why their script isn't working as expected.

@lurch has already posted how to suppress the warnings (actually you don't even need the "with" block - the simplefilter is enough). We could add those pins to an additional "fake" header in the pins database to suppress it in the case of the relevant models (A+, B+, 2B) - not sure if that's useful or not though given that sys/class needs editing just to make them accessible anyway.

pieman64, 2016-08-29:
I realise this issue was closed some time ago but I wondered if anyone knew how to suppress the error message for the PWM example? I tried various combinations based around suppressing the error message for the blink example but none worked. This is the code I have for the zero.

```
from gpiozero import PWMLED
from time import sleep
import warnings

#with warnings.catch_warnings():
#    warnings.simplefilter("ignore")
#    dunno what would go here

green = PWMLED(47)

while True:
    for i in range(11):
        green.value = i / 10
        sleep(0.05)
    for i in reversed(range(11)):
        green.value = i / 10
        sleep(0.05)
```

lurch, 2016-08-30:
I think the 'dunno what would go here' would just be your `green = PWMLED(47)` line (indented as appropriate). And you'd still want the `active_high=False` part too, so that increasing `green.value` increases the brightness of the LED.
Let me know if it doesn't work and I'll try it myself later on a real Pi Zero.

lurch, 2016-08-30:
(and of course a neater way to do your PWM-fading would be to just use http://gpiozero.readthedocs.io/en/latest/api_output.html#gpiozero.PWMLED.pulse )

pieman64, 2016-08-30:
Thanks @lurch 

This is working fine:

```
from gpiozero import PWMLED
from time import sleep
import warnings

with warnings.catch_warnings():
    warnings.simplefilter("ignore")
    green = PWMLED(47, active_high=False)

while True:
    for i in range(11):
        green.value = i / 10
        sleep(0.05)
    for i in reversed(range(11)):
        green.value = i / 10
        sleep(0.05)
```

commit 6b2e0149 (2016-04-01): Fix #148

Added recipe for power/activity LEDs (on A+/B+/2B, with notes for other
models)
commit a59f8f6b (2016-04-01): Merge pull request #242 from waveform80/built-in-leds

Fix #148