gpiozero/gpiozero #149: blink sequence method

shmick, 2016-01-18:
I was using the led blink method and was wondering if it could be possible to add a blink sequence method, where multiple on / off times could be defined?

lurch, 2016-02-01:
https://github.com/lurch/python-gpiozero/commit/ae366f3043eb452614e51c3f1916c41e3e40d72f is one possible way of doing it (which I won't be submitting as a PR, because it's only a rough first stab).

You'd create your LED object as normal, and then the `sequence` expected by `blink_sequence` is a list of tuples of `(on_time, off_time)`, e.g. you could do something like:
```python
from gpiozero import LED
from signal import pause

led = LED(17)
led.blink_sequence([(0.2, 0.3), (1.2, 0.5), (0.4, 1.5)])
pause()
```

And then the standard `led.blink()` is identical to `led.blink_sequence([(1, 1)])`

bennuttall, 2016-02-07:
This might be a case of API boat, but how about passing in a `sequence` arg to `blink()` that takes a list/tuple of numbers, on/off in sequence, then repeated:

```python
led = LED(17)

sequence = (0.2, 0.3, 1.2, 0.5, 0.4, 1.5)  # or list of 2-tuples like lurch suggested
led.blink(sequence=sequence)
```

Though I guess that conflicts with `on_time` and `off_time` if you specify those as well. They could be ignored in that case. I'm not sure really.

I think most sequences would be calculated anyway (i.e. `on_time` doubles as `off_time` halves) so that could be done more easily in a loop, though that would have to be in the foreground.

bennuttall, 2016-02-10:
I'm leaning on the side of voting that this is API bloat. There are already four (optional) parameters for `blink`. Those four parameters provide the flexibility to do clever things with `blink`. Besides `on_time` and `off_time` there's:

- `n` => Specify the number of times to cycle (e.g. `led.blink(n=10)` would blink 10 times, then stop)
- `background` => Specify whether the blink loop should take place in the background or the foreground

Used in combination, this can allow a repetitive sequence:

```python
while True:
    led.blink(on_time=2, off_time=1, n=10, background=False)
    sleep(1)  # because it doesn't do the final off_time
    led.blink(on_time=1, off_time=2, n=10, background=False)
    sleep(2)  # because it doesn't do the final off_time
```

This would go on for 2 seconds, off for 1, a total of 10 times, then swap to on 1, off 2 for 10 times, then repeat.

The only issue is as this is in the foreground you can't do anything else at the same time, without breaking out another thread (which, yes, you'd usually do with blink when `background` is `True`).

lurch, 2016-02-10:
> I'm leaning on the side of voting that this is API bloat.

Yeah, with `blink` already having *so* many parameters (don't forget that the PWMLED version of `blink` also has `fade_in_time` and `fade_out_time`), I'm inclined to agree.
Unless @shmick has a compelling use-case?

shmick, 2016-02-11:
My use case was to blink the led in various sequences ie: blink X times and then pause for Y seconds, running in the background. Having to run this in the foreground stop other things from running.

bennuttall, 2016-02-15:
@waveform80 Any opinion on this?

waveform80, 2016-02-15:
Erm... off the top of my head (and bearing in mind I'm slightly frazzled): it sounds like a decent use case, I can imagine several people might desire such a thing, but I am vaguely concerned about API bloat. Is there a simple API we could whip up for this ... quite possibly. If you look at the current implementation of `PWMLED.blink`, it'd be quite easy to split out the first chunk of that, which generates a repeated sequence of `(value, delay)` tuples, and allow the user to specify their own sequence of `(value, delay)` tuples as suggested above. How that'd play with other blink implementations I'm not sure off the top of my head (particularly RGBLED.blink). Sorry, sleep required - I'll have a think about this later in the week!

lurch, 2016-02-15:
I guess one possible way of minimising API bloat, while neatly covering the "blink X times and then pause for Y seconds" use-case, might be to add a `repeat_after` parameter to the existing `blink` method(s)? e.g. you could do `blink(on_time=0.5, off_time=0.5, n=3, repeat_after=5)` and it would rapidly blink 3 times, then be off for 5 seconds, and then rapidly blink 3 times, etc. (and perhaps an exception should be thrown if `repeat_after` is used without `n` also being used)
This is obviously much less flexible than passing in a list of `(value, delay)` tuples, but it's a much 'easier' API for the end-user to use. (it's easy to imagine people using different numbers-of-blinks of the same LED to indicate different things (rather than using multiple LEDs))

bennuttall, 2016-09-19:
I have a solution for this using source/values based on #443.

closed 2018-09-20 by bennuttall