gpiozero/gpiozero #162: Add additional ADC chips

lurch, 2016-02-01:
A little "reminder" issue, as requested https://github.com/RPi-Distro/python-gpiozero/issues/41#issuecomment-177566699

As I mentioned at https://github.com/RPi-Distro/python-gpiozero/pull/156#issuecomment-177553333 it'll hopefully be fairly straightforward to also add classes for the MCP3001, MCP3002, MCP3201 and MCP3202 ADC chips too.

pcopa, 2016-02-01:
Yes this should be simple, just a matter of  adjusting the channel count. This would address all the spi ADC chips (that I know of).

lurch, 2016-02-07:
And looking at https://raw.githubusercontent.com/RPi-Distro/python-gpiozero/master/docs/images/composite_device_hierarchy.png it would seem to make sense for MCP3001, MCP3002, MCP3004 & MCP3008 to inherit from MCP30xx; and MCP3201, MCP3202, MCP3204 & MCP3208 to inherit from MCP32xx?
Or would that just be taking things too far for the sake of 'neatness / symmetry' ?
(Is this something that can be squeezed into 1.1 ?)

bennuttall, 2016-02-10:
- [ ] MCP3001
- [ ] MCP3002
- [x] MCP3004
- [x] MCP3008
- [ ] MCP3201
- [ ] MCP3202
- [x] MCP3204
- [x] MCP3208

bennuttall, 2016-02-11:
I'm adding these now. Got MCP3002 and MCP3202 in.

[MCP3001](http://www.farnell.com/datasheets/630400.pdf) and [MCP3201](http://www.farnell.com/datasheets/1669366.pdf) seem to have two input channels but +/- rather than CH0/CH1 - I'm not sure what that means. Is it that one to read an input as 0->max, the other max->0? (if that makes sense?)

bennuttall, 2016-02-11:
Reading up on datasheets it's definitely one channel, I just don't understand what the IN+ and IN- are. I'll add them as I have them, and they can be fixed before release if there's something else needed.

waveform80, 2016-02-11:
That means that like the 3301 they operate in differential mode only (and I'd guess they don't require any input in that case); i.e. you're expected to connect both. However, if the chip's like the other 32xx and 30xx's it won't support fully signed operation; i.e. IN- will only be for background noise compensation, not for measuring a fully negative value.

waveform80, 2016-02-11:
Have a look at the existing MCP3301 class and that should shed some light on things.

bennuttall, 2016-02-11:
Yeah, just spotted it! That makes sense.

I was going to leave the `channel` parameter in, even though it has to be `0`, just so the arguments were the same as the others, but I see you removed it from `MCP3301` so I will too.

waveform80, 2016-02-11:
Yeah, not much point having a parameter you can't do anything with :)

bennuttall, 2016-02-11:
Yeah

bennuttall, 2016-02-11:
Pushed - please take a look!

closed 2016-02-11 by bennuttall

waveform80, 2016-02-11:
I think you need to remove `differential=False` from the xx01 constructors; those chips will very likely operate in differential-only mode; i.e. firstly you shouldn't be able to specify differential and secondly its value should be True.

waveform80, 2016-02-11:
Yeah, just skimmed the data-sheets and they're both differential only. Also, because both of them require no input at all you'll need to push `MCP33xx._send` up into `MCP3xxx` and override it in the xx01 classes to just return `[0, 0]` (to allow two bytes worth to be received without sending anything).

bennuttall, 2016-02-11:
Like that?

waveform80, 2016-02-11:
Yup, that's the first bit - now you need to deal with the fact they don't expect any inbound data (and will only send back two-bytes worth when the SPI CLK activates)

commit 3f9ea915 (2016-02-11): Add additional ADC chips, close #162