gpiozero/gpiozero #164: Bug in TrafficLightsBuzzer ?

lurch, 2016-02-02:
Is https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/boards.py#L264 missing `self.button` ? 
The dochelp for the `all` method a few lines further down mentions that the Button should be returned...

On a side-note: is it worth making `self._all` and the `all()` property a standard part of the CompositeDevice class? (and then maybe `close()` and `closed` could be handled automatically in CompositeDevice too?)

lurch, 2016-02-06:
And as mentioned in https://github.com/RPi-Distro/python-gpiozero/pull/131#issuecomment-180522947 the TrafficLightsBuzzer `value` property ought to include the button's value too.

Although I dunno what 'fixing' this would mean in terms of "API breakage"? :-/

waveform80, 2016-04-01:
This was fixed by #239 which also implements a "proper" `CompositeDevice` class (yeah, it's technically an API break but the benefit of a decent `CompositeDevice` implementation is too important to pass up - and I'm hoping this doesn't break any existing scripts :)

closed 2016-04-01 by waveform80