gpiozero/gpiozero #167: Rename DigitalOutputDevice._blink_led ?

lurch, 2016-02-03:
At the moment https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L181 specifies a:
```python
def _blink_led(self, on_time, off_time, n):
```
but IMHO this is mis-named. The DigitalOutputDevice is a generic base class, so having one of its method names explicitly refer to 'led' is slightly misleading/confusing.

I guess as it's an 'internal' method (i.e. it has a leading underscore) there's no problem in renaming it, but I'm not sure what to rename it **to**. `_blink_func`? `_blink_loop`? `_do_blink`?

bennuttall, 2016-02-07:
I always thought that but it's internal so fixing it didn't take priority, and I didn't want to risk breaking it. Easy enough fix though, I guess.

waveform80, 2016-02-07:
I'll bash this off quickly enough

waveform80, 2016-02-07:
Gotta love vim's bufdo with word-anchored regexes :)

commit 19c19f6f (2016-02-07): Merge pull request #173 from waveform80/blink-device-not-led

Fix #167
commit f0ea1627 (2016-02-07): Fix #167
closed 2016-02-07 by waveform80