gpiozero/gpiozero #169: Add function to use a piezo speaker

CyberKiller40, 2016-02-04:
A piezo speaker without a generator needs to be provided a square signal in order to work. There is a complicated method to do it using wiringpi, setting a pin to pwm and doing something like this:

pi@pila:~ $ gpio -g mode 18 pwm
pi@pila:~ $ gpio -g pwm 18 200
(200 is the frequency (probably), can be varied to get different sounds, can be issued again during sound emitting to change the tone dynamically)
pi@pila:~ $ gpio pwm-ms #to turn on
pi@pila:~ $ gpio -g mode 18 in #to turn off

It would be cool if some simple functions to send various beeps would be available through gpiozero.

bennuttall, 2016-02-07:
I'm interested in making musical sounds (if possible with the hardware) like the micropython team did on their bbc microbit API. We discussed it a while ago (in [#75](https://github.com/RPi-Distro/python-gpiozero/issues/75#issuecomment-156806702)). If someone can get hold of a speaker that plays decent sounds via GPIO using PWM then let's get working on an API for it.

bennuttall, 2019-02-01:
Finally! This was just introduced as part of #681, and will be released shortly.

closed 2019-02-01 by bennuttall