gpiozero/gpiozero #175: LEDBoard should have an active_high option

lurch, 2016-02-08:
So that it can be used with both common-cathode and common-anode configurations of groups of LEDs.

lurch, 2016-02-10:
In fact I guess this should be added in LEDCollection.

closed 2016-02-11 by bennuttall

lurch, 2016-02-11:
Hmm, the way you've done it there is to assign the identical `initial_value` to every LED in the LEDBoard. The way I was imagining it working, was to have the (optional) `iniital_value` be a tuple, with an exception thrown if it's a different size to the `pins` tuple, and then using the initial_value tuple-values to initialise the individual LEDClass initial-values (maybe by using something like `enumerate` or `zip`).

bennuttall, 2016-02-11:
You could do, but that's a complex init for an odd case, I'd say. Probably easier just to set them up after init. What do you think, @waveform80?

lurch, 2016-02-11:
https://github.com/RPi-Distro/python-gpiozero/pull/176#issuecomment-181906439

commit 8482fd12 (2016-02-11): Add active_high and initial_value to LEDCollection, close #175