gpiozero/gpiozero #180: Add pigpiod pins support

waveform80, 2016-02-08:
We've now got RPi.GPIO, RPIO, and a rough pure-Python pins implementation. Ought to add [pigpiod](https://github.com/joan2937/pigpio) to that mix as it's got some cool hardware PWM and looks like it ought to work on Pi 2 (not to mention its architecture means it's had rootless scripts longer than anyone!).

lurch, 2016-02-09:
Are http://wiringpi.com/ and/or https://github.com/Gadgetoid/WiringPi2-Python / https://github.com/WiringPi/WiringPi2-Python worth investigating as possible "back-ends" too? Or would they bring nothing new to the party?

waveform80, 2016-02-09:
WiringPi2 is on my list - but I need to investigate a bit more before committing to it. I don't *think* it adds anything that we won't already get from some various bits of RPIO or pigpiod but there may be people that want to use gpiozero in combination with existing scripts that use WiringPi so that's reason enough to support it.

waveform80, 2016-02-12:
The more I play with pigpiod and dig into its implementation, the more I'm impressed by it. The hardware PWM is lovely (no spamming the console like RPIO, it Just Works (tm)), it's extremely responsive (I was a little worried that the socket architecture was going to mean that timing-sensitive components like our LightSensor class would be a bit iffy - but it works perfectly!), and it's got a ton of cool side features I've yet to explore (unfortunately we probably won't be taking advantage of these because unless it's common to all the libs there's not much point in using it).

Why it's not more popular, I don't know (actually that's a lie; I can take a guess: it was later to the party, it's not packaged properly for Raspbian, and it's not got an API compatible with the one that *did* get to the party first: RPi.GPIO, all of which probably counts against its popularity). Still, just thought I should leave a comment here lamenting that it *should* be more popular - and with a bit of luck including an interface to gpiozero might get it a little more attention.

Anyway, the pin implementation for it is basically done and working well - I'll merge this in a bit when I've got the docs side sorted.

closed 2016-02-12 by waveform80

lurch, 2016-02-12:
IIRC the last time I looked at the documentation for pigpiod, I found all of the advanced features it offers a bit overwhelming. So I was happy to stick to the much simpler (and more widely-used) RPi.GPIO - especially as I never needed any of pigpiod's ultra-precise timing features.

waveform80, 2016-02-12:
Yeah, API bloat really can put people off using things (just looking at the contents page and going "how many methods?!"). It's one of the reasons I'm so paranoid about it here; I really don't want to scare off new (esp. younger) users with a huge methods list.

lurch, 2016-02-12:
I think having a nice clean "Recipes" page up-front is a good antidote to that, IMHO.

waveform80, 2016-02-12:
Definitely; copy-paste-modify turns out to be a surprisingly successful programming strategy!

The other trick is to keep the recipes brutally short. I think it's not just long contents pages that put people off; if a script is more than a screen's worth in height, it puts people off actually reading and understanding it (I need to do some more work on picraft's recipes to bring them down to that ideal...). They might still copy-paste-run it but they're less likely to read it and start hacking on it themselves.

lukaszo, 2016-02-16:
Wow, I wanted to add pigpiod support myself but you were faster :D 
PWM is amazing. I wrote some code(using pigpiod) to drive a car and move pan/tilt. It works perfectly. 
I will rewrite this code to use gpiozero. I think it would be a good way to test  this issue :)

waveform80, 2016-02-17:
If you want to have a play with it feel free to clone the repo to play with the implementation (if you need instructions, just let me know - I can put some together); I'm not entirely sure when 1.2's going to be released - so all depends how patient you want to be :)

If you look at the "latest" branch of the docs you can see the merged implementation already there: http://gpiozero.readthedocs.org/en/latest/api_pins.html

commit 0978b4c4 (2016-02-12): Merge pull request #192 from waveform80/pigpiod

Fix #180 - Add support for pigpio
commit aebe4287 (2016-02-12): Fix #180 - Add support for pigpio