gpiozero/gpiozero #182: LEDBarGraph PWM support

waveform80, 2016-02-09:
As mentioned in comments in #176 there's several things to clean up:

* [ ] Add PWM support to LEDBarGraph
* [ ] Document LEDCollection in generic API docs (don't forget the hierarchy diagram)
* [ ] Add pot->graph recipe to Recipes chapter
* [ ] Fix the description of the first example in the docs

bennuttall, 2016-02-11:
What's your plan for PWM support? A value of 0.5 for 3 LEDs = `[1, 0.5, 0]`?

waveform80, 2016-02-11:
Yeah, that's basically the idea - every LED is either on (value > LED's threshold), off (value < prior LED's threshold) or on a PWM duty-cycle linearly proportional to how far value sits between the current and prior LED's thresholds.

bennuttall, 2016-02-11:
Cool. I might have a go.

waveform80, 2016-02-11:
Go for it :)

bennuttall, 2016-02-11:
Ok got distracted. I'm sure I can tackle this - not now though.

commit 8fc4b58e (2016-04-01): Fix #182

Fixes LEDBarGraph (broken by snowpi updates) and adds PWM support and
relevant recipes
commit af94fc2a (2016-04-01): Merge pull request #241 from waveform80/bar-graph-pwm

Fix #182
closed 2016-04-01 by waveform80