gpiozero/gpiozero #189: Cannot control individual LED while all LEDBoard LEDs are blinking?

bennuttall, 2016-02-11:
I just tried this:

```python
>>> leds = LEDBoard(22, 23, 24)
>>> leds.blink()
>>> leds.leds[0]
<gpiozero.LED object on pin GPIO22, active_high=True, is_active=False>
>>> leds.leds[0].on()
```

And all three LEDs kept blinking. I expected the first LED to stay on while the other two continued to blink.

At first I thought this was a blatant bug, but now I think about it, I imagine there's nothing to stop `LEDBoard`'s blink thread when you access the LED on its own.

Is this fixable?

bennuttall, 2016-02-11:
Most of the reason I expect to be able to do this is because I do it all the time with the likes of `TrafficLights` and `TrafficHat` - but that's different as I'm still accessing the LED via the collection interface, i.e. `lights.blink(); lights.green.on()`

waveform80, 2016-02-11:
Not easily - basically you've started a blink thread which assumes it's got free reign over all the LEDs of the board. You've then tweaked the state of one of them but on the next cycle the blink thread will blithely overwrite that state. To be honest I'm very surprised that works on TrafficLights ...

\<dave goes and looks at LEDBoard and TrafficLights\>

How the hell *does* that work with TrafficLights?! It shouldn't be...

waveform80, 2016-02-11:
Erm ... it doesn't work with TrafficLights for me. I can overwrite the state of, say, green but on the next blink cycle it'll get set by the blink thread again.

bennuttall, 2016-02-11:
Ermmmmm ok, it works for `TrafficHat` but not `TrafficLights`...

bennuttall, 2016-02-11:
Wait ... it did a minute ago.

waveform80, 2016-02-11:
Really? I'm not sure how that's possible looking at the code but I can't say I've tried it!

Anyway, can it be fixed to allow overrides of individual LEDs? Hmm ... not easily. There's a couple of ideas banging around in my head but they're a bit nasty: at the moment LEDs have no idea if they're part of a board - and this is a very good thing from the point of view of avoiding cyclic references - although it's possible to work around that sort of thing with weakrefs. We could introduce the notion of an "owning" composite device to output devices and have output devices notify any owning device in the event their state is set (the owning device could then use this to notify the background blinking thread that it should no longer control that particular output device).

So, yeah, I can fix it ... but you're not going to like the look of the code afterwards :)

bennuttall, 2016-02-11:
Ok this works for me:

```python
>>> th = TrafficHat()
>>> th.blink()  # blink all including buzzer
>>> th.lights.green.on()
```

At this point, the amber and red lights continue to blink (along with the buzzer) but the green light stays on.

waveform80, 2016-02-11:
Ahhh ... TrafficHat's still got a broken blink() implementation that just calls blink() on all its subordinate objects (which'll break if background=False).

waveform80, 2016-02-11:
And you'll also find if you leave it long enough the lights will drift out of sync as the various threads controlling them get uneven time-slices

bennuttall, 2016-02-11:
Ahhh ok. Phew. I'm not going mad. So that's a separate issue? *creates issue*

Also I think I noticed this had weird behaviour:

```python
>>> th.blink()
>>> th.lights.blink()
>>> th.stop()
```

The lights are still blinking!

I don't consider this a show-stopping bug and there are more important things to do for v1.2.

We can leave this for now (or forever) if you want.

waveform80, 2016-02-11:
Yeah, that's again due to there being two separate blink implementations (one on the composite TrafficHat, and one on the composite LEDBoard) and the fact subordinate devices have no idea they're part of a larger composite device (so it's possible to fire up multiple competing blink threads in this case). Basically that'd require something similar to the fix outlined above - doable, but nasty.

waveform80, 2016-02-11:
At the very least I ought to sort out TrafficLightsBuzzer.blink to work properly (hadn't realized that was still hanging around)

waveform80, 2016-02-11:
Heh - that was fast ;)

waveform80, 2016-02-11:
Oops - hit the wrong button, sorry! Did you want me to deal with override issue?

bennuttall, 2016-02-11:
Yes please

waveform80, 2016-02-11:
Okay - I'll assign this to myself and have a go at it when I've got a mo.

bennuttall, 2016-02-11:
2 issues closed, 2 opened tonight. What a world we live in.

lurch, 2016-02-12:
I've not started testing it yet, but I'd been toying around with having a more fully-featured `CompositeDevice` class, that automatically provided `close`, `closed`, `all`, `value` etc. methods by having a `CompositeDevice._add_device` method that subclasses should use during `__init__`. I guess that'd also be a great place to hook in the 'owning' stuff that @waveform80 is talking about above, but I won't have any more time to experiment with this until mid next week.
(By providing an optional 'alias-name' to `CompositeDevice._add_device` it also automatically provided named-property accessors - I even started experimenting with automatically-created `namedtuple`s! But this was all in theory as I was writing it when I didn't have access to a Pi, so I've no idea if it would _actually_ work in practice).

Hmmm, and if we're expecting to be able to blink-all-leds apart from those that we've explicitly set a state for (and leave the "un-set" ones blinking) does that imply that the `LEDBoard.blink` method ought to grow an extra parameter to tell it which LEDs *not* to blink?
TBH I'm not sure if blinking-all-leds-except-those-we've-explicitly-set-a-state-for makes sense, but maybe that's just because I'm too tired.
Perhaps another way of looking at this should be to say that if an LEDBoard (or some other CompositeDevice) is `blink`ing, then it shouldn't be possible to set the value of the indivudual LEDs (components)?  Again, I'm too tired to think that through fully.

waveform80, 2016-02-12:
Oooh - now *that's* an interesting idea! I'll be interested to see that (and yes, creating the namedtuple automatically and then serving it up as a property on the class - entirely possible - brilliant idea).

On telling it what LEDs not to blink, my thinking went something along these lines:

* When blink starts it'd populate a list of the devices to blink (e.g. at the moment, if you look at the blink implementations all of them somewhere have `for led in self.leds` or something similar; so that'd become something like `for led in self._blinking_devices`). Initially the list could be "all devices owned by the composite" or possibly we could add a parameter to include/exclude devices.
* Device's constructor would grow an optional "parent" parameter so that devices would know what composite device owned them (this would have to be kept as a weakref proxy or something similar to prevent reference cycles).
* Finally, in OutputDevice's on/off/toggle methods, if `parent` is not `None` remove `self` from `parent._blinking_devices` prior to setting the state of the device's pin. So at the next iteration of the loop the device is no longer affected.
* There's some additional things that need to be thought about: what happens to the blink thread if all devices are removed from `_blinking_devices` (obviously it should terminate), and is a mutex required to ensure serial access to `_blinking_devices` (not sure; would probably have to look at the finished code to determine if any races exist).

Anyway, that's the rough sketch. Haven't got time to work on it right now and frankly it's probably worth waiting to see your new CompositeDevice implementation as I think it'd tie in rather nicely with that!

bennuttall, 2016-02-13:
I've been thinking about this. Would something like this work:

Every blinkable component has a `_blink_thread` property. When something blinks a component, it stores a reference to the blink thread within that property. Then, when the component state is modified, e.g. `component.on()` it kills its own blink thread.

```python
>>> th = TrafficHat()
>>> th.lights.blink()  # th.lights.green._blink_thread now set
>>> th.lights.green.on()  # th.lights.green._blink_thread now killed, th.lights.red._blink_thread still active
```

Does that make sense?

Similarly, would it be beneficial for blinkable components to maintain an `is_blinking` boolean property like `is_lit`, so that blinking can be determined?

waveform80, 2016-02-13:
I'm afraid that's not quite good enough. Consider that when you blink multiple LEDs (as in LEDBoard.blink) a *single* thread is started to blink all the LEDs as one. If we used multiple threads they'd gradually drift out of sync as not all threads get exactly the same length time-slice. So anyway, let's say that single thread goes into the subordinate object's `_blink_thread`; when that component's state is then modified it winds up killing the thread controlling *all* the LED's blinking.

So, the issue is we need a single thread to maintain synchronization but we need to modify what that thread is doing (which objects it's affecting) whilst it's running. That's why some form of shared state (like a list of objects) is required between the blinking thread and the main thread.

bennuttall, 2016-02-13:
Gah - nevermind

bennuttall, 2016-02-13:
What about `is_blinking`?

waveform80, 2016-02-13:
The `is_blinking` property ... hmm ... yes, that's doable I think. I'm just having a play with Energenie at the moment then I'll have a go at this.

bennuttall, 2016-02-13:
Oh awesome - look forward to it

waveform80, 2016-04-01:
Okay, just looking into re-doing SnowPi using the new LEDBoard stuff (based on the new CompositeOutputDevice stuff) and it's going to require doing something like this anyway (in order to enable things like snowpi.blink, snowpi.arms.blink, snowpi.arms.left.blink, etc.) so I'll target this for 1.2 as well and try and this done today.

commit d1669b80 (2016-04-01): Merge pull request #240 from waveform80/snowpi

Fix #130, fix #131, fix #189
commit d6af0293 (2016-04-01): Fix #130, fix #131, fix #189

This PR implements SnowPi, adds the ability for LEDBoard's to own other
LEDBoards as well as LEDs, and enhances blink so that manually
controlling a LED automatically stops it from blinking (no matter
whether it's blinking itself or a LEDBoard is blinking it).

It also fixes up RGBLED and Motor which I managed to break with the last
PR ...
closed 2016-04-01 by waveform80