gpiozero/gpiozero #191: Add recipe for GPIO Zero Travis build LED indicator

bennuttall, 2016-02-11:
Has to be done. Green light means tests passing, red light means the build is broken.

lurch, 2016-02-11:
... yellow (amber) light for build currently in progress? :)

waveform80, 2016-02-12:
Pre-requisites:

```
$ sudo apt-get install python-gpiozero python3-gpiozero
$ sudo pip install travispy
```

Script:

```python
from travispy import TravisPy
from gpiozero import LED
from time import sleep
from signal import pause

def build_state():
    t = TravisPy()
    r = t.repo('RPi-Distro/python-gpiozero')
    while True:
        yield r.last_build_state == 'passed'
        sleep(3600) # Sleep an hour before hitting travis again

def invert(values):
    for value in values:
        yield not value

red = LED(12)
green = LED(16)

red.source = invert(green.values)
green.source = build_state()
pause()
```

Bonus points for anyone that adds an amber light for build in progress, or even better has the red/green LEDs blink alternately as if it's Schrödinger's build and we're waiting for the waveform to collapse!

P.S. I haven't actually tested this ... apologies if it's completely broken :)

lurch, 2016-02-12:
Perhaps it's just me, but `build_state()` sounds like a verb rather than a noun (IYSWIM). And I'd be tempted to add the sleep-time as an optional `build_state()` parameter rather than have it fixed to an hour.

> the red/green LEDs blink alternately as if it's Schrödinger's build and we're waiting for the waveform to collapse!

I hope we never see the waveform80 collapse! ;-D

waveform80, 2016-02-12:
Hmm, good points :)

```python
from travispy import TravisPy
from gpiozero import LED
from time import sleep
from signal import pause

def build_passed(repo='RPi-Distro/python-gpiozero', delay=3600):
    t = TravisPy()
    r = t.repo(repo)
    while True:
        yield r.last_build_state == 'passed'
        sleep(delay) # Sleep an hour before hitting travis again

def invert(values):
    for value in values:
        yield not value

red = LED(12)
green = LED(16)

red.source = invert(green.values)
green.source = build_passed()
pause()
```

bennuttall, 2016-02-12:
There's a `travispy` module? Ace!

waveform80, 2016-02-12:
Yeah, it's not packaged for Debian/Raspbian but at least it's on pip. Seems like it's basically a Python port of the official Ruby library. The nicest thing is that while the docs mention GitHub authentication, it turns out that's not required for basic stuff like reading the build state of a public project (hence why there's no need to mess around with GitHub tokens in my example above).

waveform80, 2016-02-12:
FYI, the docs for it are here: http://travispy.readthedocs.org/en/stable/

bennuttall, 2016-02-12:
Yeah - just tried this with a `TrafficHat`. Very nice!

I've always wanted to build one to go on the office wall - for rpi documentation build status and "is the website down". Now I shall!

waveform80, 2016-02-12:
Very nice! I may have to dedicate a Pi to monitoring the test status of my various projects ... not sure how I'd multiplex them yet ... maybe a Display-O-Tron HAT or something ...

bennuttall, 2016-02-12:
I was thinking big block of wood with dymo labels, not dissimilar to [Rachel's BuildYourOwn exhibition](http://piweekly.net/wp-content/uploads/2015/06/IMG_1320-resized.jpg).

bennuttall, 2016-02-12:
Also not dissimilar to my [People in Space Indicator](https://www.raspberrypi.org/learning/people-in-space-indicator/).

lurch, 2016-02-12:
> not sure how I'd multiplex them yet ... 

28 GPIOs on the B+... do you have more than 14 projects you need to display the status of? ;)

lurch, 2016-02-12:
> for rpi documentation build status

@bennuttall I forgot to mention this to you before, but: https://github.com/gjtorikian/html-proofer/issues/299

waveform80, 2016-02-13:
@lurch No, but I'm not sure I've got that many LEDs lying around spare (or enough jumper wires to wire 'em with :)

bennuttall, 2016-02-13:
Ha. Just seen this: http://antonfisher.com/posts/2016/01/05/make-raspberry-pi-jenkins-traffic-light/

lurch, 2016-04-11:
I guess this is a great candidate for another http://gpiozero.readthedocs.org/en/v1.2.0/api_other.html#internal-devices :-)

waveform80, 2016-04-12:
Hmm, that's a nice idea

carlosperate, 2016-04-13:
Not really using GPIO Zero, but I have this CI/build server running on a pi hooked to a github repository: https://twitter.com/tinypici.
Is there a way to a leave an LED ON after a Python script finishes? That's the only reason I've been running bash scripts for the traffic light control.

bennuttall, 2016-04-13:
Use `signal.pause()` at the end of the script to keep the program alive, as per recommendations in the docs.

carlosperate, 2016-04-13:
Thanks! I'll port the script to Python then :)

waveform80, 2016-04-13:
I think he wants to leave the LED lit after the script terminates, not just leave the script running in `pause`. That can be done but currently GPIO Zero is too "nice" in that it always attempts to clean up all resources (which includes resetting all GPIOs to input, which will extinguish the LED). To be honest, this is actually quite hard with GPIO Zero at the moment; probably much easier to just use RPi.GPIO and don't call cleanup at the script's end

carlosperate, 2016-04-13:
> I think he wants to leave the LED lit after the script terminates, not just leave the script running in pause. 

Indeed that's what I meant, I guess I will stick with the bash files for now.

Out of curiosity, if I was to throw an uncaught exception to "exit", would the cleanup be bypassed and leave the LEDs ON?

waveform80, 2016-04-15:
Hmmm, if you raise an uncaught exception, all cleanup should still occur. However, I am vaguely reminded that there's some way to do a "dirty" exit of the CPython environment - it's something like `os.exit` if memory serves.

Ahh, after a brief search: [`os._exit`](https://docs.python.org/3.4/library/os.html#os._exit) - this *might* work to leave LEDs on but frankly it's an awfully dirty way of accomplishing this and not something we'd be keen to support. So, if it doesn't work or blows stuff up I'd likely say "not our problem". I'd stick to bash for now unless/until we come up with a more refined method.

On whether we should come up with a more refined method: I'm not entirely convinced this is a great use-case. I'd rather have one persistent Python script that monitored the output of other scripts and set the LED accordingly (essentially one that "owned" the LED), instead of relying on scripts to explicitly "not clean up" (i.e. scripts that temporarily own the LED and then die, hoping the system will maintain the state they've set, which raises all sorts of questions about parallel runs and so forth).

carlosperate, 2016-04-15:
hehe, "awfully dirty" still feels like an understatement, the question originated from my curiosity rather than as a real consideration (also because I've been chasing a bug somewhere else cause by improper clean up even during proper exception handling).

I do agree this use-case would not be something widely used. But for something like this (the Pi only has one function, and sets the LEDs once in a very long while) having interprocess communication seems a bit overkill, and because the scripts get executed and terminated automatically by a java process (jenkins), there is no real possibility of spawning and tracking processes individually from a "master python script".

waveform80, 2016-04-15:
Is there no possibility of even writing the exit code of the process to a file (I'm not familiar with jenkins)? e.g. in bash `{ my_script; echo $? > my_script.out; }`. That way you could simply do something like:

```python
import io
from time import sleep
from signal import pause
from gpiozero import TrafficLights

def build_status(filename='my_script.out'):
    while True:
        with io.open(filename, 'r') as f:
            s = f.read().strip()
        yield {
            '':  (0, 1, 0), # amber (building)
            '0': (0, 0, 1), # green (success)
            }.get(s, (1, 0, 0)) # red (anything else)
        sleep(1)

tl = TrafficLights(13, 19, 26)
tl.source = build_status()
pause()
```

Because the file-system can be a (crude) form of IPC too.

carlosperate, 2016-04-15:
Yes, that's really the only easy way, but I'm not a big fan of using the file system to communicate between processes. Between that and the bash script I prefer to use the script even if it leaves the pin configuration as outputs on exit. But I do realise that's a very personal opinion, for all intents and purposes using the file system is a well establish method.

lurch, 2016-04-18:
> Because the file-system can be a (crude) form of IPC too.

Perhaps FileExists and FileContentEquals could be other InternalDevices too? ;)

closed 2016-05-26 by bennuttall

waveform80, 2016-05-26:
Hate to point it out, but we've actually got [inverted](http://gpiozero.readthedocs.io/en/v1.2.0/api_tools.html#gpiozero.tools.inverted) (and negated below it) in source tools now :)

commit 0a5499df (2016-05-26): Add recipe for GPIO Zero Travis build LED indicator, close #191