gpiozero/gpiozero #193: Fails to re-create TrafficHat

bennuttall, 2016-02-12:
I've found an odd case:

```python
>>> from gpiozero import TrafficHat
>>> th = TrafficHat()
>>> th.close()
>>> th = TrafficHat()
```

which raises:

```
...
/home/pi/python-gpiozero/gpiozero/pins/rpigpio.py in _set_when_changed(self, value)
    197                 self._number, self._edges,
    198                 callback=lambda channel: self._when_changed(),
--> 199                 bouncetime=self._bounce)
    200         elif self._when_changed is not None and value is None:
    201             GPIO.remove_event_detect(self._number)

RuntimeError: You must setup() the GPIO channel as an input first
```

I've tried with `TrafficLights` but don't get the same error.

waveform80, 2016-02-12:
Hmmm ... interesting. I'm not sure what's going on there, which probably means it's something nasty to do with object lifetimes.

lurch, 2016-02-12:
Quick! Add it to the unit-tests so that you can see your Travis-indicator LEDs turn red! ;-)

waveform80, 2016-02-13:
Hehe - well, I'll definitely add it to the tests when I get to it, but I haven't even started on boards yet!

lurch, 2016-02-17:
I tried digging into this a bit further, and the problem can actually be re-created with just:
```python
>>> from gpiozero import Button
>>> b = Button(9)
>>> b.close()
>>> b = Button(9)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "gpiozero/devices.py", line 92, in __call__
    result = super(GPIOMeta, mcls).__call__(*args, **kwargs)
  File "gpiozero/input_devices.py", line 396, in __init__
    super(Button, self).__init__(pin, pull_up, bounce_time)
  File "gpiozero/input_devices.py", line 228, in __init__
    self.pin.when_changed = self._fire_events
  File "gpiozero/pins/__init__.py", line 232, in <lambda>
    lambda self, value: self._set_when_changed(value),
  File "gpiozero/pins/rpigpio.py", line 213, in _set_when_changed
    bouncetime=self._bounce)
RuntimeError: You must setup() the GPIO channel as an input first
```

i.e. it looks like something inside `Button` isn't getting properly cleaned up or something?

waveform80, 2016-04-03:
### Background

Found it. First, let's cover some basics and then dig into what's going on in a test script:

1. All GPIO pins have a "function" (input, output, and "alternates" in some cases). You can read a GPIO pin's function out of the GPIO registers; this is what RPi.GPIO's [`gpio_function`](https://github.com/waveform80/rpi-gpio/blob/master/source/py_gpio.c#L865) call does.
2. However, RPi.GPIO also keeps an internal variable called [`gpio_direction`](https://github.com/waveform80/rpi-gpio/blob/master/source/common.h#L38) which tracks what RPi.GPIO last set a GPIO's function to (input or output only as you're not allowed to set "alternate" functions with RPi.GPIO). RPi.GPIO [initializes this to -1](https://github.com/waveform80/rpi-gpio/blob/master/source/py_gpio.c#L1002) which is interpreted as "unknown", obviously `setup` [sets this to whatever direction the user specifies](https://github.com/waveform80/rpi-gpio/blob/master/source/py_gpio.c#L226), and notably `cleanup` [resets this back to -1](https://github.com/waveform80/rpi-gpio/blob/master/source/py_gpio.c#L86) (while also setting the GPIO to "input").
3. Note that a GPIO is never actually in "unknown" state - it'll be an input, output, or one of its alternate functions - RPi.GPIO simply considers it unknown until you call `setup` on it. Despite this, `gpio_function` will still return the *actual* function of the GPIO.
4. Next, `add_event_detect` [checks that a GPIO is set to input](https://github.com/waveform80/rpi-gpio/blob/master/source/py_gpio.c#L698) by looking at `gpio_direction` (the internal variable) rather than calling `gpio_function` (which queries the actual GPIO function).
5. The default pin implementation is RPi.GPIO so I'll stick to this case in the following. However, it's worth bearing in mind that RPIO also duplicates the GPIO function state in an internal variable (and thus probably has the same issue though I haven't tested it); pigpiod and the native pin implementation don't duplicate state so they won't suffer from this issue (see "duplicating state").

### Walkthrough

So, now we've got all the information, let's walk through what's going on with the following test script:

```python
from gpiozero import Button

b = Button(9)
b.close()
b = Button(9)
```

1. During `Button` construction, we first [convert the integer 9 to a pin implementation for GPIO9](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/devices.py#L443).
2. For the `RPiGPIOPin` implementation, when the pin is first constructed we [force it to an "input"](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L81).
3. Still during construction up at the `InputDevice` level, we check what the GPIO pin's function currently is and [set it to "input"](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/input_devices.py#L42) if it's not already.
4. Finally during construction, `DigitalInputDevice` sets up a [callback to watch for state changes](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/input_devices.py#L229). Under the covers, this [calls `add_event_detect`](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L213) which checks `gpio_direction` to make sure the GPIO is in "input" mode.
5. At this point, GPIO9 is in input mode, and RPi.GPIO's `gpio_direction` variable also records that GPIO9 is in input mode so everything works.
6. Now we close the Button. As part of this we [close the pin the button is using](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/devices.py#L465).
7. Down in RPiGPIOPin this [calls `cleanup` on the pin](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L101).
8. At this point, GPIO9 is still in input mode (remember that cleanup sets GPIOs to input) but RPi.GPIO's `gpio_direction` variable now records GPIO9 as "unknown" direction.
9. Now we're recreating the `Button`. As before, we convert 9 to an RPiGPIOPin implementation. However, pins are never *really* destroyed in gpiozero; they get cached and repeated requests to construct a particular pin object return the cached objects (see "pin caching" below). So the pin isn't forced to "input" here, we just [return the existing pin](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L78).
10. Now InputDevice checks what the GPIO pin's function is (as before) to make sure it's "input". As before, [`gpio_function` is called to determine what the current function is](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/rpigpio.py#L119). As this reads the actual GPIO registers it reports that the GPIO is in "input" mode, because it is.
11. Now InputDevice goes to add an event callback but this time remember that `gpio_direction` records GPIO9 as "unknown" so the call fails.

### Pin caching
The reason for caching pin objects is that some pin attributes cannot be read from hardware (pull-up notably) so this is a case where having a local copy of this information is useful. Further, it follows that two callers constructing an object representing a single pin ought to get the same object (it's one physical pin after all) which ensures that even if they set conflicting pull-ups they'll both "see" the actual state of the object (race conditions permitting ... hmm, just realized there ought to be a lock around that stuff in the native pin implementation).

### Duplicating state
[... cut ...]  should raise `AttributeError` (or more likely `ValueError` given it's not really a read-only attribute in this scenario). In this case, device implementations could simply set the function to "input" *without checking first* which would in turn ensure the internal state of RPi.GPIO (and RPIO) matched the reality. It slightly complicates future pin implementations (as it mandates `function` is always a read-write attribute), but I don't think it's a significant burden.

Phew! Back to the code...

closed 2016-04-03 by waveform80

lurch, 2016-04-03:
> However, given the pin API isn't hammered into stone yet (we've explicitly marked it "unstable") we could instead say: any attempt to set the pin's function to an unsupported value (like "output" on a strictly input-only pin) should raise `PinInvalidFunction`

Sounds good to me :+1: 

Going back to @bennuttall 's original comment, I guess `TrafficLights` allowed 're-construction' after closure (unlike `TrafficHat`), because all the sub-devices need to be in 'output' mode, and so their `function` always explicitly got set away from the default 'input' mode?

waveform80, 2016-04-03:
Oddly enough, while I'd coded InputDevice to query pin function before setting it, I hadn't done the same in OutputDevice (I should have though!) hence why TrafficLights (which was composed entirely of output devices) worked, but TrafficHat (which includes a button input device) didn't. Anyway, moot point now as neither InputDevice nor OutputDevice check pin function any more - they just set what they want.

commit 5646dd16 (2016-04-03): Merge pull request #245 from waveform80/recreate-traffic-hat

Fix #193
commit 7826e60c (2016-04-03): Fix #193

Changed pin.function so that it's always read-write, which in turn
permits InputDevice to force pin.function to "input" rather than
checking that it's not "input" first. This ensures internal state in
RPi.GPIO and RPIO reflects the reality of each pin's function (see
discussion under the ticket for more detail).