gpiozero/gpiozero #197: Offline docs how?

bennuttall, 2016-02-13:
I noticed there's now a `python-gpiozero-docs` package. First of all, do I need to ask for this to be installed by default?

I managed to browse the docs with:

```bash
epiphany-browser /usr/share/doc/python-gpiozero-doc/html/index.html &
```

But is there a better way? Perhaps we should document it. (ooh... meta)

waveform80, 2016-02-19:
I generally wouldn't install the docs package by default (largely because docs packages tend to grow to be quite large - usually several times the size of the library itself - see python-picamera-docs for example: 1.4Mb vs 100Kb for the library). Some people need offline docs, some don't so I'd just leave it as an option for those that want it (I could add it to the "Suggests" list in the main package though so it gets suggested by apt-get when installed). There's also other means of getting offline docs like PDF and EPub downloads from RTD which people might prefer too.

As for better ways of accessing the offline docs: sure. The docs integrate with [Debian's "docbase"](https://wiki.debian.org/doc-base) system for which there are several interfaces. dwww and dhelp are probably the most use on a Pi but aren't installed by default (for good reason: they do background indexing). So, for people that need offline docs (and want searching, and don't mind a small performance hit for the convenience of searching), they're a good thing to install. For people that're happy with online docs, they're not worth it.

Might be worth mentioning this in the docs? I get the impression not many people know about doc-base/dwww/dhelp/etc.

bennuttall, 2016-04-05:
Revisiting this, what's the recommended way to open the docs once you have the docs package installed?

waveform80, 2016-04-05:
Traditionally docs packages install their contents under `/usr/share/doc/<package-name>/`. There's no universal standards around docs packages; some use plain old text files, some use HTML, some even use PDFs so generally the method of using docs packages is to poke around /usr/share/doc/<package-name>/, see what's there and load it accordingly.

As mentioned above, there are *some* standards like Debian's docbase - far from universal but it's  fairly widespread nonetheless. This mandates docs in HTML format (because it's essential browser based) and it really depends what interface you want to install as to how you use it. If I recall correctly either dwww or dhelp installs a command which allows you to run something like "dwww <package-name>" and it fires up a browser with the results, or you can just run "dwww" and I think it presents an index page of all the packages it knows about.

So, for the docs package people could people familiar with docs packages could simply poke around /usr/share/doc/python-gpiozero-docs/html and load the "index.html" under there (which is the docs root, unsurprisingly), or if they prefer a friendlier interface they could install dwww/dhelp and use those instead. Or they could just download the PDF/Epub from RTD. In essence, it depends on the user's familiarity with docs packages and their reading preferences as to what they're going to pick, but I'd say we've got just about everyone covered.

bennuttall, 2016-04-05:
Hmm. Neither dwww nor dhelp worked for me. I installed them both, along with gpiozero docs. dwww did open up a browser but the search failed for me, and dhelp did nothing. Can you give it a try and see if we can provide a set of steps in the docs? I think it could be handy as an FYI.

lurch, 2016-04-05:
I guess whatever approach ends up getting adopted here, should also make its way across to picamera and picraft too? ;)

waveform80, 2016-04-05:
@lurch The approach here is actually derived from the one I took on picamera (i.e. the Makefile we use to generate the docs and the dpkg config is all derived from picamera's) - can't recall what I did with picraft but it's probably another picamera derivative docs-wise. If we bung in some new instructions though, it'd definitely be worth porting those over for the (admittedly few) people that want offline docs.

waveform80, 2016-08-15:
> Hmm. Neither dwww nor dhelp worked for me. I installed them both, along with gpiozero docs. dwww did open up a browser but the search failed for me, and dhelp did nothing. Can you give it a try and see if we can provide a set of steps in the docs? I think it could be handy as an FYI.

Just tried dhelp - works fine for me on Raspbian. Just running "dhelp" alone fired up a browser with the documentation tree on the left. I located gpio-zero's docs under Programming / Python as expected. It's worth noting dhelp's search function doesn't operate unless it's installed with Apache, and there's some minor configuration to do to get that working (uncomment some bits of `/etc/apache2/conf-available/dhelp.conf` and run `a2enconf dhelp`) - this is briefly mentioned in dhelp's man page.

I'll give dwww a try tomorrow - it may be that it too requires server config for search.

Anyway, I'll bung together an "offline" docs section to add to the docs - do you think it'd go under "Notes" or should it be its own section?

waveform80, 2016-08-15:
Dammit, should go to bed. Notes to self:

* Tried dwww; it works (actually rather better than dhelp) but like dhelp also requires some post-install config to work properly.
* It pulls in apache2 automatically (which dhelp doesn't, but should) but assumes Apache is configured for CGI out of the box (which it isn't). Run `sudo a2enmod cgi` and `sudo service apache2 restart` and `dwww` should then work much more nicely (find GPIO Zero docs from: Debian Documentation Menu / Programming / Python)
* However, search requires the index to be built; this takes a *long* time so the package install doesn't do it. A weekly cronjob is added to build it, but if you want to kick it off manually: `sudo dwww-index++` (then go make coffee...)

bennuttall, 2021-03-12:
Resolved in #931

PR #931 (2021-03-11): Unicode docs
PR #930 introduces some unicode chars into the docs (for fancy representations of the 7-segment displays). That causes some issues with the PDF generation given that LaTeX really doesn't like Unicode. Now that ReadTheDocs supports XeTeX for PDF rendering it's probably about time we switched to using this which solves most of the Unicode issues (or most of *our* Unicode issues anyway) with PDF generation.

Note that this PR builds on #930 - it shouldn't be considered until that one is dealt with.

Also closes #197
closed 2021-03-12 by bennuttall