gpiozero/gpiozero #200: Can't directly-control an OutputDevice whos source has been set

lurch, 2016-02-17:
While thinking about #189 I came up with the following example:
```python
from gpiozero import LED
from signal import pause
red = LED(21)
green = LED(20)
red.source = green.values
red.blink()
pause()
```

You *might* expect the code above to blink the red LED, but it doesn't.
I guess there's two ways of looking at this: a) this is 'undefined behaviour' and you shouldn't try directly-setting an LED that has its source property set (in which case perhaps a warning or exception should be thrown) or b) directly-setting an LED's value should 'nullify' the source property.
I guess perhaps b) follows the 'principle of least surprise' (maybe?) and 'cancelling' the `_source_thread` when a value is directly set, would be very similar to the way that the `_blink_thread` is currently cancelled when a value is directly set. But I'm not sure how/if this would work in terms of Mixins.

Also, should (nonsensical) code like that below throw some kind of exception?
```python
from gpiozero import LED
red = LED(21)
green = LED(20)
red.source = green.values
green.source = red.values
```
(attempting to do either `red.close()` or `green.close()` after this throws a `AttributeError: 'NoneType' object has no attribute 'state'`)

lurch, 2016-06-03:
I've just received my Raspio Experimenters Pack, and playing with that gave me another example that illustrates the issue being discussed here.
```python
>>> from gpiozero import *
>>> buzz = Buzzer(13)
>>> butt = Button(24)
>>> buzz.source = butt.values
```
Holding down the button then makes the buzzer beep (as expected). However if I then do
```python
>>> buzz.on()
```
there's a very quick 'click' from the buzzer as it turns on and then off again.

And in fact:
```python
>>> buzz.value
False
>>> buzz.on()
>>> buzz.value
False
```
(and pressing the button still makes the buzzer beep).

IMHO the preferable behaviour would be for `buzz.on()` to 'disconnect' the `buzz.source` (so that the buzzer is no longer controlled by the button).
But as I outlined in my original comment, another equally-valid approach would be to throw an exception if you try to call `on()` or `off()` on a component with a non-None `source`.

I think the current behaviour is simply too confusing / inconsistent.

waveform80, 2016-08-14:
Yup, fair point. Not sure which is better though - I'm swaying toward the exception approach simply because there's no "hidden" behaviour in that case (i.e. I can see the case being made that explicitly setting the value then causes surprise because the device no longer reacts to its explicitly set source, because it's no longer set).

bennuttall, 2017-09-16:
Revisiting this issue after a long time I don't consider this a bug. I think if people are going to set source and then want to do something else, they should explicitly disable source. Are you two happy for me to close this and #449?

closed 2018-02-15 by bennuttall