gpiozero/gpiozero #204: Should the 'duplicate pin checking' be more robust?

lurch, 2016-02-18:
With the variety of "pin backends" now in GpioZero (see http://gpiozero.readthedocs.org/en/latest/api_pins.html ), I wonder if the checking-for-duplicate-pins inside GPIODevice could/should be stricter?
Consider the following:
```python
>>> from gpiozero import LED
>>> from gpiozero.pins.rpigpio import RPiGPIOPin
>>> from gpiozero.pins.native import NativePin
>>> led1 = LED(21)
>>> led2 = LED(RPiGPIOPin(21))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "gpiozero/devices.py", line 92, in __call__
    result = super(GPIOMeta, mcls).__call__(*args, **kwargs)
  File "gpiozero/output_devices.py", line 104, in __init__
    super(DigitalOutputDevice, self).__init__(pin, active_high, initial_value)
  File "gpiozero/output_devices.py", line 42, in __init__
    super(OutputDevice, self).__init__(pin)
  File "gpiozero/devices.py", line 167, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "gpiozero/devices.py", line 236, in __init__
    'pin %r is already in use by another gpiozero object' % pin
gpiozero.exc.GPIOPinInUse: pin GPIO21 is already in use by another gpiozero object
>>> led3 = LED(NativePin(21))
>>> led1.pin
GPIO21
>>> led3.pin
GPIO21
>>> led1.pin == led3.pin
False
```

It's good that `LED(21)` and `LED(RPiGPIOPin(21))` are recognised as referring to the same pin, but I wonder if it should also recognise that `LED(NativePin(21))` is pointing to the same pin?

Given that the `repr()` of `RPiGPIOPin(21)` is the same as the repr of `NativePin(21)` I guess one possible solution would be to change the `_PINS` in `gpiozero/devices.py` to be a `dict` (indexed by the pin repr-strings) instead of a `set` ?

lurch, 2016-02-19:
...although with the "pull-up state of a pin being a "write only" property" as @waveform80 [mentions](https://github.com/RPi-Distro/python-gpiozero/issues/206#issuecomment-186177463), I wonder if we'd need something more sophisticated, so that the "hardware state" is remebered / shared across all pin implementations??  Alternatively we could just class it as "unsupported / undefined behaviour"...

waveform80, 2016-02-19:
Hmmm ... my general thought at the moment is: "if you go around mixing pin implementations for the same set of pins, you get to deal with any horrific consequences" on the basis that I can't think of any decent use-cases for wanting to do so. However, I've stopped short of actually checking for it on the basis that I might just be lacking in imagination (someone might have a use case) and that I'm pretty sure no-one's going to mix pin implementations accidentally.

Still, I think it's worth a mention in the docs as "unsupported behaviour - you can try it, but you get to pick up the pieces if/when it breaks horribly".

commit 22388b8e (2016-04-06): Merge pull request #256 from waveform80/last-tests-for-12

Fix #204 and start readying the release
commit ee3bd6b5 (2016-04-06): Fix #204 and start readying the release

Also re-numbers energenie sockets 1-4 (as noted by @bennuttall in
comments to #239), and adds several "real pins" tests and board tests.
The bad-PWM stuff is currently disabled as it causes segfaults when
running the tests and I can't seem to trace the cause at the moment.
Finally, I've tweaked the deb config to suggest gpiozero, removed spidev
as a mandatory dep (which'll fix installs on wheezy for py3), and
there's some more miscellaneous last-minute stuff here that I can't
recall...
closed 2016-04-06 by bennuttall