gpiozero/gpiozero #208: 'pin' parameter documentation out-of-date?

lurch, 2016-02-19:
In all the current gpiozero documentation (i.e. over many many classes) the `pin` parameter is always referred to as a (`int`). However with the pin-refactoring all these `pin` parameters can now be integers *or* pin objects.

Should the docs be updated, or would that just be too confusing for end-users?

Also, http://gpiozero.readthedocs.org/en/latest/api_generic.html#gpiozero.PWMOutputDevice says "See Notes for valid pin numbers." - a) the pin numbers are now on the Recipes page rather than the notes page b) it seems a bit weird that `PWMOutputDevice` is the _only_ place where this is mentioned (perhaps it should be removed for consistency)
....
Oh, nope, `PWMOutputDevice` is the only class on http://gpiozero.readthedocs.org/en/latest/api_generic.html that mentions it, but many classes on http://gpiozero.readthedocs.org/en/latest/api_input.html and http://gpiozero.readthedocs.org/en/latest/api_output.html also mention it. But they all incorrectly refer to the Notes page. I guess there's no way for all the 'pin' parameters to automatically have indentical docstrings? ;-)

And the Parameter-lists for http://gpiozero.readthedocs.org/en/latest/api_generic.html#gpiozero.WaitableInputDevice http://gpiozero.readthedocs.org/en/latest/api_generic.html#gpiozero.DigitalInputDevice http://gpiozero.readthedocs.org/en/latest/api_generic.html#gpiozero.SmoothedInputDevice http://gpiozero.readthedocs.org/en/latest/api_generic.html#gpiozero.AnalogInputDevice http://gpiozero.readthedocs.org/en/latest/api_generic.html#gpiozero.DigitalOutputDevice appear to be missing and/or wrong. (as are all of http://gpiozero.readthedocs.org/en/latest/api_input.html#analog-to-digital-converters-adc )

waveform80, 2016-02-19:
I'm a bit torn on this. Pins are a sufficiently advanced subject that for now, I'd like to keep at least the concrete stuff like LED, Button, etc. just mentioning integer numbers for pins (no need to scare anyone away with things like abstract classes!). I could be tempted to change the "generic" section to mentioning integers and Pin implementations, and obviously in the later pins chapter this is spelled out explicitly, but I expect only advanced users will get to that point and I'd like to keep the start of the docs as simple as possible (i.e. we should never mention this stuff in recipes unless we get around to splitting recipes into "basic" and "advanced" which we might do if some more complex recipes start cropping up).

Good point about notes being incorrectly referenced. reST does have an `.. include::` command for including the content of one file within another - unfortunately I don't think you can use that *within* a `:param:`. There's also `.. literalinclude::` for including code sections. At some point I may pull all the recipe code out of the docs into their own files and then include it this way; I did this with the [picraft docs](https://picraft.rtfd.org/) because it makes testing recipe scripts really simple (you just run them straight off - although in gpiozero's case there's the question of wiring it all up first :).

And yeah - I should go through the docs and tidy up the parameter lists when I've got a sec - unfortunately no time this week or next! PRs welcome :)

raspitv, 2016-04-16:
Another comment on this subject - perhaps it's unimportant, but I just came across this in the documentation...

gpiozero.LED(**pin**, active_high=True, initial_value=False)

Perhaps it's deliberate but use of the word **pin** there is misleading in my opinion since we're defaulting to BCM GPIO numbering shouldn't it be "port"? Because if you choose, say BCM GPIO25, it's pin 22. I always use "port" to denote port numbers and "pin" to denote pin numbers.

There's probably other reasons for it, but it stuck out as a confusing inconsistency to me. :)

waveform80, 2016-04-16:
Hmm, I take your point, but it's the first time I've ever heard them called "ports". If it were in more common usage, I'd be tempted but I've always heard them referred to as GPIO pins/BCM pins/Broadcom pins vs either physical pins/BOARD pins - "ports" is a new one for me. Assuming my experience is common, changing to "ports" would likely cause even more confusion - but perhaps my experience is abnormal? Would be interested to hear from the masses on this!

(I'm happy to concede that ports is the correct technical term, but if another terminology is in more common usage, as my experience suggests, I'm swayed toward keeping the current naming)

raspitv, 2016-04-16:
Oh I'm not saying it's wrong or you should change it, particularly if you intend to be able to use multiple numbering schemes, but I was under the impression that GPIO25 referred to GPIO port 25 and pin 25 refers to physical pin 25. (At least it's that black and white in my head). It's semantics, but I think I picked it up from the original Gertboard manual.

It is true that in Arduino we talk about digital pin 1 when it isn't physical pin 1, but Arduinos are labelled.

The thing is differentiating ports and pins is unambiguous. I just thought you should be aware of this as a potential source of confusion. There may not be a perfect answer.

waveform80, 2016-04-17:
An unambiguous scheme is definitely attractive. My problem basically boils down to:

1. At the moment, we've got a potentially ambiguous notation (GPIO pins), but so far no-one's complained about it (beyond the inevitable wrangling over BCM vs BOARD numbering).
2. You've made a good case that a completely unambiguous notation (GPIO ports with "pins" reserved for physical numbering) would be preferable.
3. However, I've no idea if switching to "GPIO ports" throughout the docs would now make things clearer or more confused ("port" to me is usually associated with things like a physical socket rather than a pin, although that's probably quite silly now I think about it :).
4. My suspicion is that anyone vaguely familiar with these things will have no problem with the switch, but I've simply no clue what effect (if any) it'd have on new-comers (a rather important demographic in the context of this library).

My default position when I don't know what to do is generally to leave things as they are until an issue becomes clear, or enough people weigh in against the status quo (though I acknowledge that the longer something like this is left, the more people become acquainted with it and the harder a switch becomes).

Time to poke the picademy alumni (@bennuttall, @MarcScott, @lesp, @martinohanlon - sorry, I don't know what Sarah / Steven / Damian / Spence / Josh are on GitHub), because at the moment I think all our [worksheets](https://github.com/waveform80/picademy_worksheets) use the "GPIO pin" notation too, and if we change this I really ought to update the worksheets to match (and we should make sure workshop spiel matches too!).

bennuttall, 2016-04-17:
I think the word pin is fine for this use.

raspitv, 2016-04-17:
This is how I tackled it in my RPi.GPIO documentation (in case it helps at all)...
http://raspi.tv/2013/rpi-gpio-basics-4-setting-up-rpi-gpio-numbering-systems-and-inputs

After explaining about BCM GPIO numbers and pin numbers and GPIO.setmode()

> **How to set up a GPIO port as an input**
> Use the following line of code…
> 
> GPIO.setup(Port_or_pin, GPIO.IN)
> 
> …changing Port_or_pin to the number of the GPIO port or pin you want to use. I’m going to use the BCM GPIO numbering and port GPIO25, so it becomes…
> 
> GPIO.setup(25, GPIO.IN)

Not trying to force my way on you at all though. Just trying to get things as clear as they can be. In my head, pins are physical pins. (e.g. one side of the Portsplus says _GPIO_ and the other side says _Pins_)

lesp, 2016-04-17:
I second the term GPIO pin. That is the official terminology is it not Ben?

Les

bennuttall, 2016-04-17:
I've also seen `channel` but I always preferred `pin`.

lurch, 2016-04-18:
It's explained fairly early-on in the Recipes page http://gpiozero.readthedocs.org/en/latest/recipes.html#pin-numbering which might be enough to avoid any confusion?

bennuttall, 2017-07-27:
@waveform80 surely this is no longer the case as of v1.4? Can we close this?

closed 2018-02-24 by bennuttall