gpiozero/gpiozero #209: 'Parameterisable' tests?

lurch, 2016-02-19:
While writing the tests for #203 and thinking about #206 I had a crazy idea for parameter-isable tests... i.e. tests where the pin-number, pin-implementation and pin-consumer could all be specified as optional parameters (with them defaulting to `2`, `MockPin` and `GPIODevice` respectively).

Because while https://github.com/RPi-Distro/python-gpiozero/blob/master/tests/test_devices.py#L47 works fine for `MockPin` and `GPIODevice`; as https://github.com/RPi-Distro/python-gpiozero/issues/193#issuecomment-184966586 shows it fails for `RPiGPIOPin` and `Button` (whereas it works fine for `RPiGPIOPin` and `LED`). So rather than having to copy'n'paste all the tests multiple times to test each of the combinations, it might be nice if we could rewrite the current `test_device_reopen_same_pin` as
```python
def test_device_reopen_same_pin(pin_num=2, pin_imp=MockPin, pin_device=GPIODevice):
    pin = pin_imp(pin_num)
    device = pin_device(pin)
    device.close()
    device2 = pin_device(pin)
    assert not device2.closed
    assert device2.pin == pin
    assert device.closed
    assert device.pin is None
```
and then we'd e.g. currently have `test_device_reopen_same_pin()` and `test_device_reopen_same_pin(3, RPiGPIOPin, LED)` passing, and `test_device_reopen_same_pin(4, RPiGPIOPin, Button)` failing.

But I've no idea if this is taking things one-step-too-far, and/or if there'd be any way to set up something like this in pytest.

Nzbuu, 2016-02-19:
https://pytest.org/latest/parametrize.html

lurch, 2016-02-19:
> https://pytest.org/latest/parametrize.html

Oooh, very neat! Something to look at over the weekend :)

waveform80, 2016-02-19:
Yup, that's exactly my intention (to use pytest's parametrisable fixtures) when writing the tests for the "real pin" classes. If you want a simple example of fixtures for functional testing, have a look at picamera's current test suite: https://github.com/waveform80/picamera/blob/master/tests/test_capture.py contains tests with fixtures, while https://github.com/waveform80/picamera/blob/master/tests/conftest.py defines the fixtures.

So, basically I'd envisage having a single set of tests which cover all of the abstract pin functionality, and we use fixtures to run those expected to work against the various concrete classes (obviously we'll need exclusions for things like implementations that explicitly don't support PWM etc.)

closed 2019-02-01 by bennuttall