gpiozero/gpiozero #212: 'state' property inconsistency

lurch, 2016-02-20:
According to http://gpiozero.readthedocs.org/en/latest/api_pins.html#gpiozero.pins.Pin.state the `state` property is 0 or 1 (i.e. an integer), or between 0.0 and 1.0 (i.e. a float) for PWM outputs.
However both `NativePin` and `MockPin` (I haven't looked at all the other `Pin` implementations yet) treat it as a boolean value. Indeed, `NativePin._get_state` says `return bool(...)`.
Is it worth modifying the `_get_state` implementations so that they match the documentation and always return a numeric value?

Feel free to close if you consider this irrelevant since the True/False and 1/0 values automatically coerce.

Also, it seems that some `Pin` implementations (e.g. NativePin) allow setting a `state` of e.g. 0.5 when PWM isn't in use (automatically converting it to 1 / True) whereas other implementations (e.g. RPiGPIOPin) throw a `PinInvalidState`. Could the documentation be tightened up on what the expected behaviour is please? :)  Or will this remain an "implementation-specific detail"?

waveform80, 2016-02-20:
Personally I'm not too bothered about the 0/1/False/True thing since as you note they automatically coerce (right down to the point that you can do things like indexing an array with a bool, e.g. ['No', 'Yes'][True]). That said, you're absolutely right that there's a certain inconsistency of treatment of other things like floats - and the docs don't really specify the expected behaviour. Yeah - let's come up with a good specification first and then tighten up the implementations to match it.

bennuttall, 2017-07-27:
Is this still relevant?