gpiozero/gpiozero #215: Pin.close() questions

lurch, 2016-02-20:
According to the [documentation](http://gpiozero.readthedocs.org/en/latest/api_pins.html#gpiozero.pins.Pin.close) for `Pin.close()`

> After this method is called, this Pin instance may no longer be used to query or control the pin’s state.

But observe:
```python
>>> from gpiozero.pins.rpigpio import RPiGPIOPin
>>> pin = RPiGPIOPin(21)
>>> pin.function
u'input'
>>> pin.state
0
>>> pin.function = 'output'
>>> pin.state = 1
>>> pin.close()
>>> pin.state
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "gpiozero/pins/__init__.py", line 130, in <lambda>
    lambda self: self._get_state(),
  File "gpiozero/pins/rpigpio.py", line 133, in _get_state
    return GPIO.input(self._number)
RuntimeError: You must setup() the GPIO channel first
>>> pin.function
u'input'
>>> pin.function = 'output'
>>> pin.state
1
```

i.e. merely setting the `RPiGPIOPin`'s `function` property is enough to automatically 're-open' the pin!

Should the `Pin` implementations have a `closed` property, like `GPIODevice`, with a corresponding `PinClosed` exception for any operations performed on a closed pin? Or is the "this Pin instance may no longer be used" statement in the documentation merely "advisory"? (or is the documentation simply wrong?)

Hmmm, I see that `GPIODevice.close()` calls `pin.close()`, which means we get this behaviour:
```python
>>> from gpiozero.pins.rpigpio import RPiGPIOPin
>>> from gpiozero import GPIODevice
>>> pin = RPiGPIOPin(21)
>>> device = GPIODevice(pin)
>>> device.value
False
>>> device.close()
>>> device.closed
True
>>> device2 = GPIODevice(pin)
>>> device2.value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "gpiozero/devices.py", line 323, in value
    return self._read()
  File "gpiozero/devices.py", line 245, in _read
    return self.pin.state == self._active_state
  File "gpiozero/pins/__init__.py", line 130, in <lambda>
    lambda self: self._get_state(),
  File "gpiozero/pins/rpigpio.py", line 133, in _get_state
    return GPIO.input(self._number)
RuntimeError: You must setup() the GPIO channel first
```

TBH I've got no idea what the expected/intended behaviour is :-/
Perhaps I'm just mucking around at too low a level, and at the "LEDs and Buttons" level that gpiozero is typically used at, everything magically comes out in the wash? ;-)

bennuttall, 2016-05-26:
I guess this is a minor bug. No one will really care that they can mess around with a closed device and later be able to read its state, but it should probably be addressed somehow (i.e. stop the ability to set the pin function when closed).

Also would it be possible to stop RPi.GPIO RuntimeErrors coming through?

waveform80, 2016-05-26:
Sorry I'm not around much at the mo (picamera 1.11 dragging on!). Briefly:

1. Pins can't *really* be closed (from this perspective the docs are simply wrong). This is because `Pin(2) is Pin(2)` is `True`; i.e. separately constructed Pin objects are actually the same instance provided they refer to the same physical pin. This is in order to ensure we maintain knowledge about things we can't query from the hardware like pull-up status.

2. `Pin.close()` is basically akin to RPi.GPIO's cleanup() - it just resets the pin to its "default" state, i.e. input (although actually this isn't entirely "default" for some GPIOs which should theoretically default to one of the alt-function states). Frankly, `close()` should probably be renamed something better like "reset".

3. The error you're getting on re-construction of GPIODevice against the same pin I'd consider an RPi.GPIO bug. The pin's actually in input state (at the hardware level) but because you've called cleanup() on it RPi.GPIO think it's in "unknown" state (despite the fact its own gpio_function() function will tell you it's in input mode...). More specifically it's because [gpio_direction](https://github.com/waveform80/rpi-gpio/blob/master/source/common.h) for that pin is -1 [after cleanup](https://github.com/waveform80/rpi-gpio/blob/master/source/py_gpio.c#L74) (it's another good example of having "multiple versions of the truth" - there's what the hardware actually is, and RPi.GPIO's idea of the hardware which is out of sync with reality).

closed 2019-02-01 by bennuttall