gpiozero/gpiozero #222: I2C GPIOs are hardcoded

lurch, 2016-02-22:
Something I just noticed, is that there's several places where you specify that GPIOs 2 & 3 (i.e physical pins 3 & 5) have fixed hardware pullups. However for very early Rev1.0 Pis, the physical pins 3 & 5 with hardware pullups are actually connected to GPIOs 0 & 1 - see http://elinux.org/RPi_Low-level_peripherals for more info.

GpioZero ought to check the board's Revision (as found in `/proc/cpuinfo` - see http://elinux.org/RPi_HardwareHistory) and use that to determine which GPIOs have the hardware-pullups.

And of course the ComputeModule doesn't have hardware pullups on any GPIO, although the intersection of ComputeModule users and GpioZero users is likely to be very small ;-)

waveform80, 2016-02-22:
Good point

waveform80, 2016-04-04:
Fixed in #246 (including for compute module users :)

closed 2016-04-04 by waveform80