gpiozero/gpiozero #226: os.system breaks event detection under CPython interpreter with RPi.GPIO pins

waveform80, 2016-02-24:
Wire a button to a GPIO (say, 20) and attempt the following in the CPython interpreter:

```python
>>> import os
>>> from gpiozero import Button
>>> btn = Button(20)
>>> btn.wait_for_press()
True
>>> os.system('echo foo')
foo
0
>>> btn.wait_for_press()
```

After the first `wait_for_press` call, hit the button and the prompt should return. However, the second `wait_for_press` call will hang forever because for some reason executing anything via `os.system` seems to disable RPi.GPIO's event handling.

After an evening of playing with this I've got numerous observations but still no clue as to the root cause:

1. I've tested this with `RPiGPIOPin` (based on RPi.GPIO), `PiGPIOPin` (based on pigpiod), and `NativePin` (the pure Python implementation). Only `RPiGPIOPin` fails in this manner; I'd hazard a guess that `RPIOPin` (based on RPIO) might as well given both libraries handle event detection equivalently (epoll wait on a sysfs file representing the pin state), but I haven't dug out a Pi 1 to test RPIO with yet.
2. This only occurs with `os.system`. Launching the same external process via `subprocess.call` works just fine so it's nothing to do with sub-processes per sé. Annoyingly, `os.system` has one of the simplest implementations of any Python command: it's basically just a C system(3) call ... so digging into this may mean digging into the C stdlib (and while I've never looked at system(3) in there, I have a nasty feeling it's a complex one ...).
3. This only seems to occur in the CPython interpreter. Attempting it from IPython works just fine, as it does from IDLE's shell (this is particularly confusing?!)
4. On the off-chance something was causing RPi.GPIO to lose its handle to the sysfs value file representing the pin I took a look with `lsof` before and after the `os.system` call but no change was observed.

As far as I can tell this was first encountered here: https://www.raspberrypi.org/forums/viewtopic.php?t=97152&p=674670 but no resolution is suggested beyond what's discussed above. I can't find any other references on the 'net to this specific issue (and it's definitely tied into `add_event_detect` in RPi.GPIO).

Next step is probably to build a custom RPi.GPIO with some debugging in the event detection thread but I have a horrid feeling that sooner or later this is going to wind up going down the rabbit hole into the C stdlib ... If anyone can shed any light on this before that happens I'll be most grateful!

lurch, 2016-02-25:
I don't have any insight, but I did manage to distil the problem into a simple non-interactive test-app using only RPi.GPIO:
```python
import os
import RPi.GPIO as GPIO
import time

# This test assumes you've wired the following GPIO pins together
TEST_PIN = 22
INPUT_PIN = 27

# We need to give the callback time to get executed, before checking in the main thread
sleep_for = 0.1

changed = False

def pin_changed(channel):
    global changed
    changed = True

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(INPUT_PIN, GPIO.IN)
GPIO.setup(TEST_PIN, GPIO.OUT)

GPIO.output(TEST_PIN, False)
assert not GPIO.input(INPUT_PIN)

GPIO.add_event_detect(INPUT_PIN, GPIO.BOTH, pin_changed)
assert not changed

changed = False
GPIO.output(TEST_PIN, True)
assert GPIO.input(INPUT_PIN)
time.sleep(sleep_for)
assert changed

changed = False
GPIO.output(TEST_PIN, False)
assert not GPIO.input(INPUT_PIN)
time.sleep(sleep_for)
assert changed

#os.system('echo foo')

changed = False
GPIO.output(TEST_PIN, True)
assert GPIO.input(INPUT_PIN)
time.sleep(sleep_for)
assert changed

changed = False
GPIO.output(TEST_PIN, False)
assert not GPIO.input(INPUT_PIN)
time.sleep(sleep_for)
assert changed
```

On my headless Pi2, python, python3, ipython and ipython3 all run the script without errors when the `os.system` is commented-out, but when the `os.system` is uncommented ipython and ipython3 still work okay, but python and python3 both fail with an AssertionError on line 47 (i.e. the first `assert changed` after `os.system`).

I guess this probably needs to be reported upstream? (although it seems there's also quite a few other event-detect tickets already open, but I'm not sure if any of them cover this particular case?)

I haven't tested with any of the other GPIO-backends.

bcroston, 2016-02-25:
Raise a ticket against RPi.GPIO and I'll see what I can do.

lurch, 2016-02-25:
Raised a ticket against RPi.GPIO https://sourceforge.net/p/raspberry-gpio-python/tickets/127/

bcroston, 2016-10-30:
This should now be fixed in RPi.GPIO 0.6.3

bennuttall, 2018-09-24:
Never got around to testing this, can confirm it's fixed.

closed 2018-09-24 by bennuttall