gpiozero/gpiozero #227: GPIO Zero vs. Rpi.GPIO worse performance with event detect

alexellis, 2016-03-02:
I've got a small program in Python 2.7 that 'watches' two PIRs and resets a Redis NoSQL counter when either are activated. If there PIRs are not reset for N seconds, then a main loop will increment the counter. 

In GPIO Zero this draws a constant 10-12% of CPU on the Raspberry PI Zero, but with the older RPI.GPIO library and GPIO.add_event_detect CPU usage barely goes above 1%. 

I've not included the code for redis_controller since it is constant between the two examples. The main counting loop that increments the counter is also constant between the two examples.

Can anyone suggest why they may be so different? I imagine that a more complex configuration in GPIO Zero would use an incrementally higher CPU percentage.

(Also posted as question on RPI Forum, but I think this is the better place for the question)

```
import time
import RPi.GPIO as GPIO
import os
from redis_controller import responder

GPIO.setmode(GPIO.BCM)

LeftPir = 27
RightPir = 17

GPIO.setup(LeftPir, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(RightPir, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

responder1 = responder("timer_1")

def pir_change(channel):
    print "Change"
    if(GPIO.input(channel) == False):
        return
    if(GPIO.input(channel)):
        responder1.clear()
        print "Ticking on channel " + str(channel)

GPIO.add_event_detect(RightPir, GPIO.RISING, callback=pir_change,bouncetime = 1000)
GPIO.add_event_detect(LeftPir, GPIO.RISING, callback=pir_change,bouncetime = 1000)

secondsPerTick = 10

secs=time.time()
while(True):
    if(time.time() - secs >= secondsPerTick):
        responder1.tick()
        print "Counting idle... "+str(responder1.get()) + " mins"
        secs = time.time()

    time.sleep(0.02)
```

```
import time
import os
from gpiozero import MotionSensor
from redis_controller import responder

LeftPir = 27
RightPir = 17

left = MotionSensor(LeftPir)
right = MotionSensor(RightPir)
responder1 = responder("timer_1")

def resetPir():
    print ("-")
    responder1.clear()

for pir in [left,right]:
    pir.when_motion = resetPir

secs=time.time()
while(True):
    if(time.time() - secs >= 10):
        responder1.tick()
#        print "Counting idle... "+str(responder1.get()) + " mins"
        print(".")
        secs = time.time()

    time.sleep(0.02)
```

lurch, 2016-03-02:
I wonder if this is related to #181 ?

waveform80, 2016-03-02:
My immediate hunch (unconfirmed, untested - don't have time this week, sorry!) is that this'll be down to the fact MotionSensor does *much* more than just set an event to be notified of a rising edge (which is all the code above is up to). Because it descends from SmoothedInputDevice it fires up a background thread which continually samples the state of the motion detector, pushing those values into a limited size queue and using a median of that queue for the value of the detector. The `queue_len` defaults to one which makes all this quite pointless (because the default motion sensor we assume, the camjam kit one, has an incredibly long notification and inhibit time of several seconds, even when the Tx pot is cranked all the way down). However other motion sensors can have considerable faster times (making them "jittery") which is why the functionality is there for those that need it.

Can't say I looked at the CPU cost before now, but hearing that we're burning a mere 12% without any serious attempts at optimization my initial reaction is "that's fine".

alexellis, 2016-03-02:
Off topic question.. I found that when set to the quickest reset time the £1/unit PIRs (like CamJam) only seem to fire every 1.5 seconds or so. Which PIRs reset so quickly? I'd like to look into getting some if they are not too expensive in comparison.

RE: CPU hit
I understand you may have to prioritise this request, but a 1% vs 12% constant CPU load for the same work (required) really puts me off. I wouldn't feel great about recommending GPIO zero - especially if the PI ends up doing more than polling a PIR..

It doesn't sound as important for you guys, but is there scope for your team to look at an option for the slow-reset PIRs?

waveform80, 2016-03-03:
Quick solution: just use DigitalInputDevice or Button which'll just treat whatever you've connected as a simple on/off device rather than something that needs "smoothing". On motion sensors that need smoothing - I haven't personally used any, but many [can apparently be hacked](http://www.neufeld.newton.ks.us/electronics/?p=208) to have a very short response time.

That said, I was frankly a bit skeptical of the 1% vs 12% claim; that sounds awfully high. So, I tested the following two variants of your code (which I think do something similar but without any external dependencies, c.f. the redis_controller module):

### 227_rpigpio.py

```python
from time import time, sleep
from RPi import GPIO

GPIO.setmode(GPIO.BCM)

LEFT_PIR = 20
RIGHT_PIR = 21

GPIO.setup(LEFT_PIR, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(RIGHT_PIR, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)

last_update = time()
last_reset = last_update

def pir_change(channel):
    print("Change")
    global last_reset
    last_reset = time()

GPIO.add_event_detect(LEFT_PIR, GPIO.RISING, callback=pir_change, bouncetime=1000)
GPIO.add_event_detect(RIGHT_PIR, GPIO.RISING, callback=pir_change, bouncetime=1000)

while True:
    if time() - last_update >= 10:
        print("Idle %d secs" % (time() - last_reset))
        last_update = time()
    sleep(0.02)
```

### 227_gpiozero.py

```python
from time import time, sleep
from gpiozero import MotionSensor

left = MotionSensor(20)
right = MotionSensor(21)

last_update = time()
last_reset = last_update

def pir_change():
    print("Change")
    global last_reset
    last_reset = time()

left.when_motion = pir_change
right.when_motion = pir_change

while True:
    if time() - last_update >= 10:
        print("Idle %d secs" % (time() - last_reset))
        last_update = time()
    sleep(0.02)
```

Using top / htop to monitor the CPU load of the running python process in each case, the gpiozero one appears to use 1% CPU time, while the RPi.GPIO one uses 0.3% which sounds a lot more likely to me. For reference, this is on a Pi 2 with no overclocking - I'm sure it'd be larger on one of the single-core platforms like an A+, but 11% still sounds like an awful lot.

I'll try and dig out an A+ tomorrow for further testing, but to be blunt this still sounds like a case of premature optimization to me (sorry, sleep deprivation - haven't got the energy to be politic right now).

alexellis, 2016-03-03:
@waveform80 thanks very much for your input on this. I think the DigitalInputDevice would be a good option in light of what you've said. Few pictures to show the readings I'm getting on htop:

http://imgur.com/a/gl8Jh

Baseline (no Python process), then the zero library (6.7+2.7+2.7 observed when waving hand in front of sensor) then RPI.GPIO.

This is running through a Raspberry PI Zero. A Model A+ would be roughly equivalent. 

> In GPIO Zero this draws a constant 10-12% of CPU on the **Raspberry PI Zero**, but with the older RPI.GPIO library and GPIO.add_event_detect CPU usage barely goes above 1%.

*If unmodified PIRs are the common sort, and fast resetting are only available through hardware-hacking, perhaps MotionSensor could inherit from DigitalInputDevice and SmoothedMotionSensor - or FastMotionSensor etc could be the current incarnation? Just seems like the least surprising arrangement.*

waveform80, 2016-03-03:
Yeah ... the headline figures on those pics are largely meaningless as we're dealing with a multi-user multi-tasking OS. As for the 6.7+2.7+2.7 - I suspect only the first one is meaningful there; the latter two are measures of the background threads but they're not *added* to the first number. Turn off your view of userland threads and you'll just see one value for the process which I'm guessing will be 6.7% ish (which sounds much more likely to me).

Just to make sure I wasn't mis-remembering things, I tried my two scripts above on an A+ (don't have a zero with a header soldered on at the mo, but an A+ is similar enough - single core, low power, and there's one spare on my desk). The RPi.GPIO script hovered around 0.9% CPU, and the gpiozero script hovered around 4.5% CPU (similar ball-park).

All in all, measuring CPU usage is a tricky/subtle thing and generally not very useful; in my time as a UNIX admin I usually relied on system load as a more useful measure. Even then, it's a tricky thing to interpret. For example, tie a button to a LED in gpiozero with source/values. You'll peg the CPU at 100%, but the rest of the system will still be reasonably responsive (the process is constantly "doing" something, but that something isn't terribly complicated, involves almost no memory churn and no I/O). The system load will probably hover around 1 after a minute (assuming an otherwise idle system). Add another button+LED with source/values; your system load will eventually reach 2 (two processes/threads constantly "doing" something). Your CPU will still be pegged at 100% though and the rest of the system will still be reasonably responsive. So what does the CPU% mean? What does the system load mean? And how come we can still get stuff done? The answers to which come down to understanding the workload, whether we've maxed out other things like IO bandwidth, whether the overall working set is now a significant percentage of available RAM, blah blah blah.

(I should mention that purely because people get freaked out by large CPU loads, and I'm fearful of getting swamped by tickets when I get around to writing loads of source/values recipes, I'll be fixing #181 before 1.2)

Finally, as mentioned, this is all desperately premature optimization: there's no complaint about performance of the class in terms of responsiveness or results, no use-case has been presented for why those few percentage points are desperately needed, and frankly if I cared about wasting CPU cycles needlessly, I wouldn't be writing in an interpreted dynamic language several thousand times slower than C. But I do because *wasting CPU cycles largely doesn't matter* - not when you've got 900 million of them to play with each second.

If anyone's *that* perturbed by a seeing a needle rise above 1 when it could be sat at 1, I'll accept PRs to optimize things but it's just not on my list (and probably won't be for the foreseeable future).

closed 2016-05-26 by bennuttall