gpiozero/gpiozero #228: RGBLED.cycle()

bennuttall, 2016-03-03:
Suggestion:

We add a `cycle` method to `RGBLED` which cycles through the colours. Parameters similar to `blink` (maybe `cycle_time`, `n`, `background`, `start_color`)

lurch, 2016-03-03:
Been having a think about this, and I guess in a way it's slightly analogous to #149 
So what about a method signature like:
`def cycle(on_time=1, fade_time=0, colors=((1,0,0), (0,1,0), (0,0,1)), n=None, background=True)`
?

Calling `rgbled.cycle()` would then display a repeated sequence of "red on for 1 second, green on for one second, blue on for one second" and `rgbled.cycle(0, 2)` would display a repeated sequence of "2 seconds smoothly fading (re-using the `lerp` code from [here](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L644) ) from red to green, 2 seconds smoothly fading from green to blue, 2 seconds smoothly fading from blue to red".
And you could also do `rgbled.cycle(0, 1, ((1,0,0), (1,1,0), (0,1,0), (0,1,1), (0,0,1), (1,0,1)) )` to fade through the full HSV range.
(So perhaps that'd be a better default-value for `colors`? Meaninig that `rgbled.cycle()` with no args would then 'flick' through "red, yellow, green, blue, cyan, magenta").

And given that `cycle` with no-args is kinda-similar to `blink` with no-args, maybe we could also add a `fadecycle` (or `smoothcycle`) which simply calls `cycle` with different default-args; similar to how `pulse` simply calls `blink` with different default-args?

Alternatively, if we wanted a simpler interface (perhaps I'm over-complicating things again) we could go with Ben's suggestion of:
`def cycle(cycle_time=1, start_hue=0, n=None, background=True)`
and then always smoothly cycle through the full range of colours using something like https://en.wikipedia.org/wiki/HSL_and_HSV#/media/File:HSV-RGB-comparison.svg
(In this case `start_hue` would be the [hue angle](https://en.wikipedia.org/wiki/HSL_and_HSV#Hue_and_chroma) which we could either specify as 0-360, or scale to 0-1 for similarity to the rest of the GpioZero API. And I wonder if a `cycle_time` of 3 would be a better default, otherwise the colours might appear to cycle too fast? Depends whether we want `cycle_time` to indicate the time to fade from one colour to the next, or the time taken to do a complete-cycle through all the colours)

Hmmm, and maybe we could add an `LEDBoard.cycle()` method too, to light up each of the LEDs one at a time? ;-)
`def cycle(on_time=1, fade_time=0, start_index=0, n=None, background=True)`

pcopa, 2016-03-08:
Fruit for thought:

I'm aware that everyone has a lot on their plate right now, but if we start something new I really think we should take in consideration a fourth channel. All new 5050 chips have 4 colors RGBW. These are getting very popular and are popping up in all kinds of applications. Just Google "rgbw 5050", also see "Adafruit rgbw".

Where should I insert a comment regarding this just as a placeholder for future implementations?

lurch, 2016-03-08:
@pcopa Please just create a new issue, and I'm sure someone will get to it eventually :)