gpiozero/gpiozero #230: Compositing robots with sensors

DeeJay, 2016-03-15:
The existing implementations of RyanteckRobot and CamJamKitRobot assume a device composed only of a pair of motors.

The CamJamKitRobot, and other commercially available kits such as Pi2Go from 4Tronix, can also be equipped with a variety of sensors, already supported by code such as LineSensor and DistanceSensor.

There is already a simple Recipe for this situation in the Docs.

But can you advise on how to implement a more 'pythonic' Class construction for a particular model of Robot, so that the end-user simply has to to say:

myrobot = MyRobot(left=(38, 37), right=(36, 35), lineleft=34, lineright=33, distance=32)  # could use defaults

and behind the scenes all the accessors are created  for the sensors as well as the motor functions?

(I am not a fluent user of OO python, so it maybe that there is an example already in the source code that I have overlooked.)

bennuttall, 2016-04-05:
I like the idea. We're just trying to get v1.2 out the door, but we should revisit for v1.3.

lurch, 2016-04-05:
Perhaps this (partially) ties in with my ideas in #219 ?

waveform80, 2016-04-05:
Yup - we'll have a look for 1.3; @lurch - I haven't checked out your polymorphic params branch yet (sooo many tests to write!) but it sounds fascinating - I'll definitely have a look when we get onto 1.3 :)

lurch, 2016-06-06:
Would this 'just' be a case of:
```python
class MyRobot(CompositeDevice):
    def __init__(self, left, right, lineleft, lineright, distanceecho, distancetrigger):
        super(MyRobot, self).__init__(robot=(left, right), lineleft=LineSensor(lineleft), lineright=LineSensor(lineright), distance=DistanceSensor(distanceecho, distancetrigger), _order=('robot', 'lineleft', 'lineright', 'distance'))
```
?

And then you _should_ (untested!) be able to do things like:
```python
myrobot = MyRobot(left=(8, 7), right=(6, 5), lineleft=4, lineright=3, distanceecho=2, distancetrigger=9)
myrobot.robot.right()
sleep(2)
myrobot.robot.stop()
print(myrobot.lineleft.is_active)
print(myrobot.distance.distance)
myrobot.distance.when_in_range = myrobot.robot.backward
myrobot.distance.when_out_of_range = myrobot.robot.stop
pause()
```

Or did you want something "more integrated" ?

DeeJay, 2016-06-06:
As the person who raised the Issue, I'd say that pretty much covers it. (But I haven't tested the suggestion.)

The 'icing on the cake' would be to use my own Class in the same way as the 'built in' Classes, and be able to create an instance of my robot with default pin assignments by saying:
`deejay = MyRobot()`
(In the same way that one can say:
`ryan = RyanTeckRobot()` )

Obviously that would mean moving the pin assignments into the Class definition, rather than specifying them when an instance is created. Can you illustrate how the constructor in the Class definition would be modified to incorporate these defaults?

If that all works as expected then I'd be happy to say 'job done' if this was simply included as a 'Recipe', if that is what is being suggested.

(As was hinted in the original Issue, this may well be a 'teaching' issue regarding some slightly-more-than-trivial OO python rather than something that needs solving by adding code to the library. For example, I deduce that _order specifies the methods that are to be made public: is that correct?)

lurch, 2016-06-06:
> The 'icing on the cake' would be to use my own Class in the same way as the 'built in' Classes, and be able to create an instance of my robot with default pin assignments by saying:
deejay = MyRobot()

That's easy, just change it to:
```python
class MyRobot(CompositeDevice):
    def __init__(self):
        super(MyRobot, self).__init__(robot=Robot((8, 7), (6, 5)), lineleft=LineSensor(4), lineright=LineSensor(3), distance=DistanceSensor(2, 9), _order=('robot', 'lineleft', 'lineright', 'distance'))
```
and then you'd be able to do `deejay = MyRobot()`. (As a comparison, look at the complete sourcecode of RyanteckRobot [here](http://gpiozero.readthedocs.io/en/stable/_modules/gpiozero/boards.html#RyanteckRobot) )

But note that this hard-codes the pins, meaning that they can't be changed. Fine if you're creating a PCB, but not so useful if you're prototyping things on a breadboard. If you wanted to add a bit more flexibility, you can combine the approach in my earlier comment with default values, like this:
```python
class MyRobot(CompositeDevice):
    def __init__(self, left=(8, 7), right=(6, 5), lineleft=4, lineright=3, distanceecho=2, distancetrigger=9):
        super(MyRobot, self).__init__(robot=Robot(left, right), lineleft=LineSensor(lineleft), lineright=LineSensor(lineright), distance=DistanceSensor(distanceecho, distancetrigger), _order=('robot', 'lineleft', 'lineright', 'distance'))
```
and then you could either do `deejay = MyRobot()`, or if you wanted to move your line-sensors onto different pins you could do e.g. `deejay = MyRobot(lineleft=14, lineright=12)`.

Python gives you quite a bit of flexibility on how you choose to do things :)  Apologies if I've accidentally confused things by choosing the same names for the parameters and the CompositeDevice sub-devices. The previous example would work just as well when written as:
```python
class MyRobot(CompositeDevice):
    def __init__(self, left_motor_pins=(8, 7), right_motor_pins=(6, 5), lineleft_pin=4, lineright_pin=3, distance_pins=(2, 9)):
        super(MyRobot, self).__init__(wheels_controller=Robot(left_motor_pins, right_motor_pins), left_linesensor_object=LineSensor(lineleft_pin), right_linesensor_object=LineSensor(lineright_pin), distance_object=DistanceSensor(distance_pins[0], distance_pins[1]), _order=('wheels_controller', 'left_linesensor_object', 'right_linesensor_object', 'distance_object'))
```
(and then you'd need to do `deejay.distance_object.when_in_range = deejay.wheels_controller.backward` etc.)

>  I deduce that _order specifies the methods that are to be made public: is that correct?

Not quite. Unfortunately the CompositeDevice class in gpiozero is quite complicated, but it's not very well [documented](http://gpiozero.readthedocs.io/en/stable/api_boards.html#compositedevice).
Every device in GpioZero has a `value` property - so using our earlier example we might get
```python
print(deejay.value)
MyRobotValue(robot=RobotValue(left_motor=0, right_motor=0), lineleft=False, lineright=False, distance=0.3)
```
(Note here that `Robot` is actually also a CompositeDevice, which we're nesting inside our `MyRobot` CompositeDevice)
The `_order` parameter simply specifies which order the sub-devices should be displayed in the `value`. So if we instead specified `_order=('distance', 'lineright', 'robot', 'lineleft')` then our `value` would be returned as `MyRobotValue(distance=0.3, lineright=False, robot=RobotValue(left_motor=0, right_motor=0), lineleft=False)`. Simply choose whatever order makes most sense for your project. Or if you don't really care about the order of items in `value` you can simply omit `_order` and it'll automatically display them in an alphabetical order, i.e. `MyRobotValue(distance=0.3, lineleft=False, lineright=False, robot=RobotValue(left_motor=0, right_motor=0))`.

DeeJay, 2016-06-08:
Thank you for your patient and thorough explanation.

As far as I am concerned there is enough information here to regard the Issue as Closed.

closed 2016-06-08 by DeeJay

lurch, 2016-06-08:
Glad I could help :)