gpiozero/gpiozero #232: Error on second execution of script with function called on button press

ukBaz, 2016-03-19:
I've taken the following example from the documentation:
```python
from gpiozero import Button
from signal import pause

def say_hello():
    print("Hello!")

button = Button(2)

button.when_pressed = say_hello

pause()
```
The first time of running it all works correctly. To stop it running I do a `<ctrl-c>` and then on second execution I get an error.
```
pi@raspberrypi ~ $ python3 ~/test_light.py 
^CTraceback (most recent call last):
  File "/home/pi/test_light.py", line 16, in <module>
    pause()
KeyboardInterrupt
pi@raspberrypi ~ $ python3 ~/test_light.py 
Traceback (most recent call last):
  File "/home/pi/test_light.py", line 6, in <module>
    button = Button(25)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 405, in __init__
    super(Button, self).__init__(pin, pull_up, bounce_time)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 262, in __init__
    bouncetime=-666 if bounce_time is None else int(bounce_time * 1000)
RuntimeError: Failed to add edge detection
```

If I run the same script a third time then it will run correctly.

bennuttall, 2016-03-19:
I believe this is was a bug in RPi.GPIO which was fixed. Can you update both RPi.GPIO and GPIO Zero and see if you still have an issue?

```bash
sudo apt-get install python3-rpi.gpio python3-gpiozero
# or
sudo apt-get install python-rpi.gpio python-gpiozero
```

should upgrade both packages.

ukBaz, 2016-03-19:
Thanks Ben.

Yep, it was my mistake. Had forgotten to do a ```sudo apt-get update``` 

All working now.

closed 2016-03-19 by ukBaz