gpiozero/gpiozero #237: 'Keyboard controlled robot' requires python3-evdev

DeeJay, 2016-03-20:
re: http://gpiozero.readthedocs.org/en/v1.1.0/recipes.html#keyboard-controlled-robot

This recipe needs python3-evdev to be installed as a prerequisite: this is not installed by default by raspbian-jessie. 

The installation instructions at https://python-evdev.readthedocs.org/en/latest/ are clear enough, but have further prerequisites that are not met by default.

The overall effect is that using this recipe is not quite 'frictionless.'

Also, it was not immediately obvious to me that the evdev library only deals with directly-connected physical devices. It does not interpret keystrokes from a 'virtual keyboard' such as might be the case if ssh'ing into a robot via a wifi connection.

bennuttall, 2016-03-20:
We're aware. See https://github.com/RPi-Distro/python-gpiozero/blob/807bdff2e4e783e474a51e95902589684037aff0/docs/recipes.rst#L570

bennuttall, 2016-03-20:
(actually it's just a comment so you'll need to view the raw to see what I mean)

waveform80, 2016-03-20:
Oooh - I wrote the code for that on the picademy cheat-sheet the other day - I'll fix that up tonight. Unfortunately I did find that curses doesn't work under IDLE (not terribly surprising) so the resulting code does have to be run from the terminal but at least there's no external deps.

commit 56bd1b3c (2016-04-10): Merge pull request #243 from waveform80/keyboard-robot

Fix #237
commit bb13e9fe (2016-04-10): Fix #237

Adds a keyboard controlled robot recipe that uses curses. Advantage is
that no external deps are required, disadvantages are that it'll only
work from a terminal (or terminal window, basically not IDLE).
closed 2016-04-10 by bennuttall