gpiozero/gpiozero #238: Asymmetric SmoothedInputDevice?

lurch, 2016-03-27:
Just a thought I had while writing the code for http://raspi.tv/2015/gpio-zero-test-drive-making-light-of-security#comment-60978 (which might itself be an interesting recipe for gpiozero?)

Specifically for a PIR (and perhaps for other devices too?) you might want the "smoothing" on the inactive->active transition to be minimal (i.e. no delay when motion first detected), but maximal on the active->inactive transition (i.e. don't report "no motion" until there's _really_ no motion).

Dunno if it's feasible or that useful, but I just thought I'd throw the suggestion out there...

lurch, 2016-03-27:
OTOH maybe we could have e.g. an `after_no_motion` callback which works in a similar way to the `when_held` in #115

pcopa, 2016-04-17:
Could this also be used to smoothen out ADC outputs?

I'm testing a setup for PH and ORP to control the aquarium or pool's chemicals needs. The sensors are hooked up to high impedance amplifiers that feed the ADC.  The ADC values jump around significantly to where a singe value makes no sense. It would be nice to have some kind of smoothing algorithm like a multimeter has.

lurch, 2016-04-18:
> Could this also be used to smoothen out ADC outputs?

There's already http://gpiozero.readthedocs.org/en/latest/api_input.html#smoothedinputdevice for smoothing 'digital' values, but I guess an option for smoothing 'analogue' values could be useful too (although perhaps it should be a separate issue to this one). I wonder if it could be a variant of http://gpiozero.readthedocs.org/en/latest/api_tools.html#gpiozero.tools.queued

lurch, 2016-04-21:
@pcopa does #292 help? :)

pcopa, 2016-04-22:
@lurch    Will look into it. Need to recover a HD failure first and i'l be out of town the weekend..