gpiozero/gpiozero #248: Add Servo

bennuttall, 2016-04-04:
We should add a Servo class, or more likely, a series of Servo classes (e.g. `Servo180`, `Servo360`...). I'm not sure what the variety of different servos is like, but perhaps there needs to be much configuration ability on init.

The way servos work is with PWM, setting a specific duty cycle to turn the servo to a specific angle. The one I have, a [Tower Pro SG90](http://www.micropik.com/PDF/SG90Servo.pdf), moves proportionately between 0 degrees and about 170 degrees when the value is set between 0.02 and 0.12, but any value outside that does nothing.

I started playing with one using `PWMOutputDevice` and got it turning to certain angles by sending various signals via the `value` property.

When thinking about implementing a class for this, I realised I probably didn't want to extend `PWMOutputDevice` as is, because I didn't need `on()`, `off()`, etc. so I guess we should introduce a `DigitalPWMOutputDevice` class for this, and all existing classes which currently extend `PWMOutputDevice` would switch to `DigitalPWMOutputDevice`, and `Servo` would extend the new `PWMOutputDevice`. Perhaps "digital" is the wrong word in this case...

I'm not certain what the servo API should look like, but I guess ideally, you could set it by angle, and maybe setting `value` between 0 and 1 would move proportionally between the minimum and maximum positions.

bennuttall, 2016-04-04:
A possible API:

```python
>>> servo = Servo180(17)
>>> servo.angle = 0
>>> servo.value
0.0
>>> servo.angle = 180
>>> servo.value
100.0
>>> servo.value = 0.5
>>> servo.angle
90.0
```

So that `value` and `angle` are coupled between min and max angles according to the rotation potential of the servo (in this case 180 degrees).

However, this means that a servo's `value` does not correspond directly to its duty cycle, like other PWM objects. This would be mapped between the duty cycle of the 0 degree position and that of the maximum angle (e.g. 180 degree position).

We could provide a generic `Servo` class which takes parameters for `max_angle`, `min_duty_cycle`, `max_duty_cycle`, or something similar. Alternatively we could provide `Servo180`, `Servo360` and so on. I'm not sure how much servos tend to vary. If they're all over the place we'll have to provide lots of configuration. If there's a decent set of standard ones then we should be able to just use those.

Any thoughts @waveform80?

WayneKeenan, 2016-04-05:
Please keep/expose on() and off()  ( synonymous to the Arduino servo attach() and detach() functions)

The attach/detach allows saving on power when the servos don't need to move, and also help keep some robots still when not in use. As many servos oscillate slightly at any set angle and this can become more pronounced on something like a robot arm (e.g. MeArm), performing an off (or detach) after a few seconds, if there is no required movement, minimises this.

A `min_angle` property, and bounding to and above it, would be useful for when the servos are installed into a device that physically restrict their movement, e.g. the joints of a robot arm.  
 
Having a setPosition(float) that takes a range 0.0 ..1.0 and maps it to min_angle .. max_angle could be handy.

Being able to set a `servo.frequency` for the underlying PWM frequency may be needed by some servos.

Also, continuous rotation servos use the 'angle' as a speed, so perhaps consider this type of Servo in the new class?

sebleedelisle, 2016-04-05:
This is great! As Wayne also mentions, I would take a look at the Arduino Servo object for inspiration : https://www.arduino.cc/en/Reference/Servo

It feels a bit bulky to have different objects for a 180 and 360, I would default to 180 as this seems by far the most common. I like the max_angle, min_duty_cycle, max_duty_cycle options as long as most of the time the defaults were the most common. Again, see the default set up for the Arduino Servo object. 

Nice to have : a moveTo or similar that interpolates from the current position to the target position with a specified duration, and ease in and ease out (either a linear ramp up and down or an exponential ease-in/out if you're feeling fancy). Otherwise you're relying on the in-built circuitry in a server to manage that, and it's not often very pleasant. (I have some experience in this area so can likely help if required).

bennuttall, 2016-04-05:
on/off or equivalent makes sense actually, to get it to disengage.

lurch, 2016-04-05:
Is using PWM necessarily the best idea for this? It's something I've not (yet) experimented with myself, but I recall reading in the past that trying to use PWM (especially from Python - see http://raspi.tv/2013/how-to-use-soft-pwm-in-rpi-gpio-pt-2-led-dimming-and-motor-speed-control) is too jittery on a RaspberryPi for accurate servo control (although I guess this may be less of an issue on the Pi2 and Pi3 than it was on the Pi1), and that's why things like [ServoBlaster](https://github.com/richardghirst/PiBits/tree/master/ServoBlaster) was developed, to allow much more accurate servo control?

And having Servos as an "object in their own right" and _not_ inherited from PWMOutputDevice also allays Ben's class-inheritance (and appropriate-method-overloading) concerns.

carlosperate, 2016-04-05:
I agree as well that maybe having different classes for 180 or 360 is a bit too much, sounds better to have one with sane defaults that can be edited in the constructor. The only thing I would maybe separate would be servos from "continuous servos",  have you guys given any though to that?

WayneKeenan, 2016-04-05:
@lurch I understand that by now* it might be possible to load Videocore's VPU with a small chuck of code to drive PWM on the GPIO pins.

*at a CamJam about/over a year ago I queried using VideoCore todo PWM in a conversation with Eben and he said that the VPU can access the GPIO. He also mentioned the functions the VPU were currently (at the time) performing were moving off the VPU, so it might become available for such 'user' programs. (I think there's a documentation/ a VPU compiler now too (?) )   

This is perhaps a far more intricate system level project todo and better kept separate from the gpiozero repository... :)

bennuttall, 2016-04-05:
> sounds better to have one with sane defaults that can be edited in the constructor

That's pretty much the philosophy I aim for. As long as there are reasonable standards we can adhere to, that makes sense.

> The only thing I would maybe separate would be servos from "continuous servos", have you guys given any though to that?

That makes sense. Will bear in mind.

carlosperate, 2016-04-05:
I haven't use the PWM in the Pis that much to be honest, but are we saying that the hardware peripheral is not reliable enough to control a servo? Or is the general conversation about alternative PWMs as a software varsion on other GPIOs?

WayneKeenan, 2016-04-05:
@carlosperate Just options...    The Pi's PWM for servo control is 'good enough' to get demos and fun things up and running, it's probably 'not good enough' for controlling a quad-copter though, but what people do with the Pi is always enlightening.

This is a demo (not mine) of using a Python wrapper around the ServoBlaster (software + DMA driven) PWM, which would seem ok for at least 4 PWM's, ServoBlaster says it supports up to at least 8.   http://fortoffee.org.uk/2015/02/me-arm-and-me-raspberry-pi/

Although ServoBlaster introduces another dependancy.  If I get a chance, I might plug in my MeArm and try controlling it using RPi.GPIO PWM support.

sebleedelisle, 2016-04-05:
Re continuous servos, I think most people these days equate a "servo" to a "hobby servo" or "RC servo" ie, the cheap plastic things with built-in controllers that use the pulse system for positional information. 

As someone who uses industrial servos with separate controllers it can get confusing but I'm not really sure if we can fix that at this stage! 

Of course the other type of continuous servo is a hobby servo that has had the positional feedback system pulled out (usually a pot), I'm not really sure how to cater for that but I think the basic servo API would suffice.

As for using PWM, it is basically PWM but with short pulse between 1 and 2ms followed by a wide gap. If the pulse is 1ms then the motor goes to 0º, if it's 2ms then it goes to 180º. If the servo doesn't get regular pulses or the timing isn't accurate enough then you get the horrible jitter effect. The pulse frequency should be once every 20ms or 50hz which isn't actually that fast. 

So it's a slightly different use case for PWM with slightly different requirements. But I'm thinking that the PWM libs I've seen should be able to handle it as long as the timing is accurate enough. (I always thought it was a weird protocol!)

cheers!

Seb 

PS A good diagram on the wiki page https://en.wikipedia.org/wiki/Servo_control#/media/File:Sinais_controle_servomotor.JPG

bennuttall, 2016-04-05:
> Although ServoBlaster introduces another dependancy. If I get a chance, I might plug in my MeArm and try controlling it using RPi.GPIO PWM support.

Or GPIO Zero PWM support? :P

WayneKeenan, 2016-04-05:
same difference isn't it? :)

carlosperate, 2016-04-05:
As far as continuous rotation "hobby" servos go, I've seen them being commercially available for robotic kits and similar things. I've perhaps wrongly inferred from @sebleedelisle comment that most continuous servos are manually modified, but I just wanted to clarify that it seems (at least to me) like a potential use case for this library as well.

The signal control basics are very much the same as with a normal servo, but the end result can be considered different enough where the API design would most likely change. So maybe being able to derive a “servo” and “continuous servo” class from the same parent could be a good aim to have in order to clearly separate the signal control from the end device implementation, as it has been done with other components in this package.

waveform80, 2016-04-05:
Cripes this thread has grown.. Okay, I'll throw some thoughts into the wind:

1. Different classes for different angles: no. That should be a constructor arg just like distances and thresholds are for the DistanceSensor.

2. I think PWM's probably the way to go (the only way?), but @lurch has a point that software PWM is (as I understand it) too jittery for servos on the Pi. That said, the Pi does support hardware PWM and not just on one pin! I know the "real" PWM is only on one pin, or two on later models, but you can do extremely accurate PWM (good enough for servos) on any pin via the DMA controller - both [RPIO](https://pythonhosted.org/RPIO/pwm_py.html#rpio-pwm-servo) and [pigpiod](http://abyz.co.uk/rpi/pigpio/cif.html#gpioServo) implement this. The major downside is that it requires root (because the DMA controller isn't exposed to non-root users, for good reasons) although that's a relatively minor downside in the case of pigpiod where only the daemon needs to run as root and the script can simply request DMA'd PWM via a socket. RPi.GPIO's PWM is entirely software (so probably not good enough for Servo control on slower / loaded Pi's but it doesn't require root).

3. The angle limits are an interesting idea - I'll bear that in mind.

Sorry, I've just skimmed all the messages here - head buried in unit-testing at the moment...

waveform80, 2016-04-05:
Oh, and @WayneKeenan  - not *exactly* the same difference - RPi.GPIO is indeed the default pin implementation but since 1.1 you can use RPIO or pigpiod as the backend instead for "proper" hardware PWM (although the setup's more involved - haven't written any serious docs about that yet largely because pigpiod isn't properly packaged for Raspbian (yet)). Oh, and you can even use a pure Python implementation but that doesn't support PWM so it's a bit of a moot point for this thread.

lurch, 2016-04-05:
So, if you try to create a Servo object using a non-hardware-DMA-able pin (i.e. a NativePin or RPiGPIOPin), should Servo's `__init__` method raise an Exception?

bennuttall, 2016-04-05:
> Different classes for different angles: no.

Ok sure. Updated my example above.

I think there needs to be a base PWM class, which this and `PWMOutputDevice` should extend. Any thoughts?

I've got an idea how the `Servo` class might work, so I'll try to throw something together. Could do with some more servos to test though.

sebleedelisle, 2016-04-05:
I guess a Servo could be the same as an ordinary PWM output as long as the
frequency was 50hz and the pulse width could be accurately adjusted between
5% and 10% (which would equate to the 1-2ms pulse mentioned earlier). If
I've done my maths right! :)

waveform80, 2016-04-05:
You can easily do 50Hz PWM with DMA on the Pi (in fact, it's damn near millisecond accurate looking at [RPIO's oscilloscope pics](https://pythonhosted.org/RPIO/pwm_py.html#example-with-oscilloscope)) but with software PWM (i.e. straight-forward bit-banging from a userspace process like that implemented by RPi.GPIO) you can't *guarantee* anything. While it might be fine most of the time, you can't be certain that something isn't going steal all your process' clock cycles for the next second or two (even with a fully preemptable kernel - can't remember exactly what Raspbian's defaults are in this area though).

In other words, if you're okay with your PWM occasionally going out to lunch and letting your servo go wild, then software PWM *might* be acceptable. But if you need to ensure your servo does exactly what you request, then you *have* to use some form of hardware assistance (whether that's a proper PWM implementation, which the Pi does have but only on a very limited set of pins, or programming the DMA controller to do bit-banging on the GPIO registers for you).

bennuttall, 2016-04-05:
It works well enough to send a servo to the right position. However, I noticed when setting mine to 90 degrees, it would twitch. This could be resolved by disengaging it once it's moved (I'm not sure how we'd time this if we did it automatically, but that could be down to the user).

sebleedelisle, 2016-04-05:
Yeah the twitching is almost definitely to do with poor timing on the
pulse, which happens often with software PWM. Or else it could be just a
really bad servo :)

bennuttall, 2016-04-05:
Likely both :)

sebleedelisle, 2016-04-05:
Do you have decent PWM on gpiozero? Or is it all done in python?

waveform80, 2016-04-05:
@bennuttall now try it on a slower Pi (like a model 1) and fire up some simultaneous threads - it'll only get worse :)

@sebleedelisle here's a rough overview of the way things currently stand:

gpiozero (which is pure python) relies on "pin libraries" to handle controlling GPIO pins (for everything from simple high/low settings up to PWM). Currently four "pin libraries" are supported and gpiozero tries to import them in the following preference order:

1. RPi.GPIO - this is the default library largely because it's extremely popular in the community, it's properly packaged on Raspbian, it's well maintained and is kept up to date with new model releases. It's written entirely in C and supports most operations by bit-banging on an mmap of the GPIO registers (edge detection is handled by epoll on sysfs files). It supports rootless control of the GPIO pins by mapping /dev/gpiomem in preference to /dev/mem. However, its PWM implementation is pure software: threads bit-banging on the GPIO registers with nanosleep for timing (in other words it's really not good enough for servo control).

2. RPIO - this is the initial fallback library largely because it's a drop-in replacement for RPi.GPIO (mostly) so it was the second one developed. It's packaged on PyPI (pip), but not on Raspbian and unfortunately at the moment it's fallen behind a bit in as much as it doesn't support the Pi 2 or 3, just the Pi 1 (and zero?). It's also root only (doesn't know about /dev/gpiomem). However, its PWM implementation uses the DMA controller to bit-bang on the GPIO registers so it's *very* accurate (see the oscilloscope images linked to earlier) and certainly good enough for servos.

3. pigpiod - this is the next fallback library. I'd love it to be the primary library (for reasons we'll get to), but there's a few obstacles to overcome first. Firstly it's not packaged at all (no PyPI, no .deb) so installation is below par. Its architecture is unusual: it uses a root daemon (written entirely in C) to control the GPIO pins and the Python library just talks to the daemon over a socket. On the downside this means you're subject to network latency, but over localhost this is minimal to the point it really doesn't matter (in fact, even remotely over Ethernet it's still fast enough to accurately do things like time a capacitor charging with an LDR, i.e. LightSensor). On the upside, this gives all scripts reasonably safe, rootless control of everything the daemon implements. And the daemon implements *everything* [and the kitchen sink](http://abyz.co.uk/rpi/pigpio/python.html): straight GPIO input/output, high/low, PWM via DMA, SPI, I2C, you name it. It does have a few flaws: the daemon always forks (no foreground mode which makes testing a pain), I've managed to get it into unusual states which required a restart on a couple of occasions, and the Python module looks like copy'n'pasted C (urgh) but forget all that: it's a great architecture and it's something I'd love to see given some attention.

4. native - this is the final "everything else has failed" fallback library. It's a brutally simple pure Python library which I whipped up in a couple of days after reading the source of all the aforementioned libraries. It's similar in architecture to RPi.GPIO and RPIO (bit-banging on an mmap), it's /dev/gpiomem aware (so it supports rootless control), but it's got some pretty big drawbacks at the moment. The primary one for this thread is that there's no PWM support (pure Python's way too slow for PWM via bit-banging and I'm not terribly interested in a root-only DMA implementation). The other is that edge detection is via the GPIO registers which causes a kernel module distress (although it works nicely otherwise). Still, it works even when nothing else is installed and even works happily under things like pypy so it's a good enough backstop for now.

See the [pins documentation](http://gpiozero.readthedocs.org/en/latest/api_pins.html) for a rough overview of all the above, along with detail of the pin interface gpiozero expects (and source links to all the above libraries), and instructions on overriding the default pin implementation.

So, that's how things stand right now. For the near future, RPi.GPIO is almost certain to remain the default library (especially as it's installed by default on Raspbian). For the future, I'd love to see pigpiod given some attention as I think it's the most promising architecture (the fact it exposes nearly all GPIO functionality over sockets is also a potential boon to other languages) but that's largely dependent on how much free time people have (I really ought to spend some more time on picamera which is long overdue a release, then there's all the picademy worksheets that need some work, etc. etc.)

sebleedelisle, 2016-04-06:
@waveform80 thanks for the info! I've used similar things in node, including the node wrapper for pigpio but very little experience with Python on the RPi so this was really helpful.

bennuttall, 2016-04-06:
Ok here's some code that *works* but isn't perfect:

```python
class Servo(PWMOutputDevice):
    def __init__(self, pin=None, active_high=True, initial_value=0, frequency=100, max_angle=180, minimum=0.033, maximum=0.254):
        super(Servo, self).__init__(pin, active_high, initial_value, frequency)
        self._max_angle = max_angle
        self._minimum = minimum
        self._maximum = maximum
        self._angle = None  # don't know what it is on init

    @property
    def angle(self):
        return self._angle

    @angle.setter
    def angle(self, value):
        if 0 > value > self._max_angle:
            raise OutputDeviceBadValue("angle must be between 0 and %s" % self._max_angle)
        self._angle = value
        self.value = ((self._maximum - self._minimum) * value / self._max_angle) + self._minimum
```

I want to create a new base class, rather than letting this extend `PWMOutputDevice`, and make `off()` disengage the servo. I don't think `on()` is necessary. I want `value` to contain a value 0 to 1 mapped to the angle potential, rather than the duty cycle.

waveform80, 2016-04-06:
Any particular reason this shouldn't extend PWMOutputDevice? You could simply make on a method equivalent to setting the maximum angle (which would be consistent with the fact value will be 1 which is the result of calling on in other PWM devices). If you *really* want to get rid of on, Python is unusual in that you can remove methods from sub-classes with a little magic, but it's unusual enough that I'd generally discourage it (simply because people won't expect it).

bennuttall, 2016-04-06:
Good point. I'll try that! on = 1 and off = 0 makes sense.

bennuttall, 2016-04-06:
Working on [servo branch](https://github.com/RPi-Distro/python-gpiozero/tree/servo)

bennuttall, 2016-04-07:
Have pushed a second commit, it's a decent start. I'm not sure how I should deal with `initial_value`. Ideally I think it would be good to allow an `initial_value` or an `initial_angle`, and a fallback.

Also I'm storing the set angle in `self._angle` and returning it when the property is read, but that's not how we generally do things (they're usually read from the device itself). The `value` is read from the actual duty cycle, which means it can be slightly off what it was set to as the servo jitters. But other than that it works pretty much as I designed above.

lurch, 2016-04-07:
If you have a servo that's capable of 0-180 degrees, and you're using it in a robot where (due to mechanical restraints) you only want the servo to be able to move between e.g. 45 and 150 degrees, how would that fit in with the current API-design? Does the user need to remember to only set values between 45 and 150, or could they map 45 to the min_angle and 150 to the max_angle and then set values between 0 (for 45 degrees) and 1 (for 150 degrees) ?

lurch, 2016-04-07:
Would a servo *ever* be active_low?? And given the strict timings mandated by servos, perhaps frequency should't be an init-param either?

WayneKeenan, 2016-04-07:
If `off()` disengages the servo then should `on()` re-engage the servo at `self._angle` or `self._maximum` or `1`?

I'd say the path of least surprise would be `on()` uses `self._angle`; this way the device the servo is mounted in, e.g. robot arm, would remain in the same position.

To remove all doubt perhaps add `attach()`, `detach()` and `attached()`  ? 
http://www.arduino.cc/en/Reference/Servo

WayneKeenan, 2016-04-07:
@waveform80 thanks for the info, very enlightening.  pigpiod looks very impressive but as you mention it could do with some packaging and Python API love; I may take that baton up as part of something I'm building, if that gets anywhere I can let you know.

WayneKeenan, 2016-04-07:
What is the philosophy of gpiozero (optionally) using additional hardware support, specifically, additional add-ons/HAT's for reading ADC inputs?

I only ask in this case because it's possible to perform user modifications of servos and buy off the shelf servos that can output analog positional feedback, for example https://www.adafruit.com/products/1404

WayneKeenan, 2016-04-07:
The Servo constructor could accept 2 optional parameters:

1. a `servoPositionSensor` parameter that is an object that can provide an angular (or 'ANGLE_UNKNOWN') reading.  The type passed would inherit from an interface class that has `getAngle` method. The implementation may read an ADC value or be some other sensor type like an optical sensor that can only reliably detect min and max positions (returning ANGLE_UNKNOWN for all ranges in between)

2. 
A `servoPostionSetter`  (rubbish name) parameter that would default to the PWMOutputDevice (so drop inheritance and use composition instead).  This way the `servoPostionSetter ` would be a object that implements an interface that provides (at least) setAngle.  This could support creating an implementation that instead of using PWM could use i2c to support something like: https://www.adafruit.com/product/815

lurch, 2016-04-07:
> or be some other sensor type like an optical sensor that can only reliably detect min and max positions

I guess something like that would also be useful for stepper motors #144

bennuttall, 2016-04-07:
> If you have a servo that's capable of 0-180 degrees, and you're using it in a robot where (due to mechanical restraints) you only want the servo to be able to move between e.g. 45 and 150 degrees, how would that fit in with the current API-design?

That's possible, but not perfect.

Let's say duty cycle of 0.1 is 0 degrees, duty cycle of 0.2 is 180 degrees they want to restrict to 45 degrees - 135 degrees 0. That means a duty cycle of 0.125 sets it to 45 degrees and 0.175 to 135 degrees. They could configure it like so:

```python
servo = Servo(pin=17, max_angle=90, min_duty_cycle=0.125, max_duty_cycle=0.175))
```

So they actually end up with a 90 degree servo, where value 0 is position 45 of the full potential, and value 1 is position 135.

Do you think we should instead provide `min_angle` or even `rotation_range`?

bennuttall, 2016-04-07:
> Would a servo ever be active_low??

I don't know. However, I noticed that my servo rotates anti-clockwise (so turning from 0 degrees to 180 degrees is anti-clockwise), which is the opposite of what I'd expect. I could swap this around if others agree. Seems arbitrary as it's getting this from the duty cycle (higher duty cycle = higher angle).

bennuttall, 2016-04-07:
> If off() disengages the servo then should on() re-engage the servo at self._angle or self._maximum or 1?

`off()` just stablises it, to stop it jittering. There's no need to re-engage it, just give it another instruction. However, perhaps `on()` should send it to the max angle?

bennuttall, 2016-04-07:
> What is the philosophy of gpiozero (optionally) using additional hardware support, specifically, additional add-ons/HAT's for reading ADC inputs?

Mostly ok, see what's been added to boards.py (or look at Boards page in docs)

lurch, 2016-04-07:
> They could configure it like so: ... So they actually end up with a 90 degree servo, where value 0 is position 45 of the full potential, and value 1 is position 135.

Yeah, sounds reasonable.

Once Servo gets added, I can imagine that a Recipe where a potentiometer controls the angle of a servo, with the servo's current duty_cycle being printed to the console, would be *very* useful :)   (for determining these min and max dutycycle values)

> > Would a servo ever be active_low??

> I don't know. However, I noticed that my servo rotates anti-clockwise (so turning from 0 degrees to 180 degrees is anti-clockwise), which is the opposite of what I'd expect.

Ahh, that's interesting - I hadn't considered overloading the meaning of active_high to mean which direction the servo rotations should be interpreted as... perhaps that should be a different parameter name, rather than re-using active_high (which I'd been interpreting in the common-anode / common-cathode LED sense) ?
(IIRC from my A-level maths, in polar coordinates increasing angles are anti-clockwise)

> off() just stablises it, to stop it jittering. There's no need to re-engage it, just give it another instruction. However, perhaps on() should send it to the max angle?

I think people would definitely expect on() and off() to be functional opposites of each other.

bennuttall, 2016-04-07:
> I think people would definitely expect on() and off() to be functional opposites of each other.

Good point. How about:

- `on()`: go to max angle
- `off()`: go to 0
- `disengage()`: stop twitching

I don't think `engage()` is necessary.

sebleedelisle, 2016-04-07:
on and off to go from max to min seems crazy unintuitive to me! To me, on
and off are states of engagement / power, not an angle.

bennuttall, 2016-04-07:
> on and off to go from max to min seems crazy unintuitive to me

I only say this because `on()` and `off()` for LED is high and low; for PWMLED are high and low; for PWM device are high and low. The behaviour matches these.

WayneKeenan, 2016-04-07:
Perhaps `detach` and `attach` need to be added to break cleanly from the nomenclature of `on` and `off` for PWM pins.  To me `on()` and `off()` indicate the presence or absence respectively of a PWM signal on a pin and not the power/dutycycle so therefore not the resulting angle.

carlosperate, 2016-04-07:
I agree with @sebleedelisle on this one, having `on` and `off` change the position of the servo would be quite unintuitive for me as well. The entire concept of having an `on`/`off` method that does not actually physically turn the servo on or off would seem a bit odd (this type of device I would probably associate with powering it on and off, but I guess that removing the control signal could also be a good interpretation) .

Of course, we should have something to ensure the servo doesn't twitch, and while using the keyword "off" could be sensible enough, I do also agree that "disengage" or "detach" would be more appropriate. For all cases, the opposite keywords, does not seem to add any value at first, but we need to take in consideration that servos can have their position changed due to external intervention, so we need to decide if a "on"/"engage"/"attach" function would try to enforce the last set angle.

lurch, 2016-04-07:
> disengage(): stop twitching

I believe if the signal is 'clean' enough, there shouldn't *be* any twitching.

For 'traditional' PWM devices, on() represents 'fully-on' i.e. a 100% duty-cycle, and off() represents 'fully-off' i.e. a 0% duty cycle (and at these two extremes, are analogous to a Digital output device). But I'm not sure mapping 'fully-on' to a servo-angle of 180° and 'fully-off' to a servo-angle of 0° makes the same semantic sense?

Although for continuous-rotation servos, perhaps using off() for stopped and on() for full-speed-forwards *would* make sense...
Hmmm, should [Motor](http://gpiozero.readthedocs.org/en/latest/api_output.html#motor) have on() and off() functions?

carlosperate, 2016-04-07:
> I believe if the signal is 'clean' enough, there shouldn't be any twitching.

You can always get noise on the signal, cheap servos, etc.

> Although for continuous-rotation servos, perhaps using off() for stopped and on() for full-speed-forwards would make sense...

Maybe, but it doesn't really fit the paradigm all that well either, as you can also go backwards at full speed.

lurch, 2016-04-07:
> so we need to decide if a "on"/"engage"/"attach" function would try to enforce the last set angle.

I believe that would only be possible using a modified servo, like the on that @WayneKeenan [linked to earlier](https://github.com/RPi-Distro/python-gpiozero/issues/248#issuecomment-206711234). For regular servos, there's no way to know what angle they've been moved to, so all we can do is drive them to a specific angle.

carlosperate, 2016-04-07:
> For regular servos, there's no way to know what angle they've been moved to, so all we can do is drive them to a specific angle.

Which would probably have to be saved as a member variable to the class. Which would make sense to do in order to be able to to implement offsets as well rather than only absolute values.

lurch, 2016-04-07:
> I don't think engage() is necessary.

I'm sure https://www.raspberrypi.org/blog/author/robzwetsloot/ would disagree with you! ;-D

lurch, 2016-04-07:
> Which would probably have to be saved as a member variable to the class.

https://github.com/RPi-Distro/python-gpiozero/blob/servo/gpiozero/output_devices.py#L895

lurch, 2016-04-07:
Hmm, thinking a bit more and I guess @WayneKeenan 's idea of passing a specific `servoPostionSetter` object to the `Servo` constructor, is a very similar idea to passing a specific `Pin` object to the `GPIODevice` constructor.

waveform80, 2016-04-07:
I should probably wade into this mammoth thread before I get too far behind. Speaking as someone who's much more software-guy than hardware-hacker (and who's playing catch-up half the time while coding things in gpiozero :) here's what I've gleaned an implementation ought to have so far:

* We obviously need some constructor parameters specifying the minimum and maximum angle that the servo is physically capable of which, absent any other parameters, would correspond to values of 0 to 1 in the `value` property (perhaps -1 to +1? Might depend on whether we deal with negative angles or not).

* It seems a common requirement is to limit a servo's motion to a particular range (to ensure it doesn't bash into fixed objects like a casing). These ought to be additional constructor parameters (which if unspecified would match the minimum and maximum range, the latter of which would be mandatory parameters).

* There's concern about servo twitching: from my understanding this is largely down to software PWM. I've heard opposing claims on this from various sources, and not being a hardware expert I don't know what to believe. Can we have some verification that a hardware-assisted PWM implementation (like RPIO or pigpiod) definitely still twitches in some configurations (cheap servos or whatever)?

* Admittedly hardware-assisted PWM currently isn't a choice for people not willing to endure some installation pain, so that probably needs addressing *before* we merge a Servo class (i.e. get pigpiod packaged properly, or perhaps throw some effort into getting RPIO compatible with Pi2/3 - not sure which is the bigger effort but both support hardware-assisted PWM).

* On the subject of disengaging the servo (which I'm assuming means stopping the PWM entirely until re-engaged), is this *purely* to avoid twitching or are there other use-cases?

* I'm coming around to the position that having on/off corresponding to max/min value (as it does in other classes) is unintuitive in the case of the Servo class. While it would be different to everything else in the hierarchy, the only reason (that I've thought of so far) for having on/off equate to value 1/0 would be to ensure that a routine passed an arbitrary device knew what to expect when calling on/off... But what use-cases are there for a routine being passed an arbitrary device? I'm struggling to think of one.

* A more interesting question is what `value` does the servo have when it's disengaged. Does it return the last `value` it was at and we assume that it hasn't moved from that position, or do we return a value explicitly denoting that it's disengaged (like `None`). I'm tempted by the latter simply because it affords a means of disengaging (and re-engaging) the servo while using a source/values style of coding.

lurch, 2016-04-07:
> On the subject of disengaging the servo (which I'm assuming means stopping the PWM entirely until re-engaged), is this purely to avoid twitching or are there other use-cases?

I suspect (but can't check right now) that if a servo isn't being sent a PWM signal, then it isn't being "driven" to a specific angle, and so could be moved by hand (subject to the resistance in the gearing) to a specific angle. But unless you're using one of the position-sensor servos that @WayneKeenan pointed out earlier, this probably has limited use?

> While it would be different to everything else in the hierarchy

As I already pointed out, Motor doesn't have on() or off()

> I'm tempted by the latter simply because it affords a means of disengaging (and re-engaging) the servo while using a source/values style of coding.

But IIRC this would be the first time any device returns a `value` that isn't a number in the range -1 to 1 inclusive?

Dunno if it helps, but here's another (lower-level) API-design for a Servo interface (to go along with the Arduino interface pointed out earlier) http://docs.micropython.org/en/latest/pyboard/library/pyb.Servo.html
Interestingly, it has no detach()-like function.

SteveAmor, 2016-05-06:
Hi All, as mentioned in #314 I've been programming servos and speed controllers, in assembly, for drones for the past 5 years. Please bear with my observations from an OOP point of view.

I don't like angles for two reasons.  One, if it's an int, you don't get enough resolution.  Secondly, they are not accurate.  It depends on the mechanical build of the servo.  I see Arduino has writeMilliseconds() which gets around the resolution issue if needed.

I don't particularly have a problem with on() and off(). However, I would say that off() need to set the PWM mark space ratio to 0%.  And on() needs to set the servo to the defined midpoint (usually 1.5msec pulse).

There are analogue and digital servos.  Analogue servos work up to about 50Hz. Digital servos can run up to a PWM rate of 400Hz.  An analogue servo will burn out if you run it at 400Hz.  The higher refresh rates of a digital servo equates to more torque (within the limits of the design of the servo).

If there is no jitter on the PWM signal, the servo will not "twitch".  If you set the PWM mark space ratio to 0%, i.e. servo.off(), the servo will consume minimal power but will not have any torque.  Generally, this is not an issue and is desired.  If a servo is off(), I would suggest the value returned is none.

By default set min = 1msec, max = 2msec, 50Hz, start position = 1.5msec.

When using a rotational servo, off() with 0% PWM duty cycle is very useful to ensure the servo really does stop rotating.  It's quite difficult to send the exact PWM to keep it steady.

I'm happy to test with other gpio libraries to test for jitter but don't know how to at the moment.

In the future, you may want to consider supporting Smart Servos and S Bus servos which use a serial protocol enabling either feedback for the position or a 1 wire type bus for servos.

lurch, 2016-05-07:
> I don't like angles for two reasons. One, if it's an int, you don't get enough resolution. Secondly, they are not accurate. It depends on the mechanical build of the servo. I see Arduino has writeMilliseconds() which gets around the resolution issue if needed.

Like the rest of gpiozero, I'm sure the angle would be a float rather than an int. (In python you actually have to go out of your way to specifically restrict something to being only an int).
Perhaps the Servo class should have a `value` property in the range 0->1, a corresponding `angle` property in the range 0->maxangle (in degrees) and *also* a `pulsewidth` property specifying the value in milliseconds? (And each of these properties are simply different interpretations of the same internal state). However with both value and angle being floats, I'm unconvinced of the need for being able to specify a milliseconds pulse-width too.

> I don't particularly have a problem with on() and off(). However, I would say that off() need to set the PWM mark space ratio to 0%. And on() needs to set the servo to the defined midpoint (usually 1.5msec pulse).

I've just read through this entire thread again, and my current opinion is that calling `off()` should set the duty-cycle to 0%, and then in the off-state querying `value` or `angle` would both return `None`.

**If** we decide to also add an `on()` function, then I think `off()` should also save the current value to e.g. `self._last_angle`, and then `on()` should set the servo to `self._last_angle`, and set `self._last_angle` to None. And if being used with a servo that has a position-sensor, then `on()` should read the servo's current angle, and drive the servo to that same angle. And maybe `on()` could also have optional keyword-only `value` and `angle` parameters (with an Exception raised if you try to use both at once)? So that you could e.g. do:
```python
servo.angle = 45 # will drive servo to 45
servo.off() # 'disengage' servo
servo.on() # will drive servo to 45 again
servo.angle = 90 # will drive servo to 90
servo.off() # 'disengage' servo
servo.on(angle=130) # will drive servo to 130
servo.off() # 'disengage' servo
servo.angle = 150 # will drive servo to 150
servo.close() # 'disengage' servo
```

>  I'm not sure how I should deal with initial_value. Ideally I think it would be good to allow an initial_value or an initial_angle, and a fallback.

Perhaps for the Servo class, we should make the initial_value / initial_angle non-optional? Or perhaps constructing a Servo with no initial_value or initial_angle should leave it in the off-state, until a value/angle is explicitly set (so constructing a servo with no initial_value or initial_angle, and then calling on(), would raise an Exception).

> Nice to have : a moveTo or similar that interpolates from the current position to the target position with a specified duration, and ease in and ease out (either a linear ramp up and down or an [...]

I already started on something very similar to this in #310
Currently it just does linear ramps, but I'll think about adding exponential ramps too (via an optional ramp='linear' / ramp='exponential' parameter) :-)

SteveAmor, 2016-05-08:
@lurch I believe we're of the same thinking :-)
Two things I'm grappling with related to the max and min values.  Most, if not all, servos should respond to a pulse width from 1000 to 2000usecs (free to move, on the bench, with no attachments causing mechanical limitations). 1500usecs will be the centre (let's call this 90 degrees). 

First grapple, you can't relate a pulse width to a true angle offset from you 90 degree centre point (it depends on the internal gearing of the servo).  How nice it would be for 1000usec to be "0 deg" and 2000usec to be "180 deg" (with reference to 1500usec being 90deg). 

Second grapple. Most servos will respond to a pulse width outside the 1000usec to 2000usec range. The user should be free to use this extended range. So what happens to the abstracted values if they are in the range of 0 to +1 and the user extends the pulse width from 900usec to 2250usec for example?

[36 more replies]

commit 02f7d20b (2016-08-30): Fix #248

Add Servo and AngularServo implementation along with docs and tests.
This is a deliberately minimal implementation designed to be added to as
we agree on new extensions (better than making an all-singing,
all-dancing version in which I get things wrong and then wind up making
backward incompatible changes to get it right :)
commit c343c737 (2016-08-30): Merge pull request #416 from waveform80/servo

Fix #248
closed 2016-08-30 by waveform80