gpiozero/gpiozero #254: DHT

bennuttall, 2016-04-06:
We should look at adding support for DHTs like the DHT11 and DHT22.

There's an Adafruit library which is pretty poor, and requires sudo. There was talk of a kernel module providing a file interface, but I'm not sure where that went. Perhaps, as it's one-wire, this could work similarly to the Temperature Sensor (#93).

zleap, 2017-02-24:
I agree on this, support for the dht11 would be most welcome.  I have tried the Adafruit library, it worked once, then stopped working,  or refused to work on another pi.

joan2937, 2017-02-25:
The DHT11/22 etc. are so called one wire devices but they use a proprietary protocol and are not compatible with the (Dallas) 1-wire bus.

I have had little joy with the kernel modules (I'm not sure if there are one or two, the last I tried was iio based).  Strangely the userland solutions are much more reliable.

See http://abyz.co.uk/rpi/pigpio/examples.html#pdif2_DHTXXD for a pigpio daemon based solution.

bootsmann, 2017-05-19:
Is there something in progress about DHT22 sensor after one year?

msa7, 2018-11-03:
any news here?

bennuttall, 2019-12-11:
If anyone wants to find some code that'll read from a DHT, that'll help.

joan2937, 2019-12-11:
@bennuttall You may not find any code suitable for gpiozero.  The DHTxx series requires accurate timing of GPIO level changes and the method used will be specific to each back-end.

This is my pigpio offering.  http://abyz.me.uk/rpi/pigpio/examples.html#Python_code/DHT.py

bennuttall, 2019-12-11:
I thought as much - I used the Adafruit driver in the past, and it's ok but slow. I've had issues using the kernel driver, and that doesn't seem to be going anywhere.

DarrenTownsend, 2020-04-26:
Just found this, which might be useful:

https://github.com/szazo/DHT11_Python