gpiozero/gpiozero #255: led.off time delay

BrianDurward, 2016-04-06:
I'm a nooby so apologies for probably asking the obvious. 
My Python programme turns on an LED using  - button.when_pressed = led.on
I then want the LED to remain lit for 10 seconds or until the switch is activated a second time.
I have tried different options with no success.
Help please.

bennuttall, 2016-04-06:
Ok sure, what you need to do is create a custom function to light the LED for 10 seconds, then set `when_pressed` to the custom function, rather than `led.on`:

```python
from gpiozero import LED, Button
from time import sleep
from signal import pause

led = LED(2)
btn = Button(3)

def light():
    led.on()
    sleep(10)
    led.off()

btn.when_pressed = light

pause()
```

I'm not sure what your exact requirements for the "or until the switch is activated a second time". Do you want it to start the 10 seconds again, if pressed after 5 seconds? I'm not sure how best to interrupt the running of `light` with another button press. @waveform80, any ideas?

Alternatively to `led.on(); sleep(10); led.off()` you could use `led.blink(10, 0, 1)` (on for 10 seconds, run once).

waveform80, 2016-04-06:
```python
from gpiozero import LED, Button
from time import sleep
from signal import pause

led = LED(4) # or whatever pin you've got it on
btn = Button(17) # again, adjust as necessary

def light_for_ten_seconds():
    led.on()
    sleep(10)
    led.off()

btn.when_pressed = light_for_ten_seconds
pause()
```

That's one way, but another is to use a "limited" blink:

```python
from gpiozero import LED, Button
from time import sleep
from signal import pause

led = LED(4) # or whatever pin you've got it on
btn = Button(17) # again, adjust as necessary

def light_for_ten_seconds():
    # the n parameter limits the number of blinks to 1
    led.blink(on_time=10, off_time=1, n=1)

btn.when_pressed = light_for_ten_seconds
pause()
```

Hmm, let's see how else we could do this ... how about a source/values version:

```python
from gpiozero import LED, Button
from time import time
from signal import pause

led = LED(4) # or whatever pin you've got it on
btn = Button(17) # again, adjust as necessary

def stretched(values):
    # Stretches any 1s (button presses) out for 10 seconds
    pressed = 0
    for value in values:
        if value:
            pressed = time()
        yield time() - pressed <= 10

led.source = stretched(btn.values)
pause()
```

These last two recipes have the advantage that button presses during the LED being lit will extend it for a further 10 seconds from the press, whereas the first won't because the handler doesn't return until 10 seconds have elapsed.

Now, let's pimp out the last one:

```python
from __future__ import division # in case you're on py2
from gpiozero import PWMLED, Button
from time import time
from signal import pause

led = PWMLED(4) # or whatever pin you've got it on
btn = Button(17) # again, adjust as necessary

def stretched(values):
    # Stretches any 1s (button presses) out for 10 seconds
    pressed = 0
    for value in values:
        if value:
            pressed = time()
        # fade out after the press
        yield max(0, 10 - (time() - pressed)) / 10

led.source = stretched(btn.values)
pause()
```

waveform80, 2016-04-06:
Doh - @bennuttall beat me to half those ideas ;)

lurch, 2016-04-06:
Here's a (slightly more complex) version of something similar I came up with the other day http://raspi.tv/2015/gpio-zero-test-drive-making-light-of-security#comment-60978
Apparently (according to the reply) it doesn't work, but I haven't yet got round to testing / debugging it myself.

But (unsurprisingly) @waveform80 's `stretched` approach looks like a much neater solution ;)

bennuttall, 2016-04-06:
Neat solution @waveform80

martinohanlon, 2016-04-06:
I like @waveform80 's long blink!

waveform80, 2016-04-06:
To be honest, that last "fading" version is probably also more easily done with a limited blink but I had to work some source/values stuff into there somewhere :)

BrianDurward, 2016-04-06:
Many thanks to all for their quick and helpful responses.
So here's the scenario.  The RPi will be operating outdoors, powered by a power pack and not connected to the internet.
The switch will be a vibration detecting switch located on the top of a wooden upright.  When the upright is contacted either forcibly or very lightly by a ball the switch will bring on an array of 4 LEDS wired in parallel (powered separately from the RPi) and switched by a transistor.   I am happy to take guesses regarding the sport!  After 10 seconds the lights will go off and the program continue as before (ie sensitive to vibrations).    I thought I would put a regular switch in parallel with the vibration switch to turn off the LEDs before the 10 seconds were up (it might be easier to just wait 10 seconds).  I will try all of the submitted ideas and feedback in due course.  BD

lurch, 2016-04-06:
I don't think a "regular switch in parallel" would work to turn the LEDs off again - even if you did get the code working to do that, you'd probably find that the vibration sensor triggers many times for each "hit" (while it's still vibrating), and so the sensor itself would cause the LEDs to switch off again!

I think if you wired up your regular switch to an additional GPIO on the Pi, then something like this might work:
```python
from gpiozero import LED, Button
from time import time
from signal import pause

led = LED(4) # or whatever pin you've got it on
sensor = Button(17) # again, adjust as necessary
btn = Button(18) # again, adjust as necessary
pressed = 0

def stretched(values):
    # Stretches any 1s (sensor readings) out for 10 seconds
    global pressed
    for value in values:
        if value:
            pressed = time()
        yield time() - pressed <= 10

def lights_off():
    # Stop the stretched function from lighting the LED
    global pressed
    pressed = 0

btn.when_pressed = lights_off
led.source = stretched(sensor.values)
pause()
```

Obviously you'd need to make sure that the button attached to GPIO18 is only pressed after you're sure your sensor has stopped vibrating.

waveform80, 2016-04-06:
Just because I hate globals :)

```python
from gpiozero import LED, Button
from time import time
from signal import pause

led = LED(4) # or whatever pin you've got it on
sensor = Button(17) # again, adjust as necessary
inhibit = Button(18) # again, adjust as necessary

def stretched(sensor_values, inhibit_values):
    # Stretches any 1s (sensor readings) out for 10 seconds
    for sensor_value, inhibit_value in zip(sensor_values, inhibit_values):
        if inhibit_value:
            pressed = 0
        elif sensor_value:
            pressed = time()
        yield time() - pressed <= 10

led.source = stretched(sensor.values, inhibit.values)
pause()
```

Incidentally that'll only work in Python 3; Python 2's zip doesn't do lazy evaluation so it'll try and enumerate the two infinite iterators ... rather unsuccessfully. If you need it to work in Python 2 bung the following at the top:

```python
try:
    # Make Python 2's zip like Python 3's
    from itertools import izip as zip
except ImportError:
    # We're on Python 3 so nothing to do
    pass
```

Zip is a hugely useful function with source/values stuff as it lets you combine as many sources together as you like in a single function and synchronously read things from each of them.

martinohanlon, 2016-04-06:
@waveform80 I also hate globals.

lurch, 2016-04-06:
Sorry, I was just being lazy :-/
(and I'm not as expert at using all the iterator-based functions as @waveform80 yet)

Does the `for x, y in zip(xiter, yiter):` syntax/notation guarantee that (as near as dammit) the xiter and yiter both get read at the same time? Or if the `xiter` took longer to produce its value, would that result in the `yiter` being read later than the `xiter`? And if that's the case, and you know that you have one iterator that's 'slower' than the other, should you put the slower iterator last in the call to `zip`? (although perhaps at the end of the day it makes little actual difference)

Not sure how well I explained that...

waveform80, 2016-04-07:
Ahh, the thing to remember is that the iterators don't have any pauses in them (by default) - it's the source thread which implements a minor delay between reading values from whatever it's been given. So, zip will indeed read the iterators damned near simultaneously (can't be precisely simultaneously but given that zip is C-based it'll be *very* close). As to what order zip actually reads its stuff in, I'm not 100% sure (haven't delved into Python's source) but I'd assume left to right.

Incidentally I said no pauses by default because you can place pauses in the iterators with the new `pre_delayed` and `post_delayed` functions, but that's just complicating matters :)

lurch, 2016-04-07:
> the thing to remember is that the iterators don't have any pauses in them (by default)

I was thinking specifically about the `slow` iterator (generator?) that you introduced back when I was first asking about generators... ;-)

closed 2016-04-07 by bennuttall

waveform80, 2016-04-07:
Yeah, pre_delayed and post_delayed are variants on that which just position their delay before or after the yield accordingly. However, that's on the generator side of things; the new source_delay thing in 1.2 introduces the delay on the source side which adds a little more flexibility to the system and simplifies things a bit (means you don't have wrap everything you assign to a source for instance).