gpiozero/gpiozero #266: Incorrect documentation?

lurch, 2016-04-08:
According to http://gpiozero.readthedocs.org/en/latest/api_generic.html `RGBLED` inherits from `CompositeDevice`, but looking at the [source](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/output_devices.py#L527), it actually inherits from `Device`?

I'm not sure if its the code or the documentation that's incorrect ;-)  (but I suspect the documentation)

lurch, 2016-04-08:
Found something else which is either a documentation issue, or something that @waveform80 's planning to add to the code but hasn't done so yet...

http://gpiozero.readthedocs.org/en/latest/api_boards.html#gpiozero.LEDBoard describes a `named_pins` parameter, but that doesn't actually seem to exist anywhere in the codebase? https://github.com/RPi-Distro/python-gpiozero/search?q=named_pins&type=Code

waveform80, 2016-04-08:
The document's incorrect - RGBLED (despite being a composite under the covers) is meant to look like a single monolithic device (which is why it doesn't expose it sub-LEDs directly).

And oops - missed **named_pins from the constructor line but it is in the code-base - it's just very very obscure (for reasons I shall explain :). So, LEDBoard has the following constructor:

```python
def __init__(self, *args, **kwargs):
```

However, we've overridden that in the docs (in `docs/api_boards.rst`) and specified it as (I've just added the `**named_pins` bit - the back-slashes are to ensure reST doesn't interpret the asterisks as emphasis tokens):

```rest
.. autoclass:: LEDBoard(\*pins, pwm=False, active_high=True, initial_value=False, \*\*named_pins)
```

So which is it? Well, we'd really like it to be what we've documented but unfortunately it's invalid syntax in Python 2. Python 3 introduced the concept of "keyword-only" args. Given that `pwm`, `active_high`, and `initial_value` all appear after an arbitrary positional `*args` they have to be specified as keyword args (they can be specified positionally). But as Python 2 doesn't have a conception of keyword-only args it doesn't allow this syntax.

The only way to implement this and remain Python 2 compatible is to do `*args, **kwargs` then pop the keyword args we're expecting from the kwargs dict in the constructor. If we didn't also have a `**kwargs` in our prototype we'd then have to check our kwargs dict was empty after popping our expected keywords and raise an exception if it wasn't. You can see an example of this in `LEDBarGraph.__init__` which doesn't allow named LEDs.

lurch, 2016-04-08:
Okay... so where's the kwargs.pop('named_pins') ?
(I suspect I'm probably misunderstanding things again :-/ )

waveform80, 2016-04-08:
Well, it wouldn't be `kwargs.pop('named_pins')` for starters: the `**` element in the prototype becomes a dict that captures all other keyword arguments. Something with the prototype `LEDBoard(*pins, pwm=False, active_high=True, initial_value=False, **named_pins)` would (in Python 3) receive the arguments `pins` (a tuple of all positional argument values), `pwm` (False if the user didn't specify it and whatever they did specify otherwise), `active_high` (similar story to `pwm`), `initial_value` (same story again), and `named_pins` (a dict of all remaining keyword arguments and their values).

However, as discussed above we can't use that prototype in Python 2 so we use a more generic `*args, **kwargs` (i.e. a tuple of all the positional args and a dict of all the keyword args), strip out the stuff we want and pass the remaining keyword args to the super-class.

Why document it as "named_pins" instead of "kwargs"? Well firstly the name is irrelevant - the name of a `**` parameter is entirely for the function's internal use - the user can never specify it because that parameter captures *all* remaining keyword args. So the only use for the name to the user is to suggest its purpose - hence I call it `**named_pins` in the docs to suggest its purpose. I'd like to call it the same in the code, but as mentioned we can't use this precise prototype in Python 2. Instead we have to capture *all* keyword args in that parameter, so calling it `**named_pins` would be a confusing misnomer in this case.

Ultimately, the class that winds up dealing with assigning names to pins (or more precisely, devices) is CompositeDevice in the devices module. This also uses `*args, **kwargs` for the same reasons as above (i.e. I'd like to use `*devices, _order=None, **named_devices` instead ... but I can't in Python 2). So, I'm afraid this all boils down to the fact that supporting Python 2 is still a necessity, at least for libraries at this point in time (as opposed to applications, projects, etc.)

lurch, 2016-04-11:
Okay, I think I see how it all works now, with LEDBoard making use of the kwargs stuff in CompositeDevice. I've got some tidied-up docstrings which I'll hopefully PR in the next day or two.

closed 2016-04-11 by lurch