gpiozero/gpiozero #268: SourceTools questions

lurch, 2016-04-08:
1) IMHO the documentation for  http://gpiozero.readthedocs.org/en/latest/api_source_tools.html#gpiozero.scaled is a bit confusing. After reading it a few times, I *think* what it means is that range min/max is the range of the 'output' values, and domain min-max is the range of the 'input' values? What happens if one of the input values is outside of domain min-max? Does (or should) the output get clamped to range min-max?

2) Given that most of the devices in gpiozero operate on values from 0->1 or -1->1, perhaps there should be two separate utilities (i.e. with the necessary min-max values hardcoded) for scaling back and forth between these two ranges?

3) I guess another useful function (perhaps?) might be one that simply called abs() on each value?

4) Again IMHO the documentation http://gpiozero.readthedocs.org/en/latest/api_source_tools.html#combining-sources for conjunction and disjunction could be a bit clearer. I think if you specifically mentioned AND and OR, then people would have a better grasp of what these functions do without having to click through to wikipedia. (After all, GPIOZero is supposed to be beginner-friendly!)

5 ) There's a couple of common errors in the http://gpiozero.readthedocs.org/en/latest/api_source_tools.html#gpiozero.sin_values and cos_values documentation. Both say "If you require a wider range than 0 to 1, see scaled()." when they actually return -1 -> +1. And both say "result of applying sign to" when I guess it should be 'sine' and 'cosine' respectively instead of 'sign'. Perhaps the documentation should clarify that sin starts at 0 and cos starts at 1? And do the examples need a source_delay or something?

6) Do we even need separate sin and cos values? Perhaps we could have just a single sin function, with an optional start_angle parameter that defaults to 0? (and could be set to 90 to get a cosine wave). I can imagine people wanting to start at sin(-90) too (when they're scaling down to 0->1).

7) Given that we've got a function for getting a sine-wave, perhaps we could also have a function for triangle-waves? (I guess with the same 'frequency' as the sin function) Perhaps with a couple of optional params that could also be used to turn it into a sawtooth wave? (Hmmm, sounds like I'm starting to re-invent the `blink` function again!!)

bennuttall, 2016-04-08:
Wow - great to see source tools land. I wonder if we should move it to `gpiozero.tools` though so only component classes are exposed directly by `gpiozero`? So an example would become:

```python
from gpiozero import LED, Button
from gpiozero.tools import inverted
```

lurch, 2016-04-08:
8) Perhaps `inverted` also needs some min/max args, so that -1 -> +1 values can get inverted to +1 -> -1 without having to go through two scaling operations?

9) Should pre_delayed and post_delayed have the same "> 0" and "is float" checks that you recently added to mixins.py ?

waveform80, 2016-04-08:
@lurch 

1. Ahh ... this was me recalling stuff from my A-level maths days (an unspecified number of decades in the past ... ahem); [domain and range](https://en.wikipedia.org/wiki/Domain_of_a_function) are fairly well understood terms in mathematics but I'm guessing not more widely. In the case of scaled, if you violate the specified domain on input, the function will potentially violate the range on output (i.e. the domain parameters are effectively your guarantee to the function that you won't pass in stuff outside those values, and the range parameters are what you expect from the function assuming you stick to that guarantee). Better docs / name suggestions are welcome!

2. The thought had crossed my mind ... any good ideas on names? I couldn't come up with anything that sounded as nice as "scaled".

3. Hmm, that's a interesting idea - erm .. "absoluted"? "natural"? "positive"? Not sure any of those is particular good...

4. I did ponder calling them "and_gate" and "or_gate" ... but they're not exactly gates. So, once again I fell back on maths (or more precisely, predicate logic). Again - better name suggestions welcome!

5. Doh - need to correct those docs. The default source_delay of 0.01 should be sufficient. Given it'll take 360 readings to complete a cycle than means each cycle should take 3.6 seconds (minimum) by default.

6. That's a much nicer idea than my sin/cos stuff. One thing I was pondering was whether to stick to degrees though. It's friendlier from one perspective (I doubt many kids will know about radians) but on the other hand, all of the stuff in Python's math library expects (quite reasonably) radians, so if they want to combine the two we've just forced them to do a load of conversion nonsense. Perhaps I should just leave the trig out for now?

7. That was more or less where I was heading: a saw-tooth wave, a square wave, whatever other waves people can come up with that might have some vague use somewhere, and then perhaps functions to combine said waves. But then, I was struggling to think of use-cases beyond making police sirens with red&blue LEDs so I thought perhaps I was going over-board and left it at sin/cos for now.

8. Yeah, that's a good idea - any better ideas than domain/range for names?

9. Probably - most of source_tools was bashed together very quickly without a great deal of thought :)

@bennuttall hehe - I was wondering that too, but thought you'd want to keep it simpler! Okay, I'll rename the module and leave it out of the main package imports.

lurch, 2016-04-10:
With the 1.2 release being imminent I won't touch `tools.py` for now, but will leave a few comments for @waveform80 to ponder over...

1. and 8. What about `input_min`, `input_max`, `output_min` and `output_max` ? (with the docstring warning that if any of the `values` go outside of the range input_min->input_max, then the output will go outside of the range of output_min->output_max)

2. I guess in order to come up with reasonable names for the conversion-functions, we possibly need to come up with names for the "0 -> 1 range" and the "-1 -> +1 range". (which may also help with documentation elsewhere?)

3. I guess `absoluted` is as good as anything (and follows the same grammar-pattern as `negated` and `inverted`)

4. Given that they're simply implemented in terms of `any` and `all`, maybe those could hint at suitable names (e.g. `all_values`, `any_values`) ? Although I was only suggesting clarifying the docstrings, rather than renaming them entirely \*shrug\*

5. But isn't it better to be explicit than implicit? If you include an explicit `source_delay` of 0.01, then even if it's the same as the default value, the reader will have a better idea how to slow down or speed up their siren.

6. I think it's worth keeping the sine-wave in there, as it can give a more "natural looking" oscillation than a simple triangle wave (if that gets added). And perhaps you could solve the radians / degrees conundrum by having both `start_degrees` and `start_radians` parameters (both defaulting to None, interpreting it as 0), and simply throwing an error if both of them are non-None?
(although if you specify that start-angle in radians, would you still want 360 "steps"?)

7. Sounds neat :)  I guess re-implementing `blink()` entirely in terms of these '**waveforms**' (LOL!) would be a good testing-ground!

waveform80, 2016-04-10:
Okay, lots of good ideas there. I'm going to incorporate these into the docs reorg I'm about to push, but given I'm still not 100% happy with the way tools.py is (I'm not sure about degrees/radians thing yet) I'm going to mark that module "experimental" (like pins, so no guarantees we're not going to change the API next release) just to give us a little lee-way in fiddling with things next time round.

waveform80, 2016-04-10:
On the subject of names for the 0-1 range, converting data-sets so that they consist of floats in the 0-1 range is frequently called "normalization", so I would be tempted to call that the "normal" range. Unfortunately "normal" has so many other meanings (it has multiple meanings in the maths domain alone, let alone science and other fields) that it'd be rather silly to pile another onto it. We could take a leaf out of vectors and call it the "unit" range perhaps?

waveform80, 2016-04-10:
Huh ... just had a thought on sine/cosine/saw/whatever waves: what you *really* want to specify in the constructor (assuming you want to synchronize it with other things), is the number of values in one cycle of the wave, i.e. the period - never mind degrees and radians and all that. In other words, if I wanted each value to represent another degress I'd specify a period of 360, but equally I could specify a value of 100 to get it to synchronize with something else that took 100 cycles to do whatever it did.

I'll stuff that in quickly while waiting on the last ticket :)

bennuttall, 2016-05-26:
@lurch is your question answered, or does something need to be done before this can be closed?

lurch, 2016-05-26:
I've just gone through all my previous points, and the only ones that haven't been addressed are:

2) Given that most of the devices in gpiozero operate on values from 0->1 or -1->1, perhaps there should be two separate utilities (i.e. with the necessary min-max values hardcoded) for scaling back and forth between these two ranges?

7) Given that we've got a function for getting a sine-wave, perhaps we could also have a function for triangle-waves? (I guess with the same 'frequency' as the sin function) Perhaps with a couple of optional params that could also be used to turn it into a sawtooth wave? (Hmmm, sounds like I'm starting to re-invent the blink function again!!)

But these are both minor points that we can revisit in future should the need arise, so I'll close this issue for now :)

closed 2016-05-26 by lurch