gpiozero/gpiozero #278: Remote pin issue caused by pi_info

bennuttall, 2016-04-13:
I think the pigpio remote pin access is a really interesting prospect. Trying the brilliant demo in the docs of creating an instance of a pin object remotely on a Pi from another machine currently fails when run on a computer that's not a Pi due to the exception raised in `pi_info()`. There needs to be a way of it detecting the return value of `pi_info()` on the host Pi, or it to fall back on a sensible default.

Similarly, will there be an issue if an unknown Pi (future product or unknown revision) is used? Or does the code in pi_info() rely on the host running Raspbian? Would it work on Ubuntu MATE, for example?

lurch, 2016-04-14:
Interesting point, I guess this would also be a problem if you're using e.g. a Rev1 Model B to control the pins on a B+... AFAIK `pi_info()` returns info about the Pi the script is running on, which won't be the case if you're using remote pins via pigpio. Off the top of my head I guess there's no way for pigpio to tell the 'parent script' what revision of Pi the remote pins are being created on, so perhaps there needs to be a 'use at your own risk' static method to override the revision of Pi used within pi_info() etc. ?

And regarding future products, pi_info should be able to 'decode' the new-style revision_info structure https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py#L503
Although looking at the code in detail, it would actually fail if the 'Type' part of the revision-code isn't in the set of {0, 1, 2, 3, 4, 6, 8, 9}. (perhaps that could be fixed? - I guess any unknown fields could simply get set to None)

AFAIK there's no Raspbian-specificness to any of this.

lurch, 2016-04-14:
> perhaps there needs to be a 'use at your own risk' static method to override the revision of Pi used within pi_info() etc. ?

... which is as simple as:
```python
import gpiozero.pins.data
gpiozero.pins.data._PI_REVISION = '0013'
```
to make gpiozero think it's _actually_ running on a Model B+ (even when the script is running on a PC). See https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py#L244 for the list of currently-recognised revision codes.

carlosperate, 2016-04-14:
Perhaps there could be a "not-a-pi" type returned by pi_info to denote this use case?

waveform80, 2016-04-15:
I dug into the pigpio interface while on the bus the other day, and there's a gpio_revision() call one can make. It's not terribly specific in the docs as to what it actually returns, but if you dig into the source, it returns the "Revision" field from `/proc/cpuinfo` (as an integer) which is more or less exactly what we want for looking up the Pi. So, that's an easy way of solving the remote info problem (although a little finesse will be required as remote info will have to be a dictionary of PiInfo instances keyed by connection rather than a single PiInfo connection as there's nothing stopping you from constructing pins on *multiple* remote Pis).

Anyway, I'll try and get this fixed and re-released fairly shortly so Ben can go around demo-ing this and persuading people it's a good architecture which we ought to be seriously thinking about for the future :)

waveform80, 2016-04-15:
Sorry, make that `get_hardware_revision` rather than gpio_revision - when you're digging into the source there's something like that a layer or two down.

closed 2016-04-15 by waveform80

dglaude, 2016-06-05:
I am a bit confused if information found by @waveform80 on April 15 has been fully used to detect the Pi we want to work with (especially for a remote pi).

In file data.py line 617 for function pi_info when revision is set to None it call function _get_pi_revision() from line 492 try to read from local '/proc/cpuinfo'.

Is there a way to use the information that seems to be gathered in line 111 of pigpiod.py ?

The problem I see, is how is the function pi_info know that we try to work remotely...
Should I trigger some pin usage first to get the library to know it?

mac$ **cat gpiozero.pi_info.py** 
```
import gpiozero
info = gpiozero.pi_info()
```
mac$ **python gpiozero.pi_info.py** 
Traceback (most recent call last):
  File "gpiozero.pi_info.py", line 2, in <module>
    info = gpiozero.pi_info()
  File "build/bdist.macosx-10.11-intel/egg/gpiozero/pins/data.py", line 640, in pi_info
gpiozero.exc.PinUnknownPi: unknown RPi revision "unknown"

This is the documentation I found about hardware revision in pigpio:
http://abyz.co.uk/rpi/pigpio/python.html#get_hardware_revision

There might be a trick however as I don't have a match between this test code (remotely):

mac$ **cat pigpio.get_hardware_revision.py** 
```
#!/usr/bin/env python
import pigpio
pi = pigpio.pi()
print pi.get_hardware_revision()
pi.stop()

```
mac$ **python pigpio.get_hardware_revision.py** 
9437331

And this CLI done on the Pi:

pi$ **cat /proc/cpuinfo | grep Revision**
Revision        : 900093

So same value in decimal and hex.

bennuttall, 2016-06-05:
Interesting. Could be useful. I think that's what @waveform80 had in mind. Though he does have a lot in his mind...

lurch, 2016-06-06:
The problem with trying to 'automatically' get the `pi_info()` for a remote Pi, is that theoretically you could have pigpiod connections to multiple remote Pis at once - which connection should pi_info return the details about?
That's why you need to 'manually' do what pigpiod.py does on line 111 - get the remote revision and manually pass it to the pi_info function.

> There might be a trick however as I don't have a match between this test code .... So same value in decimal and hex.

Not sure I understand what you're saying? 9437331 in decimal is 0x900093 in hex. With it being based around a C library, pigpio always reports the revision in decimal, whereas `/proc/cpuinfo` always reports the [revision](http://elinux.org/RPi_HardwareHistory#Board_Revision_History) in hexadecimal.

dglaude, 2016-06-06:
I am still in the mindset of writing the same code that works locally and remotely.

In that use case, there is no doubt about witch pigpiod to connect to, it is set in the environmental variable as the one and only.

I would prefer not to redo 'manually' do what pigpiod.py maybe does perfectly.

I updated the comment. I only noticed later that it was hex and decimal, but I forgot to edit.

lurch, 2016-06-06:
> In that use case, there is no doubt about witch pigpiod to connect to, it is set in the environmental variable as the one and only.

I think that would only work if either:
 * #353 got merged, and you'd set PIGPIO_ADDR and done `gpiozero.devices.pin_factory = PiGPIOPin`
 * #359 got implemented (which is just a 'shortcut' for setting pin_factory)

Currently `gpiozero/pins/data.py` (which is where the `pi_info` function lives) is pretty much standalone; I dunno whether @waveform80 would want to modify it to 'peek' at `gpiozero.devices.pin_factory` too? Maybe he'll have a cleverer approach ;)

waveform80, 2016-06-10:
PiGPIOPin was corrected to use `get_hardware_revision` (i.e. remote pi revision info) in 79a0fd2a (see [pigpiod.py line 110](https://github.com/RPi-Distro/python-gpiozero/commit/79a0fd2a278a571323a2f8dae92378381d0c9116#diff-fbefe69bc9724ee622cd38859bfd55f4R110) onwards) - i.e. *if* you use PiGPIOPin with a remote Pi then it queries the remote Pi's revision via that interface for lookup by `pi_info`. However, there's currently no way to make it *implicitly* use a remote Pi.

The idea of using some env vars to provide the defaults (#353) is very nice - that should definitely go in. I'm less hot on #359. Firstly I don't see the need for it - you could simply stick the following at the top of your script and assuming #353 was implemented it'd still work perfectly happily locally and remotely with no code alterations (assuming a recent raspbian that runs `pigpiod` by default):

```python
from gpiozero.pins.pigpiod import PiGPIOPin
import gpiozero.devices

gpiozero.devices.pin_factory = PiGPIOPin
```

The other reason is that while #353 has us reading *data* from the environment (hostnames and ports), #359 potentially has us reading executable code from the environment which makes me extremely nervous.

waveform80, 2016-06-10:
P.S. Apologies if I've missed the point of the discussion above (skimmed very rapidly as I'm swamped today trying to get some fancy stuff together for the MCR Rasp Jam show & tell tomorrow :) !

lurch, 2016-06-10:
Hmm, could
```python
from gpiozero.pins.pigpiod import PiGPIOPin
import gpiozero.devices

gpiozero.devices.pin_factory = PiGPIOPin
```
be reduced to a simple one-liner via a utility method, something like:
```python
gpiozero.set_pin_factory('PiGPIOPin')
```
?
(which would also make using MockPin in the interactive REPL much less arduous! ;-) )

> #359 potentially has us reading executable code from the environment which makes me extremely nervous.

Hmm?? I'd imagine it'd just be reading the string, and doing something like:
```python
env_default = os.getenv('GPIOZERO_PIN_FACTORY', None)
if env_default is not None:
    if env_default == 'RPiGPIOPin':
        # import and set default pin class to RPiGPIOPin
   elif env_default == 'PiGPIOin':
        # import and set default pin class to PiGPIOin
   # etc.
```
rather than directly executing environment-provided code?

I'm still +0 on whether #359 gets implemented or not though (although I can see how it'd be beneficial to some people).

dglaude, 2016-06-10:
@waveform80 

Maybe #353 and #359 should be discuss in there respective place and not here.

#353 is to just use exactly the official/standard mechanism build into PiGPIO python access library. My implementation might not be the best, but what can be wrong in a oneliner.

As @lurch say, there is nothing to be nervous about in #359, no code is executed, it is not supposed to be different that what #353 is doing. But then again, it is just an idea, if needed I can try to make an implementation and do a pull request.

Just the pin factory selection fallback sequence is configurable rather than to be burned into the code with fixed sequence/preference.

The idea is for the user to write code that is pin factory agnostic, and not to have to worry about it. It will always be possible for advance user to explicitly specify wich pin factory to use, control multiple Pi, access simultaneously local and remote pin, ... but for most educational use, having the same code working everywhere might help.

waveform80, 2016-06-10:
Good point, I'll add comments on #353 and #359

commit 4d00b069 (2016-04-15): Merge pull request #283 from waveform80/remote-pins

Fix #278
commit 79a0fd2a (2016-04-15): Fix #278

Allow the creation of remote pins easily with PiGPIOPin. Also changes
DefaultPin to a pin_factory callable which accepts the input pin number.