gpiozero/gpiozero #279: Add-on board use with pigpio?

bennuttall, 2016-04-13:
It's great that I can great an instance of an LED using a remote pin like so:

```python
from gpiozero.pins.pigpiod import PiGPIOPin
from gpiozero import LED
from signal import pause

led = LED(PiGPIOPin(12, host='192.168.0.2'))
```

However, I don't believe it's possible to create an instance of, say, a `SnowPi` attached to another Pi, because that doesn't take pin numbers on init, so I can't provide alternative pins.

Would it be possible to achieve this with an init kwarg?

```python
sp = SnowPi(host='192.168.0.2')
```

or perhaps configure the host at the top of the script, similar to RPi.GPIO's `setmode()`?

I imagine it could be useful to be able to control devices on multiple Pis at once, in the same script, so perhaps the `setmode` style configuration would not work for all cases.

lurch, 2016-04-14:
I guess this is where `DefaultPin` comes into play? http://gpiozero.readthedocs.org/en/latest/api_pins.html

Perhaps you could make a subclass of `PiGPIOPin` which is hard-coded to use a particular `host` (i.e. your remote Pi) and then set `gpiozero.devices.DefaultPin` to your new subclass? (and you could have multiple subclasses, each using a different IP address; changing the DefaultPin each time you want to address a different Pi).

All in theory, I've not actually tested any of this myself! ;-)

waveform80, 2016-04-15:
Yup, that's what `DefaultPin` is for, and @lurch is absolutely right that if you wanted a remote pin as the default that would currently mean sub-classing `PiGPIOPin` to change the default host in the constructor (if you want to be strict about it).

As a bit of background, `DefaultPin` is something I'm not terribly happy with: it works, but it's a bit crude and rather difficult to modify. I'd much prefer to replace it with a factory function/class that was called whenever a pin needed to be made with whatever arguments had been given to construct that pin (in fact this is more a documentation change than a code change - very little would change in `devices.py`). In this case instead of sub-classing PiGPIOPin you'd do something more like:

```python
from gpiozero.pins.pigpiod import PiGPIOPin
import gpiozero.devices

def make_remote_pins(number):
    return PiGPIOPin(number, host='my_pi')

gpiozero.devices.DefaultPin = make_remote_pins
```

As hinted at above, this would probably work now as I don't think `GPIODevice` cares whether the `DefaultPin` callable is actually a class or not - only that it accepts a single integer argument. What I would probably do is rename `DefaultPin` to something like `pin_factory` to make its intent clearer (remember the pins API isn't locked down yet so we can break stuff like this for now).

The one thing I'm still mulling is whether to permit more complex affairs; in other words to relax the test to say "if I'm passed a `Pin` derivative, use it, otherwise pass whatever I've been given (whether it's an integer, a string, a tuple of values, or whatever) to the pin factory and expect to get back a pin or have an exception raised. If that were the case you could potentially specific a pin factory that took an integer *and a host* (and whatever else you wanted) and act accordingly.

waveform80, 2016-04-15:
I've included an example of doing this explicitly in the docs in the fix for #278 (see http://gpiozero.readthedocs.org/en/latest/api_pins.html) in addition to explicitly renaming `DefaultPin` to `pin_factory`. If that's good enough, then we can close this. Otherwise I can look into adding a "host" constructor parameter but frankly that's going to touch a hell of a lot of stuff and I'm not sure it's a good idea.

dglaude, 2016-05-21:
I would like to do something very similar, but I don't know if it is already possible.

I want to change the default pin of GPIOZero like this:
    gpiozero.devices.DefaultPin = PiGPIOPin

(And somehow in the syntax, I should be able to specify the host if it is not localhost and/or the port if it is not the default)

This is the syntax documented as possible with NativePin in the documentation.

And I don't want to address a single pin (as documented) or to address the SnowPi like the original question... I would like to address all the GPIO (of another Pi) and be able to write:

    led = LED(16)
or
    fish = FishDish()

Is this already possible or has PiGPIOPin something missing that is present in NativePin?

If this become possible, then it would be possible to use a Raspberry PiZero, with Ethernet gadget and pigpiod started as a full featured GPIO extension. With a single USB cable and a microSD card, a PiZero could do the same things or more than the RTK.GPIO.

lurch, 2016-05-21:
Sounds like you didn't look at the api_pins.html page that @waveform80 linked to above? ;-)

Although that's only available in the bleeding-edge github version of GPIO Zero, so if you want something that'll work with the current 1.2.0 release, have a look at the `make_remote_pins` stuff in an earlier reply.

Using a PiZero as a more-capable RTK.GPIO equivalent is also something I'd been thinking about, but hadn't got round to looking at yet, and I didn't want to distract from Ryan's current Kickstarter campaign :)

dglaude, 2016-05-22:
Sorry, my question was drafted before reading this.
I did send it directly to Ben that suggested I post here.

I did read the api_pins.html from v1.2.0 to draft my question, but I did not notice the linked version was different. I start to be used with bleeding-edge, the first time I was playing with PIR I had to get the Github version (that time I was reading the latest documentation by mistake).

I believe this could be a RTK.GPIO killer (I am a backer), and I contacted Ryan as soon as I found noticed this possibility. On the down side, you have to solder the pin, and add the price of the PiZero and a microSD card. But the GPIO extension for PC/MAC will be able to work stand-alone with exactly the same code (if based on GPIO Zero and assuming it is possible to use this lib on PC/MAC).

Now seems to be the good time:

1. Latest Raspbian propose to start Remote GPIO in raspi-config [https://www.raspberrypi.org/blog/another-update-raspbian/](url)
2. Latest Raspbian can be configured as Etherner_USB Gadget from the PC/MAC/Linux that prepare the SD card: [http://blog.gbaman.info/?p=791&_utm_source=1-2-2](url)
3. PiZero is back in stock or at least is produced again
4. RTK.GPIO is backed (but you can wait for the end of the Kickstarter)

dglaude, 2016-06-06:
#359 with environment variable is even a better solution without having to use an explicit pin_factory.

lurch, 2016-08-31:
With all the work done on `pin_factory` in the 1.3 release, I believe this issue is resolved now.

bennuttall, 2016-12-21:
Following Dave's example:

```python
from gpiozero.pins.pigpiod import PiGPIOPin
import gpiozero.devices

def make_remote_pins(number):
    return PiGPIOPin(number, host='my_pi')

gpiozero.devices.DefaultPin = make_remote_pins
```

how would you change host between creating devices?

bennuttall, 2016-12-21:
How about:

```python
from gpiozero.pins.pigpiod import PiGPIOPin
import gpiozero.devices

def make_remote_pins_host(host):
    def make_remote_pins(number):
        return PiGPIOPin(number, host)
    return make_remote_pins

gpiozero.devices.DefaultPin = make_remote_pins_host("192.168.1.2")
red = LED(2)
gpiozero.devices.DefaultPin = make_remote_pins_host("192.168.1.3")
green = LED(2)
```

waveform80, 2016-12-21:
After #459 this will change but also be quite a bit easier because you can just construct a couple of factories with different hosts then use them to construct whatever pins you want, i.e. it'll be something like:

```python
from gpiozero import LED
from gpiozero.pins.pigpiod import PiGPIOFactory

host2 = PiGPIOFactory(host='192.168.1.2')
host3 = PiGPIOFactory(host='192.168.1.3')

red = LED(host2.pin(2))
green = LED(host3.pin(2))
# carry on making whatever you like using those factories...
```

bennuttall, 2017-03-28:
Reopening this issue.

1. We need to be able to easily create a pin factory object, and create objects like LEDs with pins from that factory, like so:

    ```python
    host2 = PiGPIOFactory(host='192.168.1.2')
    red = LED(host2.pin(2))
    ```

 2. We need to be able to easily change the default pin factory on-the-fly, so that we can just as easily use add-on-board classes which do not take pin numbers, like so:

    ```python
    local_th = TrafficHat()
    gpiozero.devices.DefaultPin = PiGPIOFactory(host='192.168.1.2')
    remote_th = TrafficHat()
    ```

(or at least something like this)

bennuttall, 2017-07-01:
@waveform80 what's the new syntax for the above examples?

lurch, 2017-07-03:
@waveform80 Following on from the above, would it be possible to (modify the gpiozero code, so that it's possible to) pass `factory` as a kwarg to e.g. the TrafficHat, so that you could do something like:
```python
host2 = PiGPIOFactory(host='192.168.1.2')
host3 = PiGPIOFactory(host='192.168.1.3')

local_th = TrafficHat()
remote_th1 = TrafficHat(factory=host2)
remote_th2 = TrafficHat(factory=host3)
```
without having to keep messing about with `gpiozero.devices.DefaultPin` ?

waveform80, 2017-07-03:
> @waveform80 what's the new syntax for the above examples?

```python
from gpiozero import *

factory2 = PiGPIOFactory(host='192.168.1.2')
factory3 = PiGPIOFactory(host='192.168.1.3')

local_th = TrafficHat()
Device.pin_factory = factory2
remote_th1 = TrafficHat()
Device.pin_factory = factory3
remote_th2 = TrafficHat()
```

> @waveform80 Following on from the above, would it be possible to (modify the gpiozero code, so that it's possible to) pass factory as a kwarg to e.g. the TrafficHat, so that you could do something like:

Oh! I'd been wondering about exactly that, but I hadn't managed to come up with a "clean" method of specifying a factory. I wonder if `factory` should be a constructor parameter at the Device level. If it's `None` (the default) it simply uses Device.pin_factory, otherwise it uses whatever is specified. That could be quite nice ... Or should it only be at the CompositeDevice level (as it's generally only composites that implicitly specify pin numbers)?

waveform80, 2017-07-03:
Hmm, thinking about it I'd probably call the parameter `pin_factory` partly to mirror the `Device` attribute, and partly to make it a little more explicit what purpose the factory serves.

bennuttall, 2017-07-03:
That would be neat. Are you thinking of adding this to 1.4?

lurch, 2017-07-03:
I had a brief think about it before suggesting it, and IMHO it should be at the CompositeDevice level - if you want to change the factory at the Device level you can just use the existing `factory.pin()` method.
I agree that it should default to `None`, but I was suggesting it should be a keyword-only argument (as some other args are) rather than being a positional argument.

waveform80, 2017-07-03:
> That would be neat. Are you thinking of adding this to 1.4?

Yes, definitely ... partly because it *might* affect remote pin-specs depending on the following ...

> I had a brief think about it before suggesting it, and IMHO it should be at the CompositeDevice level - if you want to change the factory at the Device level you can just use the existing factory.pin() method.
I agree that it should default to None, but I was suggesting it should be a keyword-only argument (as some other args are) rather than being a positional argument.

I absolutely agree it should be keyword-only. Not so sure about basing it on CompositeDevice though. The more I think about it, the more putting it at Device level makes things much "cleaner":

* At the CompositeDevice level this makes sense because one typically doesn't specify pin numbers. That means at present there's no method for specifying in the constructor that we wish to use a remote pin factory. This proposal fixes that.

* At the Device level it *can* make sense, but throws up the contradictory case of specifying a pin from one factory, while specifying a different factory for `pin_factory`:

```python
from gpiozero.pins.pigpio import PiGPIOFactory
from gpiozero import *

f1 = PiGPIOFactory(host='foo.local')
f2 = Device.pin_factory
l = LED(f2.pin(4), pin_factory=f1)
```

* However, this contradiction only arises because we permit the pin to be either a pin-spec (an integer GPIO number) or a `Pin` implementation. What if we *only* permitted a pin-spec? (I don't think many people are presently manually constructing `Pin` implementations when constructing devices, so I'm not terribly worried about the backwards-incompatible nature of this change - after all we're changing it in this release *anyway* from, say, `RPiGPIOPin(4)` to `rpi_factory.pin(4)`).

* In other words, Device would introduce the keyword-only pin_factory argument which would default to `None` meaning "just use whatever `Device.pin_factory` is". GPIODevice would add the pin argument (as it does currently), but it would only be permitted to be a pin-spec (or more precisely we'd simply assume it was a pin-spec). This would mean constructing solo devices for remote Pis would be entirely consistent (syntactically) with constructing composite devices (like boards) for remote Pis. In other words:

```python
from gpiozero.pins.pigpio import PiGPIOFactory
from gpiozero import *

f1 = PiGPIOFactory(host='foo.local')
l = LED(4, pin_factory=f1)
th = TrafficHat(pin_factory=f1)

# Or alternatively:
Device.pin_factory = f1
l = LED(4)
th = TrafficHat()
```

* I'm currently thinking through the implications for the pin reservation system but so far, I think this just simplifies things even further (I haven't thought of a case yet where it can complicate the mechanisms).

Any thoughts?

waveform80, 2017-07-03:
Oh boy ... this potentially clears up the rubbish "pin address" mechanism I had to come up with for pin reservations (and which has been nagging at the back of my mind ever since). Pin reservation can (should) effectively become a factory level thing (since one factory doesn't particularly care about another factory's pins are reserved; there's an exception to be made for local-only factories there but that's simple enough). That in turn means we don't need factory addresses which sorts out the whole localhost / 127.0.0.1 reservation thing, address separators, and still none of this affects the user API. The more I think about this, the better it looks. I'll have a go at throwing something together in another branch.

lurch, 2017-07-03:
> At the Device level it can make sense, but throws up the contradictory case of specifying a pin from one factory, while specifying a different factory for pin_factory:

Yeah, that was my thinking / concern...

> What if we only permitted a pin-spec?

...ooh, that might fix it ;-)  What exactly is a "pin-spec"?

Hmmm, I've just remembered that part of your 'factory' stuff was to work with e.g. IO expanders, so that you could (possibly) do something like this with the current scheme:
```
expander_factory = I2CExpander(address=0x34)
l = LED(expander_factory.pin('BANKB', 3))
```
(unless I'm misunderstanding things?) and I guess that _wouldn't_ work with the Device having just a `pin_factory` kwarg? (or would it??)

> The more I think about this, the better it looks.

Cool, glad to hear my crazy thoughts can lead to worthwhile ideas. And if this makes things "simpler" and more consistent then that's great. I think I've lost count of the number of design-iterations the "pin stuff" has been through now ;-)

lurch, 2017-07-03:
(re-opening this issue again)

waveform80, 2017-07-03:
> ...ooh, that might fix it ;-) What exactly is a "pin-spec"?

For now, simply an integer number specifying a GPIO port, e.g. 17 for GPIO17.

> (unless I'm misunderstanding things?) and I guess that wouldn't work with the Device having just a pin_factory kwarg? (or would it??)

Actually, I think that would work. Either the expander can be the factory itself (as in your example):

```python
expander_factory = I2CExpander(address=0x34)
l = LED(3, pin_factory=expander_factory)
```

Or it can be a device which provides a factory as a property, like so:

```python
expander = I2CExpander(address=0x34)
l = LED(3, pin_factory=expander.factory)
```

Not sure which is best yet (I have a hunch object lifetime issues will make the latter design simpler, but that's just speculation). Anyway ... yes, it all "just works" with the same old design. And it means I can strip out my horrid pin-address hack (just doing that on an experimental branch right now - looking much cleaner!)

> Cool, glad to hear my crazy thoughts can lead to worthwhile ideas. And if this makes things "simpler" and more consistent then that's great. I think I've lost count of the number of design-iterations the "pin stuff" has been through now ;-)

Yeah, it's been really bugging me that we've gone through many iterations and none of them has quite "felt right". Things got better with the introduction of factories (removed all that horrid `__new__` nonsense), but there were still horrid bits like the pin-address stuff. But this iteration finally feels  "good". Just ploughing through it now and so far nothing's annoying me ;)

lurch, 2017-07-03:
Ahhh, looks like you missed the pertinent bit of my example - in the `expander_factory` the `pin` method actually took two parameters, not just one. (But I dunno if that's 'legal' under the current system anyway?)

> But this iteration finally feels "good". Just ploughing through it now and so far nothing's annoying me ;)

Woohoo!

waveform80, 2017-07-03:
> Ahhh, looks like you missed the pertinent bit of my example - in the expander_factory the pin method actually took two parameters, not just one. (But I dunno if that's 'legal' under the current system anyway?)

Ooops - yup, I did. However we can certainly deal with that under the proposed system. As mentioned a pin-spec is currently just an integer GPIO number, which is fine as far as Pi-based pin factories are concerned (until/unless we want to expand that but that's an orthogonal issue to this so let's not get into that here :). However, as far as expander devices are concerned the pin-spec can be ... well, anything we want basically. Could be we permit the pin-spec to be a string like 'BANKB:3' for them, or it could be a tuple like ('BANKB', 3). Basically there's no hard requirement that pin-specs are integers in all cases.

lurch, 2017-07-03:
Ooooh. So what if we allowed a RemotePiGpioFactory to have a pin-spec of `(ip_address, pin_number)` ? Or does that start making things more complicated again? (is this a path we've already been down? I can't remember)

So I guess it's up to each PinFactory to determine what types it accepts as a pin-spec, and error as appropriate? Is there some kind of InvalidPinSpec exception? ;)

waveform80, 2017-07-03:
Damn, not going to get this finished tonight. It's actually working tolerably well, but the change affects a *huge* amount of the test suite (where we pass pin instances around a lot). And apparently I'm mentoring tomorrow (thought it was Weds/Thurs but it's Tues/Weds, sorry @bennuttall)! I'll try and get something pushed to my clone tonight in case anyone wants to play with it so far (branch is called pin-factory-param).

waveform80, 2017-07-03:
> So I guess it's up to each PinFactory to determine what types it accepts as a pin-spec, and error as appropriate? Is there some kind of InvalidPinSpec exception? ;)

Precisely, and yes - it's PinInvalidPin (which should probably be PinInvalidPinSpec but that's something people *might* actually be using already as it's been around a few versions so I'm tempted to leave well alone for now).

waveform80, 2017-07-03:
> Ooooh. So what if we allowed a RemotePiGpioFactory to have a pin-spec of (ip_address, pin_number) ? Or does that start making things more complicated again? (is this a path we've already been down? I can't remember)

It's potentially possible, but for the moment I'm content to have PiGPIOFactory own a single connection, so the host is at the factory rather than the pin level. But there's nothing stopping someone from implementing such a pin factory with such a pin-spec.

waveform80, 2017-07-05:
>> ... I agree that it should default to None, but I was suggesting it should be a keyword-only argument (as some other args are) rather than being a positional argument.

> I absolutely agree it should be keyword-only...

Sadly I may have to drop it being keyword-only. The only way to implement this and remain py2 compatible is with `**kwargs`. That's possible, but it means introducing `**kwargs` almost everywhere (`InputDevice`, `OutputDevice`, all the descendents, the list goes on) and as I do so it's feeling horribly arbitrary and inconsistent with the pile of args we already introduce on various things. For example, on `Button` we've got `bounce_time`,  `hold_time`, `hold_repeat` each of which should arguably be keyword only. Do we replace everything with `**kwargs` and rely upon the documentation as the "true" source of what can be passed? We could, but it harms quite a few Python IDEs which can no longer parse acceptable params from the declaration itself. For now, I'll go with an additional `pin_factory` param and see how it feels.

waveform80, 2017-07-05:
Okay, there's an implementation of this (with, I think, all the documentation updated too) in my clone under the pin-factory-param branch. Feel free to have a play! The test suite passes but I need to extend it to test the new stuff yet but that'll have to wait until later this week - too knackered tonight!

lurch, 2017-07-07:
> Sadly I may have to drop it being keyword-only.

IMHO the trouble with making it positional, is it then 'locks it' into each class's positional arguments signature (given that we've pledged to maintain backwards-compatibility), making it very hard (or impossible) to add additional positional arguments in future. But @waveform80 obviously knows the whole GpioZero library much better than I do, and is a much more experienced Python developer too (so obviously knows what works and what doesn't!).

> Feel free to have a play!

I'd normally offer to have a play with it over the weekend, but I'm away all of this weekend climbing mountains in Wales! :-)

waveform80, 2017-07-07:
> IMHO the trouble with making it positional, is it then 'locks it' into each class's positional arguments signature (given that we've pledged to maintain backwards-compatibility), making it very hard [...]

Yup, can't say I like this either (for precisely this reason). If we were targetting Python 3 only, I'd definitely say make this keyword-only. Unfortunately, in Python 2 the only way to make an argument keyword only is to accept `**kwargs` in the function/method and then extract the parameter from there. I've done that down in `devices.py` (for `Device`, `GPIODevice`, and `CompositeDevice`) because I figure no user will ever be directly instantiating those, and because that forces all the classes that descend from those to use keyword args when passing this stuff up the chain (so we've got a load of "good" examples of how to pass `pin_factory` up to a super-call).

However, there is one major drawback: I've been playing around with Python IDEs recently (not because I particularly like them, but in the series of tutorial videos I've been doing for picamera I'm trying to present lots of different ways of coding Python). Having the keyword args wrapped up in `**kwargs` effectively means none of the IDEs can display the available keyword args when queried (unsurprisingly, they just present the fact the function/method takes keyword args).

So, the compromise I've gone for in the pin-factory-params branch is:

1. Force keyword args at the low level because end-users won't touch those classes (so it doesn't interfere with their IDE experience), but it forces all our "leaf" classes to have "clean" implementations (using keyword args where-ever `pin_factory` gets passed up).
2. Declare `pin_factory` as an ordinary optional parameter in the "leaf" classes (so it works nicely with IDEs)
3. Lie in the documentation and state it's a keyword-only parameter because at least that way we've made it explicit that if people treat this as anything else we're free to break their code in future (sort of; in most of the docs I just mention this is "an advanced feature" and direct people to the Pins page for more information ... and then blatantly lie to them in the Pins page).

I'm a bit on the fence with this one: if you and Ben are happier with it being keyword-only, I can certainly go with that. Just thought I'd throw the usual spanner in the works ;)

> I'd normally offer to have a play with it over the weekend, but I'm away all of this weekend climbing mountains in Wales! :-)

Very nice, enjoy!

lurch, 2017-07-10:
I have no strong opinions in either direction (which I guess means the decision is down to @bennuttall :wink: ), but is `pin_factory` likely to be a commonly-used-enough feature to worry about it appearing in an IDE's auto-complete list or not? If the `pin_factory` parameter works uniformly across all `Device` and `CompostieDevice` subclasses, maybe it doesn't actually need per-device IDE auto-completions? (because the people who want/need to use it will have already read up how it works)

Just out of curiosity, how do other Python projects deal with this some-parameters-are-kwarg-only situation? Are there any IDEs that are able to pull the info out of the docstrings, and not just rely on the method-signature? ;-)  (but I guess that would rely on the docstrings being 100% accurate, which I'm sure we all know from experience isn't always the case!)

waveform80, 2017-07-11:
> I have no strong opinions in either direction (which I guess means the decision is down to @bennuttall :wink: ), but is pin_factory likely to be a commonly-used-enough feature to worry about it [...]

Sadly there are no IDEs that pull the info from doc-strings. I'm not sure it's even possible to do realistically: some environments, like ipython notebook and bpython, display the whole doc-string when requested but even that's not enough as the signature itself is actually in the .rst files (the signature also acts as the pointer that tells sphinx to pull the rest of the doc-string; see `docs/api_output.rst` for an example).

As for other libraries, I've seen all kinds but I've no idea what's the most popular mechanism:

1. some use `**kwargs` to enforce keyword-only args on all Python implementations, and document in Python 3 style (but as mentioned IDEs that display call prototypes generally display the "wrong" thing for these)
2. some just use regular optional args but document the arg as keyword only (which is currently how I've done things)
3. some just drop Python 2 compatibility move onto actual keyword-only args for new releases (don't think this is really an option for us yet)

bennuttall, 2017-07-11:
I'm happy with us saying it's keyword-only. That's sensible.

:+1:

bennuttall, 2017-07-11:
I'm happy with us saying it's keyword-only. That's sensible.

commit a99e0746 (2017-06-22): Fix #279

Permit replacement of pin_factory without closing old factory. However,
continue closing devices associated with extant pin factory at script
termination.
commit 93a1f529 (2017-07-14): Merge pull request #574 from waveform80/pin-factory-param

Fix #279
commit c820636f (2017-07-14): Fix #279 once and for all (ha! ;)

This implements the proposal discussed in the re-opened #279 to add a
pin_factory argument at the device level and remove the ability to
specify a pin instance to device constructors (they now only accept a
pin specification).

Note: there's still a couple of bits to tidy up (tests on "real" Pis,
and pin_factory.release_all needs refinement) but the test suite is now
at least capable of passing on a PC.
closed 2017-07-14 by waveform80