gpiozero/gpiozero #287: when_held missing from documentation

bennuttall, 2016-04-18:
I was just looking up the usage of `button.when_held` but found it wasn't listed under `Button`. I inspected the source and realised that's because it comes from `HoldableMixin` and the docs don't automatically show parent class members.

Is it possible to show the hold stuff for `Button` in the docs, and/or (probably and) can we add a recipe for it?

@waveform80 @martinohanlon

waveform80, 2016-04-19:
Argh - that's my fault - I'll bung it in the API (I can set the docs to just include everything from parent classes, but the parent API is now pretty bulky what with `source`, `source_delay`, `values`, etc. so for the "simpler" concrete classes like `Button` we tend to [explicitly list what gets shown](https://raw.githubusercontent.com/RPi-Distro/python-gpiozero/master/docs/api_input.rst)).

closed 2016-05-26 by bennuttall

waveform80, 2016-05-26:
Bah, beat me to it ... need more hours in my day!

bennuttall, 2016-05-26:
Trying to do what I can! Do you think I missed anything or added anything I shouldn't?

waveform80, 2016-05-27:
No, looks fine to me. If you want to test docs before committing / pushing / merging / whatever, just run "make doc" and then look at `docs/_build/html/` for the results (`docs/_build/latex` will have the PDF output).

lurch, 2016-05-27:
...but bear in mind that you'll need to do `sudo apt-get install texlive-latex-recommended texlive-latex-extra texlive-fonts-recommended` before being able to do `make doc`.

waveform80, 2016-05-27:
oh good point - I should add that stuff to the developer docs in picamera!

commit 9278a2f8 (2016-05-26): Add hold/held properties & methods to Button, close #287