gpiozero/gpiozero #289: Add note about remote access

IanHun, 2016-04-19:
Note

The second example of the Keyboard Controlled Robot will only work if the keyboard/screen are attached directly to the RPi, so will not work with remote access mechanisms (SSH and VNC).

waveform80, 2016-04-19:
Interesting - can't say I've tried evdev over SSH. I presume there must be some workaround (otherwise how would X get evdev events from a forwarded session over SSH), but I'm not sure it'll be trivial enough for the recipe. Either way, we should either document the short-coming or fix it.

lurch, 2016-04-19:
> I presume there must be some workaround (otherwise how would X get evdev events from a forwarded session over SSH)

Perhaps that's one of the intricacies of using `ssh -X ...` rather than just a plain 'console' ssh session? \*shrug\*
But I guess evdev also won't work over a serial console, whereas ncurses would.

waveform80, 2016-08-29:
Okay, so there aren't evdev messages going over SSH - it's straight X messages (which makes sense now I think about it) so this recipe will never work over SSH. I'll add a note to the docs to that effect. Incidentally, while digging into this I also found out how to detect a keyboard from evdev, so I'll enhance the recipe a little to include that (as my keyboard doesn't show up as device[0] typically and it'd be nice to have it "Just Work").

commit 07c95598 (2016-08-29): Fix #289

Tweak the second keyboard robot recipe so it detects keyboards and
doesn't throw exceptions on unknown keys. Also add a note to the recipe
that it won't work over remote connections like SSH.
commit bc7588b9 (2016-08-29): Merge pull request #412 from waveform80/remote-keyboard-robots

Fix #289
closed 2016-08-29 by waveform80