gpiozero/gpiozero #294: internal device idea: cpu temperature

drewp, 2016-04-20:
You might like to offer /sys/class/thermal/thermal_zone0/temp (divide by 1000 for C) as one of the internal devices on https://gpiozero.readthedocs.org/en/v1.2.0/api_other.html

lurch, 2016-04-20:
Nice idea. Should probably have the same API as #93

bennuttall, 2016-04-20:
http://www.stuffaboutcode.com/2015/06/raspberry-pi-cpu-temperature.html @martinohanlon

martinohanlon, 2016-04-20:
I like the idea of creating a CPU class where you could gain information about the CPU - temperature, version, clock speed, serial no, overclocking etc.

Should this form part of gpio-zero though? It feels outside the scope of the module.

Do we need a new module? A raspberry_pi module which provides access to classes relating to the computer. CPU, GPU, Audio, Display, USB, Bluetooth, Wifi, RAM, Storage, etc.

lurch, 2016-04-20:
> Should this form part of gpio-zero though? It feels outside the scope of the module.

That's what I wondered too, but @waveform80 already added most of what you listed to http://gpiozero.readthedocs.org/en/latest/api_pins.html#utilities

waveform80, 2016-04-21:
I do like the idea of an internal temperature device - that's exactly the sort of thing the internal devices are for (not representable on the GPIO pins, but specific to the Pi, and useful as a data-source for pin-controlled devices).

waveform80, 2016-04-21:
The data module (and associated pi_info function) does indeed add most of the stuff @martinohanlon lists above - but not all. The stuff included is specifically: fixed information (not temperature, which should be handled by a device as it's dynamic) and secondly difficult to get elsewhere (I don't include the number of CPU cores because you can get it trivially from /proc/cpuinfo, but memory isn't as trivial because of the CPU/GPU mem split so it is included, number of built-in USB ports is likewise difficult to calculate, and so on).

bennuttall, 2016-04-21:
I like the idea

bennuttall, 2016-04-21:
Especially with values!

dglaude, 2016-06-06:
As an advocate of remotely controling GPIO accross pigpio protocol, would prefer not to add to gpiozero stuff that are not related to GPIO and not reachable via pigpio protocol.

lurch, 2016-06-06:
I dunno - I'm not sure we should restrict gpiozero to only offering features that can be controlled over a remote pigpiod connection :-/
I'm fairly sure the overwhelming majority of gpiozero users will be running it on-Pi and not over a remote network connection.

waveform80, 2016-08-14:
Yup, still all for this. I don't see lack of pigpio support as a show-stopper; there's no reason an interface for this couldn't be added to pigpio later if it became a popular request and as @lurch notes the vast majority of gpiozero use is local. Might have a crack at this later tonight (unfortunately I've got to go play recklessly with fire on the BBQ the rest of this afternoon!)

commit 04075380 (2016-08-29): Fix #294

Still need to sort out threads for the internal devices to continually
poll their values so that the wait-states work properly
commit bb8ea524 (2016-08-29): Merge pull request #414 from waveform80/cpu-temperature

Fix #294
closed 2016-08-29 by waveform80