gpiozero/gpiozero #295: Privacy breach: local documentation shouldn't include Travis build status

EdwardBetts, 2016-04-21:
As discussed in #284.

The documentation available at http://gpiozero.readthedocs.org/en/latest/ includes the Travis build status. This is useful.

The build status shouldn't be included in the documentation intended to be read offline, specifically in the Debian package of documentation.

waveform80, 2016-04-21:
Thanks for the reminder ticket - I'll see what I can do about this for the next release

waveform80, 2016-08-30:
ARGH! Bloody PyPI doesn't work with the damned thing either. Right, sorry about this but I'm going to have to revert the change - it just breaks too many things and I've got no time to fix it now. If someone wants to do a proper fix later (whatever that may entail), be my guest.

lurch, 2016-08-30:
Was this closed again accidentally? Should it be re-opened?

waveform80, 2016-08-30:
Bah, yeah that's GitHub reacting to the "Fix" in the title without seeing the "Revert" which git bunged before it.

waveform80, 2021-03-12:
Fixed in a2b2b816510cebdbd2e4cb56906b2fe27e8bbf7e

commit 38541889 (2016-08-29): Fix #295

At least, I *think* this should fix this. It does locally, but we'll
have to see what happens when I push it to RTD!
commit 763b8f79 (2016-08-29): Merge pull request #413 from waveform80/fix-privacy-breach

Fix #295
commit e9ff6fce (2016-08-30): Revert "Fix #295"

This reverts commit 38541889cabb11685878cb2779c2ca99611a7531.
closed 2021-03-12 by waveform80