gpiozero/gpiozero #296: DistanceSensor - resistor divider is wrong.

DarrenTownsend, 2016-04-21:
The arrangement of the resistors, as described in the instructions for the Distance Sensor, is wrong, and will result in 4.85V being fed into the 'Echo' GPIO pin. What is described is NOT a resistor divider. The 330R resistor should be from the Echo pin of the HC-SR04 and the 470R resistor from the GND pin. The other ends of both resistors should be connected (normally by a jumper cable) to the 'Echo' pin of the Pi. See here: http://camjam.me/?page_id=1035 for more details (Worksheet 6, page 3).

lurch, 2016-04-21:
Unless I'm reading it wrong, I think what you've written above _is_ the same as http://gpiozero.readthedocs.org/en/v1.2.0/api_input.html#distance-sensor-hc-sr04 describes, with the only difference being that the gpiozero instructions bypass the jumper cable you mention, and just connect both resistors directly to the GPIO pin?

Of course any confusion would be cleared up by #51

DarrenTownsend, 2016-04-21:
Oops! You're right! I've just read them again after reading your reply, and they do make sense. I obviously had it in my head that the 2 mentions of 'ECHO' both referred to the sensor module pin ! Yes, diagrams would definitely help (me, at least!).

closed 2016-04-21 by DarrenTownsend

waveform80, 2016-04-21:
Oh damn, have I *still* not done a distance sensor one? Sorry - there's probably a reason (is the HC-SR04 in Fritzing? If not, that'll be why - otherwise it's just me not having time :)

DarrenTownsend, 2016-04-22:
No, there isn’t an HC-SR04 in Fritzing, although there are a couple of Fritzing projects that have one. It isn’t brilliant, however, being a 2D view from the front so the pins seem backwards. I must have read through that description a dozen times, and still didn’t pick up that it referred to the ECHO pin on the Pi, as opposed to the module! On the upside, at least I now know that putting 4.85 through a GPIO pin doesn’t always kill it !!

 

Regards, Darren.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

waveform80, 2016-05-08:
Hmm, at the least I probably ought to switch the order of 3 and 4 in those instructions so that people connect the Pi's ECHO pin to ground *first* and *then* connect to the sensor's ECHO pin. Otherwise I can imagine the scenario where someone selects, say, GPIO2 for the trig pin - it's pulled high by default, connects the sensor's ECHO to a GPIO pin via a 330 ohm resistor and accidentally fries it because they haven't yet put the other half of the voltage divider in.

Sounds like the diagram will have to wait a while though.

lurch, 2016-05-08:
https://github.com/CamJam-EduKit/EduKit3/raw/master/CamJam%20EduKit%203%20-%20Robotics%20Worksheet%206%20-%20Distance.pdf has some Fritzing-esque images of an HC-SR04. Might be worth getting in touch with @geekytim ?

GeekyTim, 2016-05-08:
What is wrong with the divider?

waveform80, 2016-05-09:
@GeekyTim nothing wrong with the divider - I'm just hoping to add a wiring diagram to our gpiozero docs for the HC-SR04 sensor (and possibly tidy up our wiring instructions which are potentially badly ordered at the moment - detail a couple of messages above). Unfortunately Fritzing lacks an image for the component but the camjam 3 robotics PDF has a nice frontal view of one - we're wondering where you got it and if it's available as a Fritzing component?

GeekyTim, 2016-05-09:
I'm pretty sure I got it in Fritzing - let me check when I get home.

Tim

GeekyTim, 2016-05-09:
It's not in the standard Fritzing, but is one of the items available from
the Adafruit library - https://github.com/adafruit/Fritzing-Library

Tim

waveform80, 2016-05-09:
Ah, great stuff - I'll have a dig into that!