gpiozero/gpiozero #300: Using L298N-Motor with gpiozero

goloplo, 2016-04-28:
Using this code on RPI2 with L298N-Motor:

`from gpiozero import Motor`
`from time import sleep`
`motor = Motor(forward=17, backward=27)`
`while True:`
`motor.forward()`
`sleep(5)`

gives the following error:

> Traceback (most recent call last):
  File "motpi.py", line 3, in <module>
    motor = Motor(forward=17, backward=27)
  File "build/bdist.linux-armv7l/egg/gpiozero/devices.py", line 122, in __call__
  File "build/bdist.linux-armv7l/egg/gpiozero/output_devices.py", line 756, in __init__
  File "build/bdist.linux-armv7l/egg/gpiozero/devices.py", line 122, in __call__
  File "build/bdist.linux-armv7l/egg/gpiozero/output_devices.py", line 306, in __init__
  File "build/bdist.linux-armv7l/egg/gpiozero/pins/__init__.py", line 179, in <lambda>
  File "build/bdist.linux-armv7l/egg/gpiozero/pins/__init__.py", line 175, in _set_frequency
gpiozero.exc.PinPWMUnsupported: PWM is not supported on pin GPIO17
Exception AttributeError: AttributeError(u'no such attribute _source_thread',) in  ignored

If I use LED(17), LED(27) is working perfect.

lurch, 2016-04-29:
I've just tried reproducing this myself on both Python2.7 and Python3.4 on each of a RPi2 and RPI3, using the latest version 1.2 of gpiozero, and in all cases your sample script worked without any problem :-/
Are you definitely using the latest v1.2 of python-gpiozero? How did you install it? Which version of Raspbian and/or Python are you using?

bennuttall, 2016-04-29:
That looks like it's not using RPi.GPIO - because its software PWM is permitted on any pin. It must be falling back to RPIO, or more likely the built-in pins implementation.

@goloplo Are you running this in a virtualenv? If so, you probably want to install RPi.GPIO in the virtualenv. If not, have you uninstalled RPi.GPIO?

@waveform80 can you tell from the stack trace which implementation is being used?

bennuttall, 2016-04-29:
@goloplo Note I've edited your post to use fenced code blocks to help make it more readable. Click edit and see how it's laid out for future reference.

lurch, 2016-04-29:
Just to keep things tied together - see my comments on #301 too.

lurch, 2016-04-29:
And see also #303 :-)
(which indeed allows motors to be used if RPi.GPIO isn't installed, by specifying `pwm=False` in the Motor constructor)

goloplo, 2016-04-29:
Bennuttall, lurch thank you.

To answer your questions:

1. I'm running this in a virtualenv.
2. RPi.GPIO is not install.

So my current set (one motor) without PWM is:
```
    | BCM Pin |           | L298N |
         17   -------------> IN1
         27   -------------> IN2
```
If I have to use (one motor) PWM should be:
```
    | BCM Pin |           | L298N |
          4   -------------->ENA
         17   -------------> IN1
         27   -------------> IN2
```
I think lurch suggestion is very good. Using `pwm=False` or defining two motor classes.
Another suggestion is for the speed of the motor maybe a range 0 to 120 should be added as on Arduino.
This way GPIOZERO can separate from the installation of RPi.GPIO.

bennuttall, 2016-04-29:
> I'm running this in a virtualenv.
> RPi.GPIO is not install.

You're better off depending on RPi.GPIO at present. It's the most reliable back-end.

> I think lurch suggestion is very good. Using pwm=False or defining two motor classes.

I agree. I think I raised the issue of why we don't have an on/off motor at some point previously, I can't remember the reasoning. Alternatively, `pwm=False` would do.

> Another suggestion is for the speed of the motor maybe a range 0 to 120 should be added as on Arduino.

All device values range 0 to 1 (or -1 to 1) so we wouldn't change that. It makes it easier if everything is proportional and components can be easily used together, for example if a light sensor has values 0 to 1, the value can be passed directly (or inversely) to a PWM device like PWMLED or Motor.

goloplo, 2016-04-29:
So it is not possible to add `Motor()` and `PMotor()` ? 
`PMotor()` will have the `pwm=True` and will use 3 pins, one for speed and to for rotation.

lurch, 2016-04-30:
Myself, @bennuttall and @waveform80 are all "software guys" and not hardware-guys or electronics-guys, so we're not always aware of all the different options.

In terms of motor-driver chips, the current `Motor` class supports (PWM_forwards, PWM_backwards) style controls, but it sounds like the `PMotor` class you're suggesting requires (Digital_forwards, Digital_backwards, PWM_speed) style controls, and the `PhaseEnableMotor` in #276 requires (Digital_direction, PWM_speed) style controls.
See also my brief discussion in https://github.com/RPi-Distro/python-gpiozero/pull/277#issuecomment-211608633 about different types of motor drivers.

goloplo, 2016-04-30:
Thanks for replys guys.
On [RPI forum](https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=90243) they are saying that the IN pins from the drive board  can be used with PWM.
I'll will try that and update the post.

lurch, 2016-04-30:
Page 4 of http://www2.st.com/resource/en/application_note/CD00003776.pdf talks about the difference between using PWM on the INput pins and using PWM on the ENable pins. The finer details are a bit over my head.

goloplo, 2016-05-02:
Hi guys,

Just an update I've done the installation on the virtual environment of the `RPi.GPIO` and it is working. I'm using just 2 IN pins per motor and it is working with PWM as well.

Thank you.

closed 2016-05-26 by bennuttall

lurch, 2016-05-26:
IMHO this should be re-opened, as I believe to 'properly' support the L298N we'd need a Motor class with (Digital_forwards, Digital_backwards, PWM_speed) style controls.

bennuttall, 2016-05-26:
Thanks. Just looking at this thread, the contents seem to be focused around whether or not this was a bug, and problems with back-end PWM support. I've closed and created a new issue #330 for adding support for more motors. Hope that helps.