gpiozero/gpiozero #305: Robot can't have it's value set

lurch, 2016-04-30:
AFAIK the following _should_ work, but it doesn't.
```python
>>> from gpiozero.pins.mock import MockPWMPin
>>> import gpiozero.devices
>>> gpiozero.devices.pin_factory = MockPWMPin
>>> from gpiozero import Robot
>>> r = Robot((1,2), (3,4))
>>> r.value
RobotValue(left_motor=0, right_motor=0)
>>> r.value = r.value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/andrew/tmp_working/python-gpiozero/gpiozero/devices.py", line 299, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/home/andrew/tmp_working/python-gpiozero/gpiozero/devices.py", line 146, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
AttributeError: can't set attribute
```

Just for comparison, this does work:
```python
>>> from gpiozero.pins.mock import MockPWMPin
>>> import gpiozero.devices
>>> gpiozero.devices.pin_factory = MockPWMPin
>>> from gpiozero import TrafficHat
>>> t = TrafficHat()
>>> t.value
TrafficHatValue(lights=TrafficLightsValue(red=False, amber=False, green=False), buzzer=False, button=False)
>>> t.value = t.value
```

If the Robot can't have it's `value` set, then I guess it doesn't make sense for it to use the `SourceMixin`?

lurch, 2016-05-03:
This must be because the Robot [inherits](http://gpiozero.readthedocs.io/en/latest/api_boards.html#base-classes) from CompositeDevice, whereas the TrafficHat inherits from CompositeOutputDevice (which gives it extra on / off / toggle methods, which probably don't make sense for the Robot class).
So therefore I guess the fix is to give Robot it's own `value` property getters and setters, just like Motor has. And I guess the natural 'value' for a Robot is simply a tuple of (left_motor_speed, right_motor_speed).

waveform80, 2016-05-08:
Doh - absolutely correct on all counts (should have its own value getters/setters which should return a `(left, right)` tuple).

PR #342 (2016-05-28): Add value-setter for Robot class
Fixes #305
commit b3035d30 (2016-05-28): Add value-setter for Robot class
Fixes #305
closed 2016-05-28 by bennuttall