gpiozero/gpiozero #309: Connect to iRobot Create

keshavsaharia, 2016-05-02:
Thank you @bennuttall for showing me this awesome library at picademy! I wanted to quickly propose an addition to gpiozero to allow it to interface with the iRobot Create ([open interface](http://www.irobot.com/filelibrary/pdfs/hrd/create/Create%20Open%20Interface_v2.pdf)). If it makes more sense for this to be a third-party library, let me know and I can create a repository for it and share the link instead.

The general idea behind connecting to the Create is quite simple, it uses the `serial` library:
```python
connection = serial.Serial("/dev/ttyUSB0", baudrate=57600, timeout=1)
```

Here is the code snippet I was planning to run today:
```python
import struct
import sys, glob
import serial
import time

for p in glob.glob('/dev/tty[A-Za-z]*'):
	print(p)

VELOCITYCHANGE = 200
ROTATIONCHANGE = 300

class iRobot:
	def __init__(self, port):
		self.connection = serial.Serial(port, baudrate=57600, timeout=1)
		self.direction = [False, False, False, False]
		self.sendCommandASCII("128")
		self.sendCommandASCII("131")
	
	def beep(self):
		self.sendCommandASCII("140 3 1 64 16 141 3")
		
	def stop(self):
		self.setDirection("none")
		self.motionChange()
	
	def wait(self, t):
		self.setDirection("none")
		self.motionChange()
		time.sleep(t)
		
	def setDirection(self, d):
		self.direction = [d == "up", d == "right", d == "down", d == "left"]
		
	def left(self, t):
		self.setDirection("left")
		self.motionChange()
		time.sleep(t)
		
	def right(self, t):
		self.setDirection("right")
		self.motionChange()
		time.sleep(t)
	
	def forward(self, t):
		self.setDirection("up")
		self.motionChange()
		time.sleep(t)
	
	def backward(self, t):
		self.setDirection("down")
		self.motionChange()
		time.sleep(t)
		
	def motionChange(self):
		velocity = 0
		velocity += VELOCITYCHANGE if self.direction[0] is True else 0
		velocity -= VELOCITYCHANGE if self.direction[2] is True else 0
		rotation = 0
		rotation += ROTATIONCHANGE if self.direction[3] is True else 0
		rotation -= ROTATIONCHANGE if self.direction[1] is True else 0

		# compute left and right wheel velocities
		vr = velocity + (rotation/2)
		vl = velocity - (rotation/2)
		
		# create drive command
		cmd = struct.pack(">Bhh", 145, vr, vl)
		self.sendCommandRaw(cmd)
	
	def sendCommandASCII(self, command):
		cmd = ""
		for v in command.split():
			cmd += chr(int(v))
		self.sendCommandRaw(cmd)

	# sendCommandRaw takes a string interpreted as a byte array
	def sendCommandRaw(self, command):
		try:
			if self.connection is not None:
				self.connection.write(command)
		except serial.SerialException:
			print("Lost connection")
			self.connection = None

		print(' '.join([ str(ord(c)) for c in command ]))

robot = iRobot("/dev/ttyUSB0")
robot.beep()
robot.forward(2)
robot.left(1)
robot.right(2)
robot.left(1)
robot.backward(2)
robot.stop()
```

Let me know if you have any questions or suggestions!

lurch, 2016-05-02:
Looks good! IMHO it would be nice if the API you've developed here more closely matched the existing [Robot class](http://gpiozero.readthedocs.io/en/v1.2.0/api_boards.html#robot). This would involve removing the `t` (time) parameter from each of your existing methods, and your 'example script' would then become:
```python
robot = iRobot("/dev/ttyUSB0")
robot.beep()
robot.forward()
time.sleep(2)
robot.left()
time.sleep(1)
robot.right()
time.sleep(2)
robot.left()
time.sleep(1)
robot.backward()
time.sleep(2)
robot.stop()
```

See also http://gpiozero.readthedocs.io/en/v1.2.0/recipes.html#robot
And if you can add the optional `speed` parameter to your forward/backward/left/right methods (with 0 representing stopped, 0.5 representing half-speed, and 1 representing maximum speed) then that'd be neat too. You could also add a `value` property, which returned the value of `self.direction` (although again it'd more closely match the existing `Robot` API if the `value` property returned a tuple of `(left_wheel_speed, right_wheel_speed)`, with each speed being between 0 and 1 inclusive).

Maybe naming your class `iRobotCreate` would make it visually easier to differentiate from the existing  (generic) `Robot` class?

bennuttall, 2016-05-26:
I think this is a good case for a gpiozero standard, so a new library can provide interfaces that match gpiozero ones.

closed 2018-09-20 by bennuttall