gpiozero/gpiozero #310: fade_in_time / fade_out_time for Robots?

lurch, 2016-05-02:
While thinking about #309 it occurred to me that maybe it'd be useful for Robot's `forward` / `backward` / `left` / `right` / `reverse` methods to have a `fade_in_time` parameter, and the `stop` method to have a `fade_out_time` parameter (and possibly a corresponding `background=True` parameter for each too).
This would sorta-work like the fade parameters on the PWMLED.blink method, and would allow you to smoothly increase the Robot's speed from 0 to e.g. maximum-speed, without suddenly (violently?) flipping from 0 to maximum-speed.

Hmmm, thinking about it a bit more, maybe this would be best accomplished by adding a `fade_to(value, fade_time=1, background=True)` method to PWMOutputDevice, and then PWMOutputDevice.on could become `on(fade_in_time=0, background=True)` and PWMOutputDevice.off could become `off(fade_out_time=0, background=True)` (with the `background` parameters being ignored if the fade_times are 0).
If `on` is called as `on()` then it'll simply switch the PWM duty-cycle to 100% and immediately return, just as it does now. If `on` is called as `on(2)` then it'll spend two seconds smoothly increasing the PWM value from whatever-it-is-currently to 100%, and then return at the end of the two seconds (leaving the PWM at 100% afterwards). If `on` is called as `on(2, False)` then it'll create a new background thread, and immediately return; with the background thread then taking 2 seconds to increase the PWM value from whatever-it-is-currently to 100%.

And then in the `Motor` class, you'd want to be able to do e.g.
```python
motor.value = 0.5
motor.fade_to(-0.8, 4)
```
which would spend 2 seconds fading-down the `motor.forward_device` from 0.5 to 0, and then spend 2 seconds fading-up the `motor.backward_device` from 0 to 0.8.

Neat idea, even if I say so myself! ;-)

pcopa, 2016-05-02:
I've got no experience with robot motors, but have used serial motors in
the past and this is a good idea to prevent the full speed forward to full
speed backwards issue. A delay between forward and backward may also help.
What about shunting/shorting the anchor to stop the motor faster.

lurch, 2016-05-03:
I started working on this yesterday, and it's still a work-in-progress, but so far it's coming together really well... :-)
(technical details: I've renamed all the 'blink' machinery to 'sequence' (`_sequence_thread`, `_stop_sequence`, etc.), I'm now using that for both blinking and fading, and I've pulled it out into a reusable `SequenceMixin`. So far I've got DigitalOutputDevice, PWMOutputDevice, RGBLED, Motor and LEDBoard all using the new mixin.)

@pcopa As mentioned elsewhere I'm not that experienced with electronics...
What's a "serial motor"?
What does "shunting/shorting the anchor" mean?

goloplo, 2016-05-03:
Hi lurch,
It is a good idea `genius.like()`  it will give an artistic and more humanistic twist to any robot. I'm thinking here to a robot arm, it will make the movement more human like. In animation there is a six law: [Slow in slow out](https://www.youtube.com/watch?v=fQBFsTqbKhY).
Regards.

waveform80, 2016-05-08:
Great idea - just a hint on implementations: remember that source can accept *any sequence* including a finite one. For example, to fade in a PWMLED you can simply do:

```python
from gpiozero import PWMLED
from time import sleep

led = PWMLED(17)
led.source_delay = 0.01
led.source = (i / 1000.0 for i in range(1000))
sleep(10)
```

So it's effectively like assigning a temporary waveform to the device and leaving it in the state at the end of the waveform. That said, it's asynchronous (involves a background thread) and precludes using source with the imperative methods (although it's probably reasonable to assume people won't mix the two on a single device...).