gpiozero/gpiozero #314: Add RC Electronic Speed Controller (ESC)

SteveAmor, 2016-05-06:
Given #248 to add a servo, I thought I would suggest adding ESCs as well. Control of ESCs may help shape the thoughts on control of servos.  I'll also add some servo info in #248.  I've been programming servos and speed controllers, in assembly, for drones for the past 5 years.  Please bear with my observations from an OOP point of view.

Standard ESCs respond to the same PWM that a standard servo would.  A 1msec pulse at 50Hz will switch the motor off and a 2msec pulse will run the motor at full speed.  Pulses between 1 and 2msecs proportionally control the motor speed (brushless or brushed).  Modern ESCs will be happy with a PWM frequency of 400Hz.

There is a recent enhancement called OneShot which reduces the pulse width to a value between 125usec and 250usec (stop and full speed respectively).  This allows faster refresh and faster reaction time.

The third type of ESC is a bidirectional ESC.  Here, the minimum pulse width (125usec or 1msec) is max speed in one direction and the maximum pulse width (250usec or 2msec) is max speed in the other direction.  Half the pulse width (200usec or 1500usec) is motor stop.

Just to throw another feature in, most modern ESCs can have their firmware updated via the "servo" lead using a 1 wire protocol.  You may want a passthrough mode to be able to download new firmware to them and potentially configure internal parameters (that's for the future - just useful to know for now perhaps).

If an ESC receives no PWM (i.e. 0% duty cycle) it will be in a mode which prevents the motor from running.

I would suggest that when a Standard ESC object is created, by default set min = 1msec, max = 2msec, 50Hz, start position = 1msec.  Allow them to be changed and when you call ESC.on() it then starts the PWM duty cycle.  ESC.off() will set the PWM duty cycle to 0%.

Elimination of jitter on the PWM signal is important.

lurch, 2016-05-07:
While 'internally' it's implemented the same as a Servo, maybe 'externally' an ESC should have the same API as a [Motor](http://gpiozero.readthedocs.io/en/latest/api_output.html#motor) ?

waveform80, 2016-06-15:
Interesting stuff to know but as @lurch says this should probably have an API similar to Motor. That said, it's looking like several devices use PPM (according to [this EE SO](http://electronics.stackexchange.com/questions/129961/how-to-get-the-pwm-frequency-and-duration-of-each-pulse) this is PPM implemented via PWM) as an interface. I'm tempted to make a derivative of PWMOutputDevice called PPMOutputDevice and have it deal with frequency and pulse-width calculations based on constructor parameters, then derive Servo from that (and possibly ESCMotor in future).

joan2937, 2016-06-15:
That EE SO link may be misleading (or just plain wrong).  As far as I am aware servos and ESCs are driven by PWM signals.  

PPM may be used on the radio link between a  RC transmitter and RC receiver to encode the multiple channels of PWM to be sent to the various actuators.

lurch, 2016-06-15:
@joan2937 From what I understand, PWM and PPM are more or less the same thing, and from the microcontroller point of view you do indeed drive servos using PWM signals.

I think it's the __interpretation__ of the signal where the difference between PPM and PWM comes into play, i.e. when a signal is being interpreted as PPM, it's the width of the individual pulses that matters; whereas when a signal is being interpreted as PWM, it's the ratio of the on-time to the off-time that matters. https://en.wikipedia.org/wiki/Servo_control
And from a quick google, it seems people tend to use PWM / PPM / PDM to mean different things (and possibly incorrectly). It's quite possible Im using the terminology incorrectly too :-S

joan2937, 2016-06-15:
@lurch My understanding was guided by the posts from https://www.raspberrypi.org/forums/viewtopic.php?p=904341#p904341.

waveform80, 2016-06-15:
@joan2937, @lurch thanks for the info. Given the profusion (confusion?) of terms in the wild, I'll try and come up with something suitably uncontroversial for the class names - suggestions welcome!

Incidentally, @joan2937 - many thanks for pigpiod! I've been playing with GPIO Zero's various pin implementations most of the afternoon with some cheap servos and the pigpiod results are perfect - rock solid, no jitter at all (hooked up the oscilloscope while testing and the differences between soft and hard PWM are pretty obvious - to say nothing of the effect on servos).

bennuttall, 2016-06-15:
> Incidentally, @joan2937 - many thanks for pigpiod!

+1 to that. I demonstrated remote GPIO to Eben and we ended up putting it in the Raspbian image off the back of that! It's also used by Scratch.

lurch, 2016-06-15:
> Given the profusion (confusion?) of terms in the wild

Heh. I guess strictly speaking, what a Servo interprets *is* Pulse Width Modulation (as it's only the width of the pulse that matters), and what we usually refer to as PWM would more accurately be Pulse Proportion Modulation ;-)

joan2937, 2016-06-15:
When Tim mentioned pigpio was going to be used in Scratch I was a bit dubious (for reasons best not gone into).  As I said to him "That's good news - I suppose - just hope it doesn't bring too many bugs crawling out from the dark spaces."

pcopa, 2016-06-15:
Off topic: The pigpio code with the pigpiod daemon is working fine on my
rpi3
I had some AM2320's (I2C Temp&Hum sensor) laying around and wanted to see
if I could find a way to hook them up to gpiozero.   The only code I could
find was pigpio using the pigpiod daemon and I did look into adapting the
code into gpiozero, but gave up because it used the daemon.

SteveAmor, 2016-06-16:
For note, `active_low` wouldn't be required for an ESC as 1000usec should be off and 2000usec would normally be full power.

lurch, 2016-06-16:
Ahhh, just re-read your initial comment and see that most (?) ESCs are unidirectional, so I guess they'd want the same API as my proposed [UnidirectionalMotor](https://github.com/RPi-Distro/python-gpiozero/issues/315) and only bidirectional ESCs would have the same API as [Motor](http://gpiozero.readthedocs.io/en/stable/api_output.html#motor). In fact, from the software PoV is there any difference between a continuous-rotation servo and a bidirectional ESC?

SteveAmor, 2016-06-16:
Ok :-) `active_low` would actually be quite good for a bidirectional ESC as it's usually a pain to change the motor wires to make the motor rotate in the opposite direction.  A bidirectional ESC usually has 1460usec as its mid point where it doesn't run the motor.  Home made continuous-rotation servos tend to have varying mid point values where they are at rest (depending on the tolerance of the resistors you use in the feedback loop).  Other than that, I don't think there is a difference between a continuous-rotation servo and a bidirectional ESC and both should have `active_low` :-)