gpiozero/gpiozero #320: Broken link in documentation

lurch, 2016-05-10:
I've just noticed that the "Contribute" link on http://gpiozero.readthedocs.io/en/latest/index.html#development gives a 404 error.

bennuttall, 2016-05-24:
@waveform80 is that a case of changing the link in README to an absolute link, or can it be fixed otherwise?

waveform80, 2016-05-26:
Good question - I'll have a play when I get a sec (probably not too hard)

waveform80, 2016-08-14:
Yeah, an absolute link fixes it. I'd kinda like to include "Contributing" as a chapter in the docs, though (actually I think it *has* to be a chapter or at least a sub-section if you link to it from elsewhere in the docs).

I'll see what I can do (although I suspect it's going to be a horrid hack because GitHub's using file-like addressing, i.e. CONTRIBUTING.md while sphinx will assume URLs shouldn't be modified - i.e. it'll try and link to something like CONTRIBUTING.md but it should be CONTRIBUTING.html - anyway, I'll come up with some sort of workaround).

commit 335ccdfb (2016-08-14): Fix #320

Yup, dirty horrid hack but the result is okay.
commit 9804e653 (2016-08-14): Merge pull request #403 from waveform80/fix-contributing-link

Fix #320
closed 2016-08-14 by waveform80