gpiozero/gpiozero #325: Shutdown button

goloplo, 2016-05-18:
It is possible to add a shutdown button class? 
It can be for RPi used as desktop or for a RPi robot or other aplications. The idea is that when I use my robot I don't have screen access so I want to shutdown it safely.

waveform80, 2016-05-19:
It's a nice idea, but I'm not sure it warrants a whole class. Basically this is just associating the shutdown request with a button press, which could be done like so:

```python
from gpiozero import Button
from subprocess import check_call
from signal import pause

def shutdown():
    check_call(['sudo', 'poweroff'])

shutdown_btn = Button(17)
shutdown_btn.when_pressed = shutdown
pause()
```

lurch, 2016-05-19:
> It's a nice idea, but I'm not sure it warrants a whole class.

Exactly what I was thinking ;-)
Possibly worth adding it to the Recipes?  Alternatively I guess you could use the 'hold' stuff, so that it doesn't shutdown unless the button is held down for e.g. 2 seconds? (to prevent accidental shutdowns)
```python
from gpiozero import Button
from subprocess import check_call
from signal import pause

def shutdown():
    check_call(['sudo', 'poweroff'])

shutdown_btn = Button(17, hold_time=2)
shutdown_btn.when_held = shutdown
pause()
```

And if #261 gets fixed, you could have e.g. a 1-second press doing a reboot, and a 3-second press doing a shutdown ;)

goloplo, 2016-05-19:
Ok, but this means that the button will be predefine at pin 17? Why not let the user to choose the pin?
If I do a robot .py code, I want to define it shutdown(21).

lurch, 2016-05-19:
> If I do a robot .py code, I want to define it shutdown(21).

You can do that in the sample code above by just changing the `17` to `21` - there's nothing hardcoded.
( And don't forget that these are "GPIO numbers" and not "physical pin numbers" http://gpiozero.readthedocs.io/en/stable/recipes.html#pin-numbering )

goloplo, 2016-05-20:
Yes I know, but this means it will not become part of the next release?

waveform80, 2016-05-21:
I think it should be part of the recipes - it's a decent demonstration of the sort of thing that can be done with a button push (and I especially like @lurch's suggestion of using `when_held` to avoid accidental shutdowns), but no I don't think we'll be adding a class for this.

bennuttall, 2016-05-24:
We won't add this to the gpiozero source, but we will add it as an example to the recipes page of the docs.

goloplo, 2016-05-29:
It is working perfect, thanks.

commit 963b9628 (2016-05-29): Add Shutdown button example to recipes, close #325
closed 2016-05-29 by bennuttall