gpiozero/gpiozero #326: RGBLED Initialization failed

serenagrl, 2016-05-22:
Hi,

When I tried to create a new RGBLED,

led = RGBLED(red=26, green=19, blue=13)

I get the following error:
TypeError: integer argument expected, got float.

bennuttall, 2016-05-22:
Are you using the latest version of gpiozero?

Try updating with

```bash
sudo apt-get update
sudo apt-get install python-gpiozero python3-gpiozero
```

and see if you can reproduce.

Can you confirm that this occurs with nothing else in the file, just:

```python
from gpiozero import RGBLED

led = RGBLED(red=26, green=19, blue=13)
```

Thanks

bennuttall, 2016-05-22:
(didn't mean to close)

serenagrl, 2016-05-23:
Yes, I'm using the latest version. 1.2.0

The code failed exactly at that line.

bennuttall, 2016-05-23:
I can't reproduce this. Can you please paste in the full stack trace of the error? That'll help.

Also if you're using a virtualenv, please let us know that's the case.

Thanks

lurch, 2016-05-23:
Also might be useful to say what version of Python you're using, which model of Raspberry Pi you're running on, and the complete source-code of your script that is causing this error.

serenagrl, 2016-05-23:
I'm using Python 3.5.1 on Raspberry Pi 2 running on Raspbian Jessie. 

Code:
import gpiozero as gpio
_led = gpio.RGBLED(26,19,13)

Stacktrace:

Traceback (most recent call last):
  File "07-rgb-led.py", line 4, in <module>
    _led = gpio.RGBLED(26, 19, 13)	# Create RGB LED.
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/devices.py", line 122, in __call__
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/output_devices.py", line 565, in __init__
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/output_devices.py", line 565, in <genexpr>
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/devices.py", line 122, in __call__
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/output_devices.py", line 302, in __init__
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/output_devices.py", line 47, in __init__
  File "/usr/local/lib/python3.5/site-packages/gpiozero-1.2.0-py3.5.egg/gpiozero/pins/rpigpio.py", line 122, in output_with_state
TypeError: integer argument expected, got float

lurch, 2016-05-23:
I've just been investigating this, and found out why @bennuttall  wasn't able to reproduce it - despite the confusing error message, this bug actually only occurs in the very _latest_ Github version of GpioZero (so I've got no idea how @serenagrl set up their Pi), and was introduced (by me) in #324 

Mea culpa! The following code is actually all that's needed to reproduce it:
(it works with MockPWMPin, so all the unit-tests pass, but not with a 'real' RPiGPIOPin - see also #212 )
```python
from gpiozero import PWMOutputDevice
device = PWMOutputDevice(5)
```

#327 fixes it.

bennuttall, 2016-05-24:
Thanks @lurch.

@serenagrl any reason you were using the latest code from github and not the released version?

Two options: reinstall the latest released version, or update to the very latest version from github (with the fix) and try again.

Let me know you get over this issue and I can close this.

Thanks

lurch, 2016-05-24:
@bennuttall Until #327 gets merged, the very latest github version won't contain the fix yet ;-)

bennuttall, 2016-05-24:
Oh

serenagrl, 2016-05-24:
I'm sorry that I don't know what you mean by getting a different version. I'm new to RPi and Raspbian.

For your information, gpiozero fails on all 3 of mine and my friends' newly installed Raspberry Pi. On my fren's RPi 3, it failed complaining something about magic number. For RPi 2, it complained that "ImportError: no module found gpiozero".

I tried uninstalling - sudo apt-get remove python3-gpiozero 
and then reinstalling - sudo apt-get install python3-gpiozero

But I still get the error. I quite like the simplicity of gpiozero and so I searched around and I saw somewhere that we can install it with:

pip3.5 install gpiozero

So that's what I did. After that, everything was working fine until I tried to use the RGBLED. So, I actually don't know that I got a later version. I thought I was using 1.2.0 since everything shows that number.

Thanks.

lurch, 2016-05-26:
Just out of curiosity, why are you using Python 3.5 and not the Python 3.4 that ships by default with Raspbian?

Could you get your friend to file a separate issue about his "magic number problem"?

bennuttall, 2016-05-26:
@serenagrl gpiozero is installed by default on the latest Raspbian image. If you have an older image you can install it with:

```bash
sudo apt-get update
sudo apt-get install python-gpiozero python3-gpiozero
```

Where did you find instructions saying to install with `pip3.5 install gpiozero`? Installing with pip is an alternative, and any version installed with pip will override the installation with `apt`.

closed 2016-05-26 by bennuttall

serenagrl, 2016-05-27:
@lurch I was curious about the new async and await syntax and wanted to see if it is the same as C#.

I will talk to my friend and see if he can raise the case.

@bennuttall I always run the update and upgrade commands of apt-get when I boot-up my RPi and I have tried to reinstall gpiozero many times. I know it comes by default, I can see the message when I ran the install command but it just doesn't work. I just type import gpiozero in a python3 shell and it will complain no module named 'gpiozero'.

I can't recall where I found the instruction to use pip. It was a comment from someone on a post and I decided to give it a shot. Since I am using python 3.5.1, so I use pip3.5 instead and it worked perfectly after that.

lurch, 2016-05-27:
The problem you're running into is that when you use apt-get to install python3-gpiozero, that only works with the 'system' python3 (i.e. Python 3.4) that was _also_ installed with apt-get. Because you're using Python 3.5 which *wasn't* installed via apt-get, then Python 3.5 isn't able to make use of any libraries (such as gpiozero) that were installed via apt-get, which is why you need to use pip3.5 to install the libraries (into a  location that's usable by Python 3.5).

If I get time next week I'll try setting up Python 3.5.1 (I'm assuming you must have built it from source?) on one of my Pis, and see if I can reproduce your pip-install problem.

serenagrl, 2016-05-27:
@lurch Thanks for the explanation. I'm so n00b in all this *HeHe*

Is there a way to get Python 3.5 to make use of the libraries obtained from apt-get?

Yeah, I downloaded the source, compiled and installed.

lurch, 2016-05-27:
> Is there a way to get Python 3.5 to make use of the libraries obtained from apt-get?

No idea, it's something I've never tried.

bennuttall, 2016-05-27:
@serenagrl It looks for "local" packages first. First your current directory, then your pip installed packages, then your apt-get installed packages.

For further information on where Python looks for modules see `sys.path`:

```
>>> import sys
>>> sys.path
['', '/usr/lib/python3.4', '/usr/lib/python3.4/plat-arm-linux-gnueabihf', '/usr/lib/python3.4/lib-dynload', '/usr/local/lib/python3.4/dist-packages', '/usr/lib/python3/dist-packages']
```

Try uninstalling the pip versions first.

PR #327 (2016-05-26): Don't (implicitly) set an initial_value in PWMOutputDevice's init method via the super() call
fixes #326  (which is a bug that got introduced with #324)

Unfortunately we can't add a unit-test for this, due to the problems with multiple PWM tests causing segfaults (which is something I'm also very close to fixing).
commit 3ee15418 (2016-05-26): Don't (implicitly) set an initial_value in PWMOutputDevice's init method via the super() call

fixes #326